shining-man/bsc_fw #1: Doku request

stxShadow, 2022-10-20:
Moin Florian ! 

vielen Dank für dieses Projekt. Ich betreibe den YT Kanaln "Meine Energiewende" und in unserem Discord Chat sind eine Leute wirklich "heiss" auf Dein Projekt. -> NEEY Balancer abzufragen ist bisher leider nur ein Traum.

Daher würde ich gerne im Namen des Kanals um etwas Doku bitten. Wir hatten auch schon auf "meintechblog.de" geschrieben. Du erreichst mich unter jens@solwatch.eu.

Viele Grüße
Jens

shining-man, 2022-10-21:
Nicht Florian, sondern Tobias;-)
Ich habe mit der Doku begonnen und werde sie nach und nach erweitern. 
Die Doku ist im Wiki.

dersch81, 2022-10-21:
Dein Projekt macht ziemlich Lust es einzusetzen. Ich wollte das Meiste tatsächlich über den Pi auf dem VenusOS mit NodeRed abbilden. Aber mit deinem Projekt macht das dann etwas mehr Spaß! Wo bekomme ich deine Draft Platine her? Würdest du mal eine Batch davon herstellen lassen? 

Ich habe hier die JBD als auch die NEEY rumliegen und fange bald an meine zwei 16s Pakete als auch das ESS zu bauen. 

Da VenusOS sich nur eines der über USB angebundenen BMS als Master nimmt und damit die Statistik nicht stimmt weil das zweite, dritte, vierte, etc BMS nicht beachtet wird suche ich noch eine Lösung Venus die Daten aufbereitet zur Verfügung zu stellen. Mit deiner Lösung wäre das wahrscheinlich am besten ohne dabei über weitere Middleware zu gehen. Aber bezgl des WIE habe ich noch einige Fragezeichen. Also wir brauchen DOKU, MEHR DOKU :)