shining-man/bsc_fw #9: MQTT Verbindung mit Benutzername und Passwort

shining-man, 2022-11-10:
### Discussed in https://github.com/shining-man/bsc_fw/discussions/8

<div type='discussions-op-text'>

<sup>Originally posted by **flashy02** November 10, 2022</sup>
Hallo,
habe gerade gesehen, dass ich bei der Einstellung im Web-Interface keine Möglichkeit habe MQTT-User als auch PWD einzugeben. Soll dies noch ergänzt werden? Als Basis-Sicherheit im privaten Netzwerk finde ich dies schon sinnvoll, bzw. wäre in meinem Falle auch notwendig damit ich mich mit meinem MQTT-Broker verbinden kann.
Viele Grüße
flashy02</div>

shining-man, 2022-11-10:
Aktuell ist tatsächlich nur eine unsichere Verbindung möglich. 
Ich werde es aber mit in das nächste Release aufnehmen.

flashy02, 2022-11-10:
Super, danke.

shining-man, 2023-02-06:
Funktion ist im Release v0.3.0 enthalten.

closed 2023-02-06 by shining-man