shining-man/bsc_fw #15: Seplos BMS via RS485

DarthPride, 2022-12-10:
Es wäre klasse, wenn das Anbinden eines Seplos BMS via RS485 und BT (neue Hardwareversion) und damit das Auslesen der Daten via MQTT möglich wäre.

Protokoll:
[SEPLOS BMS Communication Protocol_V2.0.pdf](https://github.com/shining-man/bsc_fw/files/10200059/SEPLOS.BMS.Communication.Protocol_V2.0.pdf)

DarthPride, 2022-12-16:
Ich hatte gesehen, dass in der aktuellen Fw schon Seplos als BMS für die Serial-Anschlüsse wählbar ist.

Steht hier schon Funktion hinter?

shining-man, 2022-12-16:
Nein. Ich bin aktuell dabei das Protokoll zu implementieren.

shining-man, 2022-12-24:
Anbei eine erste Testversion. Bitte Testen und Rückmeldung.
BMS Adresse muss vor erst die 1 sein. Wird später einstellbar. Ich weiß bloß noch nicht genau wie und wo. Theoretisch können ja bis zu 15 BMS an einer Schnittstelle hängen.
[firmware.zip](https://github.com/shining-man/bsc_fw/files/10299083/firmware.zip)

DarthPride, 2022-12-25:
Habe die neue Version gerade frisch ausgesetzt. Konfiguration ESP32 + RS485-Board mit RX an GPIO16 & TX an GPIO17.

Leider scheine ich Probleme mit der MQTT-Verbindung zu haben. Sobald ich MQTT versuche zu aktivieren, wirft er alle BT-Verbindungen raus, zeigt MQTT als disconnected an und übermittelt auch keine Daten.

MQTT disabled findet er die BT-Verbindung zum Neey wieder und connected auch vernünftig.

Im Log steht nichts bzgl. des MQTT-Verhaltens.

Da ich nur über die MQTT-Übergabe sehen kann, ob die Seplos-Anbindung funktioniert, kann ich leider nicht weitertesten.

Oder kann ich die Verbindung über Serial 1 irgendwie anders testen?

shining-man, 2022-12-25:
Dies ist eine Testversion, mit ein paar Baustellen. Eine davon ist MQTT.
Mir genügt das Logfile. Einmal mit aktiviertem und einmal mit deaktiviertem MQTT.

shining-man, 2022-12-26:
Jetzt sollte auch wieder MQTT gehen.
[firmware.zip](https://github.com/shining-man/bsc_fw/files/10302011/firmware.zip)

DarthPride, 2022-12-26:
MQTT funktioniert wieder. Ansonsten keine BT/OW-Geräte dran.
Eine Verbindung zum Seplos bekomme ich nicht. Serial 0 & 2 unbelegt, Serial 1 ist Seplos -BMS (IN).

RS485 Board blinkt einmal bei TX (GPIO17) während des Boots des ESP32 für ca. 1sek auf, bleibt danach dunkel. Rx bleibt komplett dunkel. Kein Datentransfer per MQTT für Serial 1.

Hardware ist mit einem Tasmota-Berry-Treiber getestet, hier bekomme ich Werte vom Seplos zurück.

Log nach Reboot:
I (75) LOG: Free Space total=1507328, used=28672, logSize=10873
I (89) MAIN: BSC V0.2.5
I (90) MAIN: bootCounter=1
I (90) MAIN: A
I (1501) MAIN: B
I (1501) MAIN: C
I (1501) MAIN: Init WLAN...
I (1502) MAIN: Verbindung zu Kellernetz
I (1521) MAIN: [WiFi-event] event: 0
I (1637) MAIN: [WiFi-event] event: 2
I (1734) MAIN: [WiFi-event] event: 4
I (1760) MAIN: [WiFi-event] event: 7
I (1763) MQTT: mqttConnect()
I (2649) MAIN: IP-Adresse = 192.168.XXX.XXX
I (2651) MAIN: Init WLAN...ok
I (2655) MAIN: MDNS responder gestartet
I (2658) MAIN: Starte Webserver...
I (2660) MAIN: Starte Webserver...ok
D (5606) MAIN: -> 'task_alarmRules' runs on core 1
D (5606) MAIN: -> 'task_ble' runs on core 0
I (5607) MAIN: Init BLE...
D (5607) MAIN: -> 'task_onewire' runs on core 1
D (5607) MAIN: -> 'task_bscSerial' runs on core 1
I (5608) MAIN: Free Heap: 126948
D (5609) MAIN: -> 'task_i2c' runs on core 1
D (5609) MAIN: -> 'task_canbusTx' runs on core 1
I (5612) CAN: loadCanSettings(): u8_mBmsDatasource=0
I (5613) I2C: I2C begin=1
I (5614) BSC_SERIAL: initSerial u8_mSerialNr=0, funktionsTyp=0
I (5614) CAN: Init CAN ok
I (5617) I2C: Display not found
I (5617) BSC_SERIAL: initSerial u8_mSerialNr=1, funktionsTyp=3
I (5619) BSC_SERIAL: initSerial u8_mSerialNr=2, funktionsTyp=0
I (5829) MAIN: Time: Mon Dec 26 18:54:02 2022

I (5913) MAIN: Init BLE...ok
I (9709) MQTT: mqttConnect()
I (9710) MQTT: MQTT connected

shining-man, 2022-12-26:
Sorry, kleiner Fehler meinerseits.
Als Hinweis. Bluetooth funktioniert in dieser Testversion nicht richtig.
Wenn es nicht geht, bitte weider den Log.
[firmware.zip](https://github.com/shining-man/bsc_fw/files/10304444/firmware.zip)

DarthPride, 2022-12-26:
Ist das absichtlich eine .elf-Datei?
Hat erheblich länger zum updaten gebraucht als die .bin-Files.
Nach Update keine Veränderung. 

Beim Versuch die Version neu zu flashen schmeißt er mich immer nach einer Weile mit einem Komminunikationsfehler "download data fail" raus. Hatte vorher leider kein Log gezogen, wollte schnell neu flashen.

shining-man, 2022-12-26:
Das elf file war ein Fehler. Jetzt das bin file.
Was ist denn da heute los bei mir...
[firmware.zip](https://github.com/shining-man/bsc_fw/files/10304531/firmware.zip)

shining-man, 2022-12-26:
Das elf File kannst du so nicht für ein Update verwenden. Da wird es immer zu einem Fehler kommen. Sorry.

DarthPride, 2022-12-27:
Gerade frisch geflashed.
Er setzt MQTT auf disconnect, wenn ich Serial 1 verbinde. Nehm ich Serial 1 wieder raus, verbindet MQTT wieder.
TX-LED auf dem RS485 Board blinkt nun rhytmisch, Rx bleibt dunkel.
Mit Serial 1 auf Seplos gesetzt wird der ESP32 erheblich träger...

Log nach Reboot mit gesetztem Serial 1, Serial 0/2 leer:

I (113) LOG: Free Space total=1507328, used=176128, logSize=56403
I (168) MAIN: BSC V0.2.5
I (169) MAIN: bootCounter=1
I (2296) MAIN: Init WLAN...
I (2296) MAIN: Verbindung zu Kellernetz
I (2315) MAIN: [WiFi-event] event: 0
I (2453) MAIN: [WiFi-event] event: 2
I (2565) MAIN: [WiFi-event] event: 4
I (2601) MAIN: [WiFi-event] event: 7
I (2603) MQTT: mqttConnect()
I (3465) MAIN: IP-Adresse = 192.168.XXX.XXX
I (3466) MAIN: Init WLAN...ok
I (3472) MAIN: MDNS responder gestartet
I (3474) MAIN: Starte Webserver...
I (3479) MAIN: Starte Webserver...ok
D (3479) MAIN: -> 'task_onewire' runs on core 1
D (7248) MAIN: -> 'task_bscSerial' runs on core 1
D (7248) MAIN: -> 'task_ble' runs on core 0
I (7249) MAIN: Init BLE...
I (7250) BSC_SERIAL: initSerial u8_mSerialNr=0, funktionsTyp=0
I (7252) BSC_SERIAL: initSerial u8_mSerialNr=1, funktionsTyp=3
I (7252) BSC_SERIAL: setReadBmsFunktion SEPLOS
D (7252) MAIN: -> 'task_alarmRules' runs on core 1
D (7253) MAIN: -> 'task_canbusTx' runs on core 1
I (7253) CAN: loadCanSettings(): u8_mBmsDatasource=0
I (7254) CAN: Init CAN ok
D (7254) MAIN: -> 'task_i2c' runs on core 1
I (7255) I2C: I2C begin=1
I (7256) I2C: Display not found
I (7256) MAIN: Free Heap: 99684
I (7264) BSC_SERIAL: initSerial u8_mSerialNr=2, funktionsTyp=0

shining-man, 2022-12-27:
Ich brauche den ganzen Log wenn der ESP mit Seplos ein Weile lief (z.B. 1 Min). 
Den Log dann am besten als File.

DarthPride, 2022-12-27:
[Log.txt](https://github.com/shining-man/bsc_fw/files/10306694/Log.txt)

Hier eine Logfile über ca. 4min, wenn ich den Zeitstempeln glaube.

Gelaufen hat der ESP32 aber bestimmt 15min, nur seh ich das im Log nicht.

shining-man, 2022-12-29:
Anbei ein neuer Teststand. Ich bin jetzt vorerst wieder auf das alte Filesystem zurück. Das bedeutet, dass nach dem Aufspielen der Firmware alle Einstellungen weg sind. Bitte wieder ein Log-file wenn es noch Probleme gibt.
[firmware.zip](https://github.com/shining-man/bsc_fw/files/10321857/firmware.zip)

DarthPride, 2022-12-30:
Per Update aufgespielt, nun sieht man was im Log bzgl. Seplos, leider kommen noch keine Daten per MQTT, sieht aus als wenn die Kommunikation noch nicht funktioniert.

[Log Seplos Checksum wrong.txt](https://github.com/shining-man/bsc_fw/files/10323373/Log.Seplos.Checksum.wrong.txt)

shining-man, 2022-12-30:
Das Senden hatte ich mit dem Logic-Analayer überprüft, das funktioniert. Das macht er auch bei dir, aber er empfängt nichts.
Welche ID hast du am Seplos eingestellt? Du brauchst ID 1! Welche ID ist denn normal eingestellt?
Passt deine Verdrahtung? Welche Hardware hast du und wie ist sie verschalten?

GPIO Zuordnung:
Serial 0: RX 16, TX 17
Serial 1: RX 23, TX 25
Serial 2: RX 35, TX 33

DarthPride, 2022-12-30:
Hardwareversion 10E (die aktuelle, aber ohne BT im Display)

Versionsstände BMS gem. SEPLOS-Software:
Softwareversion 16.4
Protokollversion 2.0
Verkabelung RS485 => A+ auf 2/Orange, B- auf 1/Orange/Weiß und Gnd auf 3/Grün

Die ID müsste eigentlich per default 1 sein, ich habe keine Dip-Switches gesetzt:
https://drive.google.com/drive/folders/1b0b540PZPS8IGyUCsAx5B30VrpIoRjl-
Zumindest wüsste ich nicht, wie ich das sonst ändern könnte.

Das ganze läuft mit dem Tasmota-Berry-Script problemlos, da ich bekomme alle Werte des BMS angezeigt:
https://github.com/Lu-Fi/tasmota-seplos-driver
Daher kann es eigentlich nicht am ESP/RS485-Board liegen...

shining-man, 2022-12-30:
Wie in dem Anhang ersichtlich ist die default Adresse 0. Entweder die Dip-Switches auf Adr. 1 einstellen oder mit dem angehängten Teststand testen. Habe die Adresse auf 0 umgestellt.
[firmware.zip](https://github.com/shining-man/bsc_fw/files/10324669/firmware.zip)

DarthPride, 2022-12-30:
Mit dem ersten Dip hoch (was dann ja ID1 wäre) ist die Verbindung zum BMS via Seplos Software nicht mehr möglich. Bekomme auch kein Rx bei der alten Firmware des BSC.

Bei allen Dips unten (also ID0?) und der neuen Firmware leider auch keine Antwort vom BMS, Rx leuchtet nicht.
Habe beide RS485 Ports am BMS getestet, bei beiden kein Feedback.

Getestet mit Serial 0 & 1.

shining-man, 2022-12-30:
Welche Baudrate hat das Seplos eigentlich? In der Spec. die du mir anfangs gegeben hat steht 9600. In dem Code um letzten Post steht 19200. Was ist richtig? Läst sich das am Seplos einstellen? Ich sende mit 9600 Baud.

DarthPride, 2022-12-30:
19200 ist korrekt, so verbinde ich auch die Seplos-Software.

Ich glaube in den Specs steht 9600, weil das auch mal für eine ältere Hardwareversion stimmte.

Dokumentation in China....

shining-man, 2022-12-30:
Das erklärt zumindest warum es nicht funktioniert. Anbei mit 19200 Baud und ID0.
[firmware.zip](https://github.com/shining-man/bsc_fw/files/10325943/firmware.zip)

DarthPride, 2022-12-31:
Kommunikation klappt jetzt, aber offenbar gibt es einen Fehler bei den empfangenen Daten:
[Log.txt](https://github.com/shining-man/bsc_fw/files/10326910/Log.txt)

shining-man, 2022-12-31:
Ich habe einmal die Checksumauswertung auskommentiert. Bitte wieder ein Log, auch wenn keine Fehler kommen sollten.
[firmware.zip](https://github.com/shining-man/bsc_fw/files/10327488/firmware.zip)

DarthPride, 2022-12-31:
Das scheint das Problem gewesen zu sein. Ich bekomme nun jede Sekunde korrekte Werte per MQTT in meine ioBroker-Instanz:

[Log.txt](https://github.com/shining-man/bsc_fw/files/10327586/Log.txt)

Die Zellspannungen passen soweit (in mV, genauso wie die anderen Werte.
Bei den Temperaturen zeigt er im ioBroker nur 3 Werte an, er müsste aber 6 anzeigen. Zumindest liefert das BMS soviele.
Und ggf. macht es Sinn, die Abfrageintervalle anzupassen bzw. anpassbar zu machen. Jede Sekunde wäre im Grunde nicht nötig, alle zB. 5sek würden auch reichen. 

Vielen Dank, dass du dich da so engagierst 👍 

Guten Rutsch!

shining-man, 2022-12-31:
`Vielen Dank, dass du dich da so engagierst`
Gerne doch:-)

Anbei einer Version mit korigierter Cheksum-Brechnung. Bitte testen und wieder ein Log-File zurück.
[firmware.zip](https://github.com/shining-man/bsc_fw/files/10327838/firmware.zip)

Es werden nur ein paar wenige Werte (Gesamtspannung, Gesamtstrom) sekündlich versandt. Alle anderen Werte werde ca. alle 10 Sekunden gesendet. Dies lässt sich auch schwierig einstellbar machen (zumindest wenn es häufiger als 10s werden sollte), da doch einige Daten per MQTT versandt werden müssen (7x BT BMS, 3x serial BMS, 64 Temperatur Sensoren, ...). Diese werden aufgeteilt über die 10s, damit nicht aufeinmal ein so hoher Traffic entsteht, der dann wieder Bluetooth blokiert. Noch dazu ergibt es eine ungünstige RAM Auslastung, wenn ich auf einmal soviel in den Sendebuffer schiebe.

Die drei letzten Temperaturwerte fehlen, da in die interen BMS Datenstruktur nur drei Werte vorgesehen sind. Alle bisherigen BMS (JK-BMS, JBD-BMS, NEEY) hatten nicht mehr als drei Sensoren. Ich habe die drei übrigen Werte noch hinzufügen, zumindest zum Versandt via MQTT.

DarthPride, 2023-01-01:
[Log.txt](https://github.com/shining-man/bsc_fw/files/10328448/Log.txt)

Frohes Neues Jahr :-)

Sieht soweit gut aus, es kommen weiterhin Daten im ioBroker an.

Der Temp5 (also der letzte) zeigt aber einen ungewöhnlichen Wert an (3669,2).

shining-man, 2023-01-01:
Jetzt sollte auch Temp5 stimmen.
[firmware.zip](https://github.com/shining-man/bsc_fw/files/10328666/firmware.zip)

DarthPride, 2023-01-03:
Sieht soweit gut aus, das System läuft jetzt seit zwei Tagen stabil.
Das Seplos geht halt nach einigen Stunden in einen Sleep-Mode, was dazu führt das das BSC keine neuen Daten bekommt, aber dass sollte ja kein Problem mehr sein, wenn meine Batterie am Deye WR hängt.

Von meiner Seite aus kann der issue zu. Danke :-)

shining-man, 2023-01-03:
Die Unterstützung für das Seplos-BMS ist im aktuellen main-Branch und kommt mit ins nächste Release.

closed 2023-01-03 by shining-man