shining-man/bsc_fw #16: Bluetooth Connection - Statusanzeige / Anzeige gefundener Geräte

DarthPride, 2022-12-11:
Bei der Anbindung meines Neey Balancers als BMS0 über BT ist mir folgendes aufgefallen:

1. Übersicht der gefundenen BT-Geräte ist auf 5 beschränkt. Bei mir sind es immer deutlich mehr, so dass ich immer wieder rebooten musste, bis er mir irgendwann die MAC des Neey angezeigt hat
=> Eine längere Liste wäre klasse, ggf noch durchnummeriert und mit Überschriften

2. Ich bekomme kein Feedback, ob die BT Verbindung steht. Ich hatte gestern abend eine stabile Verbindung, was ich aber nur anhand des provozierten Alarm1 gesehen habe.

3. Er baut die BT Verbindung nicht zuverlässig auf. Ich musste gerade ca. 10 mal rebooten, dann hat er sie aufgebaut.

shining-man, 2022-12-11:
Das bin File der Version 0.2.5
[firmware_025.zip](https://github.com/shining-man/bsc_fw/files/10201639/firmware_025.zip)

kmn13, 2022-12-11:
Vielleicht einmal kurz meine Erfahrung wenn ich nur 1 NEEY verbinden möchte (ich habe hier >10 BLE Geräte herumfunken). 
Meine Infos beziehen sich auf die 0.2.5. - da wurde ja einiges verbessert 👍

zu1) aktuell ist es so, dass die Suche nach dem erforderlichen BLE Geräte bei "Nichterfolg" alle 11s neu gestartet wird, da taucht dann auch bei mir immer eine andere Liste auf und irgendwann ist das Gerät dann drinnen.
Was ich dazu empfehle (spezielle wenn vielleicht mehrere NEEYsv vorhanden sind): in der NEEY App kann man den Geräten bessere Namen als "GW-24S4EB" geben und einfacher identifizieren. Der Name wird ja auch in den Ergebnissen angezeigt.

zu3) @DarthPride Kurze Frage: welche ESP32 Chip bzw. Board verwendest du? 
Ich hatte sie so mancher Hardware massive Probleme: wie finde Devices gar nicht oder keine Verbindung. Von 5 günstig erhältlichen Boards laufen bei mir nur 2 wirklich ohne Fehler (D1 Mini und Keystudio ks0413).

DarthPride, 2022-12-11:
Zu 1) Ich hatte das Problem nur am Anfang, da ich die MAC-Adresse des Neey nicht kannte und ad hoc keinen BT Sniffer gefunden habe, der sie mir ausgibt. Da hätte eine längere Liste geholfen. Sobald die MAC-Adresse eingetragen ist, hat sich das ja erledigt. Ist der Austausch der ständigen Aktualisierung gegen eine auf Knopfdruck „Verfügbare BT Devices anzeigen“ mit Ausgabe einer vollständigen Liste ggf. einfacher und RAM schonender?
Wäre im Grunde aber nur eine Erleichterung, es geht auch ohne. Gibt sicherlich größere Baustellen 😀

Zu 3) ich habe nun auch die 0.2.5 installiert. Jetzt klappt die Verbindung auch trotz häufigem Reboot zuverlässig, wird im Log ja auch entsprechend ausgeworfen.

@kmn13 
Ich nutze einen AZDelivery ESP32S Dev Kit C V4 NodeMCU von Amazon. Hatte ich ursprünglich mal für was anderes gekauft und hier noch rumliegen.

shining-man, 2022-12-11:
Ich habe das Dashboard um eine Kachel mit dem Verbindungsstatus der BT-Devices erweitert. Habe es in dem aktuellen main branch gepusht. Kann daher getestet werden.

[firmware.zip](https://github.com/shining-man/bsc_fw/files/10202202/firmware.zip)

DarthPride, 2022-12-11:
Super, danke :-) 
Ich mach mal zu, die genannten Punkte sind ja geklärt/umgesetzt

closed 2022-12-11 by DarthPride