shining-man/bsc_fw #17: Onewire - 9+ DS18B20-Sensoren führen zu Onewire-Freeze

DarthPride, 2022-12-13:
Ich habe gerade nacheinander mehrere DS18B20-Temperatursensoren angeschlossen.
Bis zum 9. Sensor hat alles problemlos inkl. Übergabe per MQTT funktioniert.

Beim Anschluss vom 10.-12. Sensor hat das System sich aufgehangen:
- Unten wurden nur die bereits vergebenen 9 Sensoren (nicht fett gedruckt) angezeigt.
- Die zugeordneten Sensoren oben sind bis auf den 1. komplett rausgeflogen
- Beide Punkte sind nicht durch Reboot behebbar
- Beide Punkte lassen sich durch Ändern/Speichern nicht beheben

Ich habe leider gerade keine Zeit um das mittels Neuinstallation zu reproduzieren. Würde ich morgen abend nachholen.

shining-man, 2022-12-13:
> Beim Anschluss vom 10.-12. Sensor hat das System sich aufgehangen:

Was genau meinst du mit aufgehangen? Hat das System einen Restart gemacht, oder hat die Weboberfläche nicht mehr reagiert?
Ist das "Aufhängen" passiert als du Sensor 10 angeschlossen hast, oder erst beim Speichern von Sensor 10?

> Die zugeordneten Sensoren oben sind bis auf den 1. komplett rausgeflogen

D.h. die Liste mit den Onewire Adressen war dann leer? War dies erst nach einem Neustart, oder nach dem anschließen von Sensor 10, oder nach dem Speichern der Sensoradresse von Sensor 10?

> Beide Punkte sind nicht durch Reboot behebbar

Wie ist der Zustand nach einem Reboot? Gehen die ersten Senoren wieder, oder ist dann nichts mehr erreichbar?

Sind das orginale DS18B20 oder China-Klone? Wo kommen die Sensoren her? 
Bei den Klonen muss man damit rechnen, dass nicht alles so toll funktioniert: https://github.com/cpetrich/counterfeit_DS18B20
Was passiert wenn du die drei Problem-Sensoren als die ersten drei nutzt?

Für mich ist noch nicht ganz klar, wann das Problem genau auftritt Ich werde einmal ein paar Logs einbauen um das Problem einzugrenzen.

Ich habe aktuell 18 Sensoren in Betrieb, ohne irgendwelche Ausfälle. Bei mir ist der erste Sensor als "OW Adr. (9)" angeschlossen.
Was pasiert, wenn du auch erst mit Adr. 9 beginnst?

DarthPride, 2022-12-13:
Ich habe Sensoren 1-9 belegt und damit Slot 0-8 belegt, dann 10-12 angeschlossen. Dann kam der Freeze.
Die OneWire1-Seite zeigt immer denselben Zustand an: 
Erster Sensor gesetzt, OneWore enabled und 9 Sensoren unten in der Liste die nicht aktualisiert werden.
Ich kann Werte ändern und augenscheinlich speichern, bei Reload/Reboot wieder erster Sensor gesetzt, OneWire enabled und 9 Sensoren in der Liste. Auch wenn ich die Sensoren abziehe bleibt die Ansicht so.
Datenübermittlung via Mqtt ist ebenfalls Tod.

Die Sensoren sollten eigentlich Originale sein. Funktionieren tun alle, die waren vorher schon an ESP8266 in Verwendung.

ich schau morgen mal ob ich das reproduziert bekomme.

shining-man, 2022-12-15:
Ich sehe zwei Probleme, die eingeltich nichts gemeinsam haben:
a) Es kommt beim Anschließen des Sensors 10-12 zu einem Freeze. => Ich selber kann das Problem nicht nachstellen. Im Anhang ein Teststand mit ein paar mehr Logs über die Serielle-Schnittstelle. Wenn du damit Testen kannst und mir dann den Log geben kannst, dann kann ich mir das Problem weiter anschauen.
b) Das Speichern der Adressen funktioniert nicht. => Funktioniert das Speichern generell nicht, oder nur das Speichern der OW-Adressen? Was passiert wenn du nur drei Adressen eingibst, speicherst und rebootest. Funktioniert das speichern dann? Ist im Log ein Eintrag, dass er Probleme beim Speichern hat?

[firmware.zip](https://github.com/shining-man/bsc_fw/files/10240520/firmware.zip)

DarthPride, 2022-12-16:
Offenbar hat es nichts mit der Anzahl der Sensoren zu tun.
Ich hatte gestern an einer neuen Installation auf einem neuen ESP32 alle 18 Sensoren dran inkl. Datentransfer via MQTT.

Abends habe ich es dann aber wieder hinbekommen, den Onewire1-Screen einzufrieren. Diesmal nur mit 18 hinterlegten Sensoren, obwohl nur noch einer dran war.

Ich vermute, dass hier auch das Problem liegt.
Wenn ich ihm spontan hinterlegte Sensoren klaue, findet er das nicht so gut.

Aber (und hier wird es merkwürdig):
Nachdem ich den ESP32 gestern abend komplett aus gemacht habe, hat er heute morgen beim Neustart die Seite korrekt (=1 Sensor angeschlossen) aufgebaut. Gestern abend hat ein einfacher Reboot & Strom an/aus für >10sek dafür nicht ausgereicht.
Dasselbe Verhalten hatte ich einen Tag vorher mit der ursprünglichen Konfiguration auch schon. Eine Nacht stromlos liegen gelassen = Wunderheilung.

Ich spiele die neue Firmware gleich auf, ggf. bekomme ich das Problem noch mal nachgestellt.

shining-man, 2022-12-16:
Zum Logfile:
Ich vermute, dass das Logfile corrupt ist. Wird der Browser mit dem Laden des Logfiles fertig, oder lädt er beständig?
Das corrupte Logfile hatte ich auf meinem Testsystem auch schon einmal. Das Problem ist, wenn ein Reboot zur falschen Zeit während einem Schreibvorgang kommt...

Lösche das Logfile einmal unter Einstellungen->System-> Log löschen und schaue ob sich das Verhalten ändert.

DarthPride, 2023-01-08:
Funktioniert (zumindest bei mir) wie gewünscht, kann daher auch zu

closed 2023-01-08 by DarthPride

shining-man, 2023-01-08:
Was war denn das Problem, dass es zu den Freezes kam und wie hast du es gelöst?

DarthPride, 2023-01-08:
Es lässt sich aktuell schlicht nicht mehr reproduzieren. Es laufen nun 18 Sensoren seit mehreren Tagen zusammen mit einem Neey und dem Seplos ohne Probleme.

Was auch immer es im Detail war, aktuell tritt das Problem nicht mehr auf.