shining-man/bsc_fw #19: Reboot mit NEEYs nach mehreren Stunden Laufzeit

kmn13, 2022-12-26:
Ich habe nun einmal das Thema mit den Reboots nach längerer Laufzeit aus der Discussion in ein Issue gezogen.

_Originally posted by @shining-man in https://github.com/shining-man/bsc_fw/discussions/12#discussioncomment-4402957_

shining-man, 2022-12-26:
Was ergeben denn die Lebenszeichen. Hängt sich ein Task auf?
Gibt es einen Backtrace der Inforamtionen liefert?

Ich habe mir jetzt einen Testaufbau gemacht, womit ich 5 NEEY (5x ESP32) simulieren kann und testen kann. Leider verhät es sich hier anders und es kommt über Tage hinweg zu keinem Reboot.
Mir ist aber im Zuge des Dateisystemwechels von SPIFFS auf LittleFS aufgefallen, dass mit LittleFS häufiger Reboots gibt, wenn 5 NEEYS verbunden sind und gleichzeigt MQTT mit Passwort aktiv ist. Deaktiviere ich MQTT läuft es bei mir mit den siumulierten NEEYs ohne Probleme.
Steht der Reboot in einem zeitlichen Zusammenhang mit einer Debugausgabe? Kommt direkt vor dem Reboot noch eine Debugausgabe auf der Seriellenschnittstelle oder im Logfile? -> Dann könnte es helfen einmal die Debugausgaben komplett zu deaktiveren.
```
//#define DEBUG_ON_FS
#define DEBUG_ON_HW_SERIAL
//#define LOG_TO_SERIAL
```
So würde auf der Seriellen-Schnitstelle noch die Rebootmeldung mit dme Backtrace kommen, aber mehr nicht.

Hast du MQTT aktiv? Wie läuft es wenn du dies deaktivierst. Ein Problem könnte vielleicht sein, dass sich BT und WLAN die Antenne teilen. D.h. der ESP muss umschalten. Ich weis nicht ob der hohe Traffic durch die 5 NEEYs dann zum Problem wird.
Versuchen könntest du einmal MQTT komplett zu deaktivieren.

kmn13, 2022-12-26:
Später als geplant, hier eine Zusammenfassung meiner Logs.
Die Reboots treten leider nun doch bei mir auch auf...das erste Mal konnte ich eines nach ca 56h finden.
Ebenso habe ich vielleicht auch ein Indiz, warum ich das BSC in meinem Netzwerk nicht mehr erreichen konnte.

Hier einmal der Reihe nach:
Info vorneweg: ich hatte für jede Task einen "alive" Eintrag, diese kamen fast immer. Jeodoch wird eine  Verzögerung auch durch "TaskRunSate" angezeigt, somit habe ich meine Meldungen zur Übersichtlichkeit aus den Logs geschmissen (war schon bei 130Mb ;-))....
Die Ausgabe `BLE_HANDLER: NEEY connected: ....` wir am Ende von `BleHandler::run()` ausgegeben.

- Fall 1) Disconnect nach ca. 56h, keine BLE Verbindung mehr und danach Reboot der CPU

Nach einer ersten Meldung `NimBLEClient: disconnect; reason=520,` beginnen sich nacheinander alle BLE Teilnehmer zu trennen. 
Danach wird wieder versucht die Verbindungen aufzubauen. Das klappt nur kurz und endet mit `E NimBLEClient: Connect timeout - no response` (ab _Zeile 216_ im Log).... danach werden die Devices zwar noch gefunden, jedoch keine Verbindung möglich: `E NimBLEClient: Connection failed; status=520`

Nach einige Retrys endet das Spiel dann mit `ASSERT_PARAM(-218959118 0), in arch_main.c at line 358` (_Zeile 5658_) und die CPU rebootet.
Danach klappte in diesem Fall wieder alles und die Devices waren verbuden...
[02_RebootNach56h.log](https://github.com/shining-man/bsc_fw/files/10303312/02_RebootNach56h.log)

- Fall 2) Nach BLE disconnect und reboot startet WiFi im AP Modus anstatt im STA Modus

Gleiches Spiel wie bei Fall 1, jedoch startete nach dem Assert und Reboot Wifi nicht mehr im STA Modus sonder als AP?
_Ca. Zeilen 5925-5970_
Somit ist auch keine Verbindung möglich. Nach dem AP hatte ich letzten nicht gesucht um festzustellen, ob der Webserver auf diesem Weg noch erreichbar ist...
[03_RebootNach4h_Wifi-AP.log](https://github.com/shining-man/bsc_fw/files/10303372/03_RebootNach4h_Wifi-AP.log)

kmn13, 2022-12-26:
in meinem Fall ist noch kein MQTT aktiv.

kmn13, 2022-12-26:
Die Logs ins Filesystem hatte ich nicht nicht aktiv, sondern so konfiguriert:

shining-man, 2022-12-26:
Beim Reboot kommt ja auch ein Backtrace. Hast du das *.elf File von der Version noch, die nach 56h den Rebot macht? Dann könnten wir den Schuldigen vielleicht einfacher ausfindig machen. Oder hast du den Backtrace schon einmal verfolgt?

shining-man, 2022-12-26:
Bei dem Reboot meldet das System noch ein ASSERT_PARAM
Dazu siehe: https://github.com/espressif/esp-idf/issues/7805

shining-man, 2022-12-26:
Man könnte auch noch versuchen den stack vom BT-Task auf 4096 zu vergößern.

kmn13, 2022-12-26:
> Beim Reboot kommt ja auch ein Backtrace. Hast du das *.elf File von der Version noch, die nach 56h den Rebot macht? Dann könnten wir den Schuldigen vielleicht einfacher ausfindig machen. Oder hast du den Backtrace schon einmal verfolgt?

Von dem Log selbst leider nein, aber ich habe es jetzt geschafft, das Problem schneller zu reprodizieren.
Wenn ich nachdem alle NEEY verbunden habe meine WLAN Verbindung kappe (Router Reboot) dann beginnt die Misere mit den BLE Disconnects und dann sind keine Verbindungen mehr möglich bis zum ASSERT....
Es macht keinen Unterschied, ob die Stacksize 3000 oder 4096...

```
I NimBLEClient: Connected event
I (852335) NimBLE: GATT procedure initiated: exchange mtu

ASSERT_PARAM(-218959118 0), in arch_main.c at line 358
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40090564: 00f01d00 00004136 f01d0000
=> 0x40090564: btdm_sleep_check_duration at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/controller/esp32/bt.c:1091
Core  0 register dump:
PC      : 0x4009056b  PS      : 0x00060534  A0      : 0x80086619  A1      : 0x3ffb58e0
=> 0x4009056b: r_assert at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/controller/esp32/bt.c:2019
A2      : 0x00000000  A3      : 0xf2f2f2f2  A4      : 0x00000000  A5      : 0x3f40ebd7
A6      : 0x00000166  A7      : 0xfffffffc  A8      : 0x8000814b  A9      : 0x3ffb5850
A10     : 0x00000000  A11     : 0x3ffb5873  A12     : 0x3ffb581f  A13     : 0x00000038  
A14     : 0x00000000  A15     : 0x3ffb5824  SAR     : 0x00000004  EXCCAUSE: 0x00000000
EXCVADDR: 0x00000000  LBEG    : 0x40086521  LEND    : 0x40086529  LCOUNT  : 0x00000000
=> 0x40086521: r_assert_param at ??:?
=> 0x40086529: r_assert_param at ??:?

Backtrace:0x40090568:0x3ffb58e00x40086616:0x3ffb5900 0x4012ba25:0x3ffb5920 0x40019fb5:0x3ffb5940 0x4001a1f2:0x3ffb5970 0x40043955:0x3ffb5990 0x400892ac:0x3ffb59c0 0x40044712:0x3ffb59e0 0x40045a8b:0x3ffb5a10 0x400874a4:0x3ffb5a30 0x40019d11:0x3ffb5a60 0x40055b4d:0x3ffb5a80 0x4012be1b:0x3ffb5aa0 0x4012c494:0x3ffb5ac0
=> 0x40090568: r_assert at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/controller/esp32/bt.c:2018
=> 0x40086616: r_assert_param at ??:?
=> 0x4012ba25: r_platform_reset at ??:?
=> 0x40019fb5: ?? ??:0
=> 0x4001a1f2: ?? ??:0
=> 0x40043955: ?? ??:0
=> 0x400892ac: llcp_length_req_pre_handler at llc_llcp.c:?
=> 0x40044712: ?? ??:0
=> 0x40045a8b: ?? ??:0
=> 0x400874a4: ke_task_schedule at ke_task.c:?
=> 0x40019d11: ?? ??:0
=> 0x40055b4d: ?? ??:0
=> 0x4012be1b: r_rw_schedule at ??:?
=> 0x4012c494: btdm_controller_task at ??:?
```
...aber leider kann ich damit nicht viel anfangen.

Muss mich einmal auf die Suche machen, warum generell keine Verbindung mehr zu den NEEYs aufgebaut wird...

> Bei dem Reboot meldet das System noch ein ASSERT_PARAM Dazu siehe: [espressif/esp-idf#7805](https://github.com/espressif/esp-idf/issues/7805)

Die Info hatte ich auch gefunden. So wie ich des jedoch gesehen habe gibt es leider noch keine finale Lösung...

shining-man, 2022-12-30:
An meinem Teststand konnte ich das Problem nachstellen, wenn ich, wie du beschrieben hast, den Router ausschalte.
Das Problem tritt meines erachtens auf, da das WLAN dann durch die Suche des Routers, das BT blockiert. Geteilte Antenne, etc..
D.h. eine Störung in den Verbindungen lässt sich hier meines erachtens nicht vermeinden. Wäre auch nicht ganz so schlimm, wenn es wieder zu einem sauberen Connect käme und nicht zu einem Reboot.
Ich habe jetzt den Connect der BT-Devices etwas umgebaut. In meinen bisherigen Tests funktioniert dies jetzt ohne Probleme. Auch bei Router ausschalten.
Bisher habe ich ein BT-Device verbunden und sofort Daten angefordert. Daa gleiche dann mit dem nächsten, u.s.w..
Jetzt verbinde ich erst alle BT-Devices und erst wenn dies erfoglreich war, dann fordere ich die Daten, von den BT-Devices an. Damit ist in der Verbindungsphase kein unötig hoher Traffic.
Vielleicht kannst du es bei dir auch einmal Testen. Das Problem ist, dass ich in meiner Testbench keine Eigenheiten des NEEY simulieren kann.

`Fall 2) Nach BLE disconnect und reboot startet WiFi im AP Modus anstatt im STA Modus`
Konnte bisher theoretisch nicht sein, da er, wenn er einmal mit einem WLAN verbunden war (STA), nach einem Reboot keinen AP mehr gemacht hat. Ich denke, dass das Problem durch einen Speicherüberlauf kam. Er war dann in einem Zustand bei dem nur noch ein Reset geholfen hat. -> Hier kann man erst einmal beobachten ob das Problem weiterhin auftritt.

kmn13, 2022-12-30:
Sorry, dass ich solange nix mehr beigetragen hatte....hatte auch noch mehrer Tests mit unterschiedlicher Anzahl und Core versucht...
Aber die Grippe hat mich niedergerafft und lässt nicht locker...

Bis zu deinem Post mit den geteilten Antennen war mit der Umstand nicht so bewusst - macht aber einiges erklärbar ;-)

Klar, mit WLAN Störungen muss gerechnet werden. Spätestens wenn zuhause (wie bei mir) gelegentlich jemand den Router neu startet weil sich manchmal das 4G Netz verabschiedet ;-)^^

Das Thema mit dem Traffic beim Verbinden könnte hinkommen, meine Beobachtung war:
Nachdem ich in der NIMBLE explizit ein paar Logs umgeschrieben hatte (damit ich nicht den vollen Log aufdrehen muss, weil da kackt die serielle ab)...da ging es bei einem durchlauf plötzlich ... sieht dann so aus, als ob in diesem Fall die Logs ausreichend verzögert hätten ...

Natürlich kann ich mit den richtigen NEEYs testen. Ich kenne das manches verhält sich dann mit der echte HW doch etwas anders als die Simulation ... auch wenn es vielleicht nur etwas timing ist...

Das mit dem Umschalten auf AP ist mir nur 1x bewusst aufgefallen. 
Aber ich werde das weiter beobachten ;-)

kmn13, 2022-12-30:
Habe schon mal was getestet ;-)

- Aktueller V0.2.6 Stand:

Also mit der aktuellen Version verbinden sich die NEEY wieder nach einem WLAN aus. Wobei der dann das Wifi im AP Mode anläuft (mein WLAN benötigt länger als 30s zum booten ;-)...
Ich sage einmal works as coded `if (!connected /*&& !firstWlanModeSTA*/) {`

- Modifizierter Stand um im STA Mode zu bleiben

Testweise habe ich das auskommentierte `/*&& !firstWlanModeSTA*/` wieder aktiviert um einen Verbindung zum WLAN zu erzeugen.
Das klappt beim 1. WLAN aussetzer ganz gut und nach einer Weile sind alle BLE Devices wieder verbunden und schicken die Daten.
Einen 2. Aussetzer innherhalb von 2-3min verkraftet das System leider nicht, dann werden die BLE Devices zwar gefunden, aber Verbindung ist keine möglich.
Es wird dann eine Weile versucht Verbindungen herzustellen, bis dann der ESP mit dem bekannten Assert die Fliege macht:
```
ASSERT_PARAM(-218959118 0), in arch_main.c at line 358
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40090564: 00f01d00 00004136 f01d0000
=> 0x40090564: btdm_sleep_check_duration at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/controller/esp32/bt.c:1091
Core  0 register dump:
PC      : 0x4009056b  PS      : 0x00060d34  A0      : 0x80086619  A1      : 0x3ffb5860
=> 0x4009056b: r_assert at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/controller/esp32/bt.c:2019
....
```
[BLE_TestNeey_CrashToCloseWLANShutOff.log](https://github.com/shining-man/bsc_fw/files/10326496/BLE_TestNeey_CrashToCloseWLANShutOff.log)

Ist die Zeit jedoch länger...bei meinem Test ca. 5min... so klappt es wieder mit dem Verbinden ohne Absturz.

- Was ist mir noch aufgefallen:
Manchmal kommt nach dem Disconnect das Wifi Event 9 = ARDUINO_EVENT_WIFI_STA_LOST_IP
Wenn des Event 9 auftritt, dann ist dies das letzte Wifi-Event und es ist keine Verbindung mehr möglich (auch nach 10min nicht).
Im Router war der ESP mit IP Adresse eingebucht.
```
I (1299) MAIN: [WiFi-event] event: 0
I (1396) MAIN: [WiFi-event] event: 2
I (1798) MAIN: [WiFi-event] event: 4
I (2432) MAIN: [WiFi-event] event: 7
I (65098) MAIN: [WiFi-event] event: 5
>>> skipped: ca alle 3s ein event 5
I (123073) MAIN: [WiFi-event] event: 5
I (124395) MAIN: [WiFi-event] event: 4
I (126594) MAIN: [WiFi-event] event: 7
I (126720) MAIN: [WiFi-event] event: 5
I (185081) MAIN: [WiFi-event] event: 9
```

- Verständnisfrage zur verwendeten connect(advDevice, false) Funktion im _BLEHander:bleNeeyBalancerConnect()_:

Beim ersten Aufruf wird bewusst `deleteAttributes=false` gesetzt, bei den beiden anderen Aufrufen jedoch ohne diesem Argument und somit _default_: `deleteAttributes=true`.
Gibt es da einen speziellen Hintergrund?

shining-man, 2022-12-31:
```
    Verständnisfrage zur verwendeten connect(advDevice, false) Funktion im BLEHander:bleNeeyBalancerConnect():

Beim ersten Aufruf wird bewusst deleteAttributes=false gesetzt, bei den beiden anderen Aufrufen jedoch ohne diesem Argument und somit default: deleteAttributes=true.
Gibt es da einen speziellen Hintergrund?
```
Das `connect` wird nur im Falle eines Reconnect mit `deleteAttributes=false` aufgerufen, da er aus der vorherigen Verbindung schon die Servicedaten hat und nicht noch einmal holen muss. Bei dem ersten Verbinden, bzw. wenn der den Client nicht kennt, dann wird `connect` mit `deleteAttributes=true` aufgerufen.

```
Was ist mir noch aufgefallen:
Manchmal kommt nach dem Disconnect das Wifi Event 9 = ARDUINO_EVENT_WIFI_STA_LOST_IP
```
Wenn der WiFi-Event 9 kam, bleibt er dann in diese State, oder kommt anschließend noch einmal ein anderer Event?

shining-man, 2022-12-31:
Im Falle des WiFi-Event 9 (SYSTEM_EVENT_STA_LOST_IP) könnte man einen Reconnect veranlassen.
Add the case `SYSTEM_EVENT_STA_LOST_IP` in `onWiFiEvent()`:
```
    case SYSTEM_EVENT_STA_DISCONNECTED:
    case SYSTEM_EVENT_STA_LOST_IP:
```

kmn13, 2022-12-31:
> ```
> Was ist mir noch aufgefallen:
> Manchmal kommt nach dem Disconnect das Wifi Event 9 = ARDUINO_EVENT_WIFI_STA_LOST_IP
> ```
> 
> 
>     
>   
> 
> Wenn der WiFi-Event 9 kam, bleibt er dann in diese State, oder kommt anschließend noch einmal ein anderer Event?

Der bleibt dann im State 9...da kommt dann nichts mehr...auch nach 10Minuten noch nicht...

Ich habe deinen Vorschlag (mit kleiner Anpassung, siehe unten) eingepflanzt und es hilft. Durch den getriggerte Re-Connect wird die Verbindung wieder aufgebaut - passt 👍

```
I (514758) MAIN: [WiFi-event] event: 5
I (579370) MAIN: [WiFi-event] event: 9
I (581697) MAIN: [WiFi-event] event: 4
I (583457) MAIN: [WiFi-event] event: 7
```

Mit `SYSTEM_EVENT_STA_LOST_IP` klappt es leider nicht sondern nur mit `ARDUINO_EVENT_WIFI_STA_LOST_IP`.
Die Enumwerte der beiten Datentypen `system_event_id_t` und `arduino_event_id_t` (welcher für WiFiEvent_t definiert wird) unterscheiden sich ab Wert "8":

```
arduino_event_id_t:
  ARDUINO_EVENT_WIFI_STA_CONNECTED, //4
  ARDUINO_EVENT_WIFI_STA_DISCONNECTED, //5
  ARDUINO_EVENT_WIFI_STA_AUTHMODE_CHANGE, //6
  ARDUINO_EVENT_WIFI_STA_GOT_IP, //7
  ARDUINO_EVENT_WIFI_STA_GOT_IP6, //8
  ARDUINO_EVENT_WIFI_STA_LOST_IP, //9
  ARDUINO_EVENT_WIFI_AP_START, //10

system_event_id_t:
  SYSTEM_EVENT_STA_CONNECTED,  //4
  SYSTEM_EVENT_STA_DISCONNECTED,   //5
  SYSTEM_EVENT_STA_AUTHMODE_CHANGE, //6
  SYSTEM_EVENT_STA_GOT_IP,  //7
  SYSTEM_EVENT_STA_LOST_IP,  //8
  SYSTEM_EVENT_STA_BSS_RSSI_LOW, //9
...
  SYSTEM_EVENT_AP_START, // 15
```

shining-man, 2023-01-08:
Ich habe noch einmal ein paar Änderungen gemacht um die 7 Devices stabiler zu bekommen.
Geändert habe ich vor allem, dass ich nur noch alle 1000ms Daten vom NEEY anfordere. 
Und die Sendefrequenz der MQTT Nachrichten habe ich vergrößert. Mal sehen ob dies was bringt.
Kannst du einmal einen Test mit deinen 7 NEEY's machen?
Ich habe einen neuen Branch erstellt: bt_stability

kmn13, 2023-01-09:
Hi, hab mit den Teststand mal gezogen und raufgespielt.

Die Verbindungen (auch Erstverbindung) dauert nun ca. 3x solange.

Folgendes ist mir aufgefallen: es scheint so, als würde die Daten doppelt/mehrfach übertragen.
Normalerweise kamen je NEEY zuerst 241 und 59byte, dann 500ms Pause. Nun kommen 1-8 mal 241 Pakete vom gleichen NEEY mit 4-6ms Pause. 
Bei ein paar Versuchen hat er 1 NEEY nicht verbunden angezeigt und auch keine, wobei die LED am NEEY dauerhaft blau war (siehe Test02.log).
Nach einem WLAN off/on wurden dann alle NEEY verbunden. 

Folgende Logs habe ich erstellt (beim ...02.log habe ich die Datenausgabe von MAC aud ID geändert um besser interpretieren zu können):
[BTStabilityTest01.log](https://github.com/shining-man/bsc_fw/files/10377092/BTStabilityTest01.log)
Device 0 wird nicht verbunden...obwohl am NEEY als verbunden angezeigt (blaue LED).
Nach dem Reset werden alle verbunden. Jedoch werden je nach NEEY oft die Pakete mehrmals abgerufen (zumindest laut Zeitstempel)
Nach WLAN Verlust verliert er dann die Verbindung zu allen NEEYs und baut auch keine mehr auf.

[BTStabilityTest02.log](https://github.com/shining-man/bsc_fw/files/10377093/BTStabilityTest02.log)
Es werden nur 6 von 7 NEEY Verbunden (NEEY leuchtet jedoch blau) und dann werden die Daten auch nur von den 6 NEEY gezogen.
Nach WLAN off/on werden im Endeffekt dann alle 7 verbunden, jedoch mit vielfachen Paketen.

Mehr Tests kann ich heute leider nicht mehr durchführen ;-)

shining-man, 2023-01-10:
`Normalerweise kamen je NEEY zuerst 241 und 59byte, dann 500ms Pause. Nun kommen 1-8 mal 241 Pakete vom gleichen NEEY mit 4-6ms Pause.`
Ich denke das kommt vom NEEY. Der NEYY fordert ein polling von 10-20ms, das verfehle ich knapp mit 1000ms;-) Da ich nicht so häufig abfrage vermute ich, dass der NEEY noch einmal die Daten in seinen Sendebuffer schreibt. Genau kann ich es aber nicht sagen. Wäre für mich aber erst einmal kein Problem.

Das Problem mit dem ersten NEEY habe ich an meinem Testaufbau mit den Nachbau NEEYs auch. Habe das Problem auf meine "ESP-NEEYs" geschoben. Kann es mir aber gerade nicht ganz erklären. Schaut für mich ein bisschen nach einem Speicherproblem aus. 

Für mich wäre erst einmal Interessant wie es mit der Langzeitstabilität ausschaut. Ohne Routerneustart o.ä., einfach nur bei sauberer bestehender WLAN-Verbindung. Am besten mit MQTT.
Die NEEYS disconnecten sich jetzt ab und zu. Das ist mir aufgefallen. Aber er verbindet sofort neu und alles ist wieder ok.
Wie verhält es sich nach 24,48,96,..h Laufzeit?

kmn13, 2023-01-11:
Den Test mit der Langzeitstabilität + MQTT kann ich noch machen.
Muss mir aber erst mal wo einen MQTT Server einrichten, da ich diese Funktion eigentlich nicht benötige habe ich das bisher nicht gemacht.

shining-man, 2023-01-11:
Mit MQTT wäre toll, will dir das aber nicht einfach aufhalsen.
Mit geht es aktuell darum die 7 NEEYs Langzeit stabil zu bekommen. Das verhalten nach einem Reboot bekommen wir dann schon noch hin. Wenn du aktuell keinen MQTT Broker am laufen hast, dann einfach erst einmal ohne.  Wenn das läuft du du Lust hast, wäre es schön wenn du es mit MQT testen könntest.

kmn13, 2023-01-12:
Ich verstehe das schon nicht auf aufhalsen ;-) ich versuche die Tage mal einen MQTT einzurichten und dann mache ich die Test. 😎

shining-man, 2023-01-13:
Ich habe in dem Branch bt_stability eine neue Version gepusht, die die Probleme beim Routerneustart und dem Nicht-Verbinden von Device 1, beheben sollte. Zumindest an meiner Teststation (5x NEEYs, 2x JK-BMS, MQTT) läuft es auch nach vielen Router-Neustarts immer wieder ohne Probleme an. Ich bin gespannt wie es sich bei echten NEEYs verhält.

kmn13, 2023-01-13:
Super 🙌
Habe den neuen Stand bereits gezogen, die ersten Tests sehen gut aus...alle 7 Knechte verbinden sich wie erwartet.
Auch die ersten Reconnects laufen wie erwartet.😎
MQTT habe ich nun auch dazu gepackt ;-)

Ich pflanze das System nun auf meinen anderen Rechner und lass alles einmal ein paar Tage werkeln (mit seriellen Log).

kmn13, 2023-01-16:
Der erste Test hat sich leider aufgelöst, da mein Rechner meinte er muss neu starten...ab im Log nich auffälliges.

Danach habe ich den 2. Lauf gestartet, dieser hat leider nach ca 29,5 Stunden den ESP mit `Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.` zum Reboot gezwungen 😥

[BTStabilityTest02 - Reboot nach 29,4h.log](https://github.com/shining-man/bsc_fw/files/10429512/BTStabilityTest02.-.Reboot.nach.29.4h.log)

Der Backtrace zeigt bei mir: `0x400968dc: prvTimerTask at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/timers.c line 852`

shining-man, 2023-01-17:
Das Exception könnte als Ursache das Starten des Verbindungstimers für den reconnect der WLAN Verbindung haben. Ist aber nur eine Vermutung, da der Fehler nach erfolglosen connectWiFi() kam.
Ich habe jetzt den kompletten Verbindungsaufbau umgebaut. Hat mir sowieso nicht so gut gefallen.
Der Aufbau der Verbinung läuft jetzt in einem gesoderten Task, ohne Timer.

Ich konnte die Ändeung noch nicht ausgiebig testen. Könnte sein, dass es an der ein oder anderen Stelle erst einmal ein kleiner "Rückschritt" ist.

Als positiv betrachet ich, dass die NEEYs jetzt scheinbar erst einmal so weit laufen.
In welchen Abständen verlieren bei dir die NEEYs eigentlich die Verbindung? Ich meine hier nicht den Verbindungsverlust bei WLAN-Ausfall. Wenn sie einer verliert, reist er dann alle mit sich?

kmn13, 2023-01-17:
Kann deine Änderungen leider erst morgen testen ;-)

Zu den Verbindungsabbrüchen (habe gerade nur das Log vom Fehlerfall und ein paar Stunden danach bei der Hand)....

Wenn sich nur 1 NEEY mal kurz vertschüsst, dann bleiben in der Regel die anderen verbunden. Bei einem WLAN Abriss springen allerdings alle über die Kante.
Im Fall vom 29,4h log (hat 34 MB darum lade ich es mal nicht hoch) war der erste WLAN Ausfall (ungeplant) der Fall wo das BSC dann abgeraucht ist.
Davor traten nur einzelne NEEY Abbrüche auf, wo alle wieder die Verbindung herstellen konnten.
Die Zeit zwischen den Disconnects lag so bei 0,1-138min und im Schnitt 29,6min.

Auf das Log vom aktuellen Test habe ich erst morgen Abend zugriff...falls das noch interessant ist...

shining-man, 2023-02-06:
Die Probleme mit Reboots und Exceptions wurden in der Version v0.3.0 behoben.

closed 2023-02-06 by shining-man