shining-man/bsc_fw #20: MQTT keine Alarme

WagnerSystems, 2023-01-04:
Hallo,
teste gerade die Version V0.2.5, laut Log aus dem Web-Interface. 
Auf einem: AZDelivery ESP32 DevKit C V4

Bei mir werden keine Alarme per MQTT übertragen. 
Im Web-Interface werden die Alarme angezeigt. 
Temperaturen vom JK-BMS etc werden seriell ausgelesen und übertragen.

Noch eine Kleinigkeit, die Bezeichnungen der Temperaturen haben einen Schreibfehler, tempature anstatt temperature

`
bsc:bsc/bms/serial/0/tempature/0:.* tempature_0
bsc:bsc/bms/serial/0/tempature/1:.* tempature_1
bsc:bsc/bms/serial/0/tempature/2:.* tempature_2
`

shining-man, 2023-01-04:
Ich habe die Alarme an einem Digitaleingang und an einem Temperatursensor getestet.
Es kam sowohl auf dem Dashboard als über MQTT.

Was war den die Quelle des Alarms der bei dir nicht geht?

`Noch eine Kleinigkeit, die Bezeichnungen der Temperaturen haben einen Schreibfehler, tempature anstatt temperature`
Den kenne ich tatsächlich. Ich habe es bisher nicht korrigiert, da sonst die Daten in Grafana nicht mehr ganz durchgängig sind. Ich werde es aber gkleich einmal korigieren, enn schön ist es nicht.

WagnerSystems, 2023-01-04:
Ich habe beim BMS 7, die drei Alarme aktiviert.
Teste gleich mal einen DI.

[image]

WagnerSystems, 2023-01-04:
Beim DI wird der Alarm übertragen.

shining-man, 2023-01-04:
Ist das Problem nur beim "Alarm keine Daten" oder auch bei den Spannungsüberwachungen?
Welches BMS hast du eingestellt?

WagnerSystems, 2023-01-04:
Ich habe jetzt mal das Verbindungskabel BMS<->ESP32 abgezogen -> kein Alarm. 
Dann ESP32 per "Reboot"-Button neu gestartet -> kein Alarm "Alarm keine Daten".
Kabel wieder eingesteckt -> jetzt kommen Alarm 2 und Alarm 3.

WagnerSystems, 2023-01-04:
O.k., jetzt geht auch Alarm 1. Aber erst nachdem ich auf alle 3 Speichern Buttons gedrückt habe und einen Reboot gemacht habe.

shining-man, 2023-01-06:
@WagnerSystems 
Kannst du es mit dem aktuellen Branch noch einmal testen?

WagnerSystems, 2023-01-06:
Änderungen an den Alarmen werden bei mir jetzt ohne Neustart übernommen.

shining-man, 2023-01-06:
Gibt es noch weitere Probleme, oder kommen die Alarme jetzt wie sie sollen?

WagnerSystems, 2023-01-07:
Die Alarme kommen wie sie sollen, Danke.

closed 2023-01-07 by shining-man