shining-man/bsc_fw #22: release 0.3.0 not booting

juergen-miessmer, 2023-02-06:
I just tried release 3.0.0.  BSC is not booting up.

ets Jun  8 2016 00:22:57<\r><\n>
<\r><\n>
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)<\r><\n>
configsip: 0, SPIWP:0xee<\r><\n>
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00<\r><\n>
mode:DIO, clock div:2<\r><\n>
load:0x3fff0030,len:184<\r><\n>
load:0x40078000,len:12620<\r><\n>
ho 0 tail 12 room 4<\r><\n>
load:0x40080400,len:2916<\r><\n>
entry 0x400805c4<\r><\n>
?

shining-man, 2023-02-07:
Does the boot message loop or does it come once?

If the message comes once after powering up, that is ok. No other messages are output to the interface. 

Background:
The BSC uses all three hardware serials that the ESP32 has. For more debug/log output on the debug port, I would have to use a software serial. Since a software serial on the ESP32 only works reasonably well at a low baud rate and also costs performance, I decided against it. For example, if an exception occurs, I won't get a backtrace over the soft serial anyway.

I have redirected the log output to a file that can be opened via the web interface.

In case of an update, the device should still be reachable under the same IP address. A new installation should create an access point named BSC.

juergen-miessmer, 2023-02-07:
New installation. Message came only once. Had no BSC acces point listed on my mobile. Tried it several times.
I compared the message to the 0.1.5 release and then thought it hangs at boot. Thanks for the explanation.
Tried it again this morning. BSC acces point was now listed and it works. Maybe an issue with my mobile....

closed 2023-06-23 by shining-man