shining-man/bsc_fw #25: Wunsch: auf allen Web-Sites einen „Home“-Button

testpaul999, 2023-02-17:
Guten Tag,

ich würde mir auf allen Web-Sites einen „Home“-Button wünschen.
Am liebsten oben in der Satusleiste.
Dann muss Mann sich nicht immer „zurück-klicken“.

danke

testpaul999, 2023-02-20:
🤩 wäre toll.
Ja habe kann selber kompilieren.
Lerne gerne dazu.
Danke

WagnerSystems, 2023-02-21:
Hallo,
hier ein patch-File mit den Ergänzungen.
[home.patch.txt](https://github.com/shining-man/bsc_fw/files/10793370/home.patch.txt)
Ich habe das Haus Icon durch 2 Pfeile ersetzt weil ich nicht weis wie das mit den Lizenzen bei dem Icon ist.

Für ein Icon einzufügen musst du nur  `&#10094;&#10094;` durch z.B. `<img src='../home.png'>` zu ersetzen und das icon ins root vom spiffs Dateisystem zu schreiben.

testpaul999, 2023-02-21:
Danke, ich schau mir das heute Abend mal an.👍🏻

testpaul999, 2023-02-22:
@WagnerSystems 
Ich gebe zu das ich mit vstudio keine großen Erfahrungen habe.
Kann ich die Home.Patch automatisch „ausführen“?
Oder geht nur die tippel tappel Tour, alle files einzeln?

danke für die Unterstützung

WagnerSystems, 2023-02-22:
Das geht über Git. 
`git apply home.patch` 
Damit werden die Änderungen in die richtigen Dateien eingetragen.

WagnerSystems, 2023-02-22:
Kann erst heute Abend danach schauen.

WagnerSystems, 2023-02-22:
Du musst in der letzten Zeile die Leerstelle löschen.

testpaul999, 2023-02-22:
Sorry, nö.
[image]

WagnerSystems, 2023-02-22:
Versuch mal ein
`git reset --hard`
und dann den neuen Patch im Anhang
[home.patch.zip](https://github.com/shining-man/bsc_fw/files/10806169/home.patch.zip)

testpaul999, 2023-02-22:
Hi, 
Also es blieb dabei, dass die params.h weiterhin einen Patch failed brachte params.h:525
Habe also den Teil rausgenommen und somit hat der rest geklappt.
Dann händisch die eine Zeil in der params.h geändert und rennt. :-)
Danke

WagnerSystems, 2023-02-23:
@shining-man  Ich denke du kannst dieses Issue dann schliessen.

testpaul999, 2023-02-23:
Hallo,
Im Prinzip ja, wäre nur schön, wenn das so übernommen wird in das nächste Release.
Weil wenn sich grundlegend etwas ändert, passt das Patch file nicht mehr.
Danke

closed 2023-02-23 by testpaul999