shining-man/bsc_fw #30: Wunsch: Erweiterung Mqtt Meldungen um Balance Current

testpaul999, 2023-02-27:
Guten Tag,
Bitte die mqtt Meldungen um den Wert Balance-current erweitern.
Sprich mit wieviel Strom wird aktuell gebalanced.
Die Werte sollten hier zu finden sein.
Analyse aus https://www.akkudoktor.net/forum/open-source-software-projekte/jkbms-auslesen-ueber-ble-bluetooth-oder-rs485-adapter-mittels-eps-iobroker/#post-649

Current_Balancer = (((int)receivedBytes_main[63+offset] << 8 | receivedBytes_main[62+offset])*0.001); 

if((receivedBytes_main[139+2*offset] & 0xF0) == 0x0) {
  Balance_Curr = (((int)receivedBytes_main[139+2*offset] << 8 | receivedBytes_main[138+2*offset])*0.001); 
}
else if ((receivedBytes_main[139+2*offset] & 0xF0) == 0xF0) {
  Balance_Curr = ((((int)receivedBytes_main[139+2*offset] & 0x0F) << 8 | receivedBytes_main[138+2*offset])*-0.001); 
}
Serial.print("Current Balancer = ");
Serial.print(Current_Balancer, 3);
Serial.println("V, ");

Und dann wäre es toll, wenn die Temperaturen ihren Sensoren zugeordnet werden.
Aktuell sehe ich im mqtt 0,1,2
Schön wäre T1,T2 und MOS-Temp.

Danke für die tolle Arbeit.

shining-man, 2023-03-02:
Mit dem Code-Schnippsel kann ich nicht wirklich etwas anfangen.
Wenn noch weitere Daten benötigt werden, dass ist die Basis dafür nicht irgend ein Code, sondern eine Spezifikation der Schnittstelle.
Meines Wissens wird der Valance-Current nicht über dies RS485 Schnittstelle übertragen.

Die Nachkommastellen bei der Temperatur schaue ich mir einmal an. Die sollten eigentlich vorhanden sein.

Bei dem Status Laden, Entladen, Balance muss ich erst schauen ob die Info an der RS485 Schnittstelle übertragen wird.

testpaul999, 2023-03-02:
Guten Abend, doch die Werte werden übertragen. Das sieht man an dem Projekt, was ich hier auch verlinkt habe. Analyse aus 
https://www.akkudoktor.net/forum/open-source-software-projekte/jkbms-auslesen-ueber-ble-bluetooth-oder-rs485-adapter-mittels-eps-iobroker/#post-649
Hier werden die Werte auch an mqtt übergeben, daher frage ich ja.
Wenn ich helfen kann…?
danke

shining-man, 2023-03-02:
Dort werden sie aber via Bluetooth gelesen und nicht über die Serielle-Schnittstelle.
In der Spezifikation die ich zur RS485 Schnisttstelle von JK habe, ist da auch nichts erwähnt.
Dein Code-Schnippsel bezieht sich auf das kopieren aus dem BT-Frame. Das RS485 Protokoll hat einen ganz anderen Aufbau.

testpaul999, 2023-03-02:
Mea culpa, das hatte ich so nicht auf dem Schirm.
Schade, aber nicht zu ändern.
Bleibt meine Bitte:
Status Laden, on/off
Status Entladen, on/off
Stautest Balance, on/off und die 
Temperaturwerte mit Nachkommastellen erweitern.

shining-man, 2023-03-02:
Den Issue am besten so lange offen lassen, bis es erledigt ist.

testpaul999, 2023-03-20:
Danke für den Tipp.
Grüße

shining-man, 2023-03-20:
@gropi75 Ich habe folgendes Protokoll, das hoffentlich stimmt. So ganz sicher bin ich mir da nicht immer...
http://www.jk-bms.com/Upload/2022-05-19/1621104621.pdf

gropi75, 2023-03-20:
Ich habe das gleiche Dokument bei mir bzw. habe [mpp-solar](https://github.com/jblance/mpp-solar/issues/112#issuecomment-850585642) verwendet. Inhaltlich passen sie zueinander.
Ich hatte Probleme mit dem 0xB6 - System Working hours. Sonst passt die Beschreibung ganz gut.

shining-man, 2023-06-23:
Is implemented in the current master branch

closed 2023-06-23 by shining-man