shining-man/bsc_fw #32: Wunsch: Kaskadierung zur Nutzung von mehr als 3 JK-BMS

papala24, 2023-03-07:
Hallo shining-man,

das ist mal ein richtig tolles Projekt. Vielen Dank für das zur Verfügungstellen.
Mein Wunsch wäre bis zu 6 JK-BMS einzubinden.
Momentan sind es zwar bloß 3 und somit in der jetzigen Version ausreichend, aber ich möchte in Zukunft gerne noch 3 weitere Packs mit je einem JK-BMS hinzufügen. Da ich ein Freund der Kabelverbindung bin würde ich mich über die Möglichkeit der Kaskadierung freuen.
Im Google-File werde ich mich erstmal mit 3 maximal bestückten Platinen eintragen.
Falls eine Kaskadierung möglich, und auch zeitlich machbar ist, was benötigt man dann außerdem?

Vielen Dank.

Gruß papala24

shining-man, 2023-03-07:
Eine Kaskadierung ist machbar. Sollte theoretisch gar kein allzu großer Aufwand sein.
Ich bin auch ein Verfechter von Kabelverbindungen.
Du brauchst dann nichts weiter als die entsprechende Anzahl der BSC. Diese lassen sich dann über ein Flachbandkabel verbinden. Die einzelnen BSC sollten dabei nicht zu weit auseinander sein.

papala24, 2023-03-08:
Perfekt. Danke für die schnelle Antwort.

shining-man, 2023-06-23:
Mit der serial extension können bis zu 8 weitere serial devices mit dem BSC verbunden werden.

closed 2023-06-23 by shining-man