shining-man/bsc_fw #41: Status und Namen/Kennzeichnungen der BSC Board's

zippeliniot, 2023-04-15:
Es währe es auch gut, dem je Board in de Konfiguration ein Namen und Status zu geben , zB. Name: **BCS01** Staus = **Testboard**, **IP xxx.xxx.xxx.xxx**  , und zB name: **BCS02** Staus : **Produktivboard**, **IP xxx.xxx.xxx.xxx** , sodass man die Board's auseinanderhalten kann , und in de Steuerung anhand des Staus auch die i/o Aktionen (workflows) unterschiedlich steuern kann.

ergänzend: 

juf — Today at 16:31
sieht gut aus, auch mqtt sieht sauber aus. Wäre es möglich dem System einen eindeutigen Hostnamen zu definieren? Ich habe ja zwei laufen und beide heißen „bsc“. Vielleicht in den Systemeinstellungen? Und manuelle IP-Adresse?

shining-man, 2023-04-15:
Sollen der Status ein Textfeld sein oder eine Combobox (Auswahl)?
Was soll mit dem Status passieren?
Für MQTT kann ja schon ein Name (Topic) angegeben werden, würde es genügen diesen auch als Namen für das WLAN zu nutzen?

tueftler0, 2024-01-13:
Gerade mit Metaworld drüber geredet dass es praktisch wäre - Und schon gibt es hier einen alten Eintrag :)

Eigentlich würde es reichen, wenn es kein "MQTT-Device Name" wäre, sondern ein allgemeiner "Device Name".
- MQTT nutzt diesen dann
- Zusätzlich in der GUI Anzeige oben mit Bindestrich getrennt hinter "BSC"
- Und für den Router als MDNS-Name

Dann wäre es eigentlich schön transparent und man könnte die Testsysteme auseinander halten ;)

shining-man, 2024-01-14:
Die Änderung ist im aktuellen Stand mit drin

closed 2024-02-04 by shining-man