shining-man/bsc_fw #42: Wunsch: Einbinden des MPP MPI 10k Wechselrichters

WagnerSystems, 2023-04-22:
Hallo shining-man,
BT Verbindung mit JK-BMS funktioniert bei mir sehr gut. Vielen Dank dafür.
Jetzt würde ich mir wünschen,das ich auch noch meinen Wechselrichter anschließen kann. 
Ich habe einen MPP Solar MPI 10k. Der wird aber auch unter anderem von Infisolar, EASun, Gowe und FSP als OEM vertrieben.

Der Anschluss beim MPI ist seriell. Eine PC-Software für Windows, Mac und Linux32/64 kann man sich beim Hersteller downloaden [https://www.mppsolar.com/v3/download/](url).
Ich habe 2 pdf Dateien mit Protokoll-Beschreibung gefunden.
[mpp_infinisolar_10k_rs232_commands.pdf](https://github.com/shining-man/bsc_fw/files/11300721/mpp_infinisolar_10k_rs232_commands.pdf)
[mg_mgx_rs232_commands.pdf](https://github.com/shining-man/bsc_fw/files/11300722/mg_mgx_rs232_commands.pdf)

Kann aber auch selbst Mitschnitte von der Kommunikation liefern.

Danke.

LuckyDogs

shining-man, 2023-04-22:
Willst du den Wechselrichter auslesen, oder sollen Ladestrom,Ladespannung,... zum Wechselrichter geschrieben werden?
Um wleche Werte geht es konkret?

WagnerSystems, 2023-04-22:
Na, am liebsten natürlich das volle Programm. 
Ich habe leider keinen Plan wie das im BSC umgesetzt werden kann, aber falls du ein Beispiel hättest könnte ich bestimmt die Funktionen ergänzen die ich möchte. Würde die natürlich dann auch teilen. 

Am wichtigsten wären mir 
- PV Eingangsspannung / -strom
- Lade-/ Entladestrom vom Akku
- Übertragung der Werte per MQTT

shining-man, 2023-04-22:
Ich schaue es mir bei Gelgenehit einmal an.
Kannst du mir noch deine Mitschnitte geben?

WagnerSystems, 2023-08-09:
Hier jetzt noch der Mitschnitt von Wireshark
[MPI10k.pcapng.zip](https://github.com/shining-man/bsc_fw/files/12301113/MPI10k.pcapng.zip)