shining-man/bsc_fw #43: Fehler beim compilieren V 0.4

WagnerSystems, 2023-04-22:
Bekomme beim compilieren des aktuellen New-features branches den folgenden Fehler

`Building in release mode
Retrieving maximum program size .pio/build/bsc/firmware.elf
Checking size .pio/build/bsc/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  18.5% (used 60756 bytes from 327680 bytes)
Flash: [=======   ]  72.1% (used 1133721 bytes from 1572864 bytes)
Building .pio/build/bsc/firmware.bin
esptool.py v4.5-dev
Creating esp32 image...
Merged 25 ELF sections
Successfully created esp32 image.
esp32_create_combined_bin([".pio/build/bsc/firmware.bin"], [".pio/build/bsc/firmware.elf"])
Generating combined binary for serial flashing
scons: *** [.pio/build/bsc/firmware.bin] python /home/volker/vmware/bsc_0.4/scripts/merge_bin_esp.py --output_name /home/volker/vmware/bsc_0.4/.pio/build/bsc/bsc-complete.bin --bin_path /home/volker/vmware/bsc_0.4/.pio/build/bsc/bootloader.bin /home/volker/vmware/bsc_0.4/.pio/build/bsc/partitions.bin /home/volker/vmware/bsc_0.4/doc/boot_app0.bin /home/volker/vmware/bsc_0.4/.pio/build/bsc/firmware.bin --bin_address 0x1000 0x8000 0xe000 0x10000: No such file or directory
=========================================================================================================== [FAILED] Took 5.89 seconds ===========================================================================================================`

shining-man, 2023-04-22:
Schaut so aus als hätte er mit der esp32_create_combined_bin ein Problem. Bzw. mit den Pfad der bin files.
Kommentier in der platformio.ini die Zeile "post:scripts/postbuild_bin.py" aus, dann sollte es gehen.
Wenn du das combines binary brauchst, dann musst du schauen an welchem Pfad das Problem liegt. Sind die 4 bin Files die er braucht alle vorhanden? Stimmen die Pfade?

WagnerSystems, 2023-04-23:
Die Pfade stimmen, habe gestern die merge_bin_esp.py jeweils mit nur einem Pfad und bin_address ausgeführt. Da kam kein Fehler.
Ohne postbuild_bin.py funktioniert es.

shining-man, 2023-06-23:
postbuild_bin.py entfernt, da es nicht notwendig ist.

closed 2023-06-23 by shining-man