shining-man/bsc_fw #45: Mehrere Seplos, wie auswerten?

frankol, 2023-04-28:
Hi,
ich habe zwei gleiche Seplos 10C und habe diese über die interne rs485 Schnittstelle verbunden und die DIP Switch am Master 2:on / am Slave 1:on gesetzt.
Diese über den rs485 adapter am esp32 angeschlossen und in der WebUI serial0 auf Seplos gestellt. 

Ich kriege aber nur die Werte vom Master. Wie soll ich es einstellen, dass beide Packs ausgelesen werden können?

Danke!

shining-man, 2023-04-29:
Das geht mit dem Relase noch nicht. Du brauchst dazu den New-Features branch.
Dort musst du serial2 nutzen und kannst dann darunter eintragen bis zu welcher ID er abfragen soll.

[image]

Ich habe selber kein Seplos, aber so viel ich weiß muss der erste Pack ID0 (kein Dip-switch auf on) und der zweite Pack dann ID1, usw. haben.

Bisher hat dies aber noch keiner getetstet, zumindest habe ich bisher noch keine Rückmeldung, dass es funktioniert. Daher gerne eine Rückmeldung über Erfolg oder Misserfolg.
Wichtig ist, dass du serial2 nutzt. Nur an serial2 wird die Abfrage von mehreren IDs an einem Port unterstützt.

frankol, 2023-04-30:
cool, danke für die Info. 
Ich habe die New-Feature Version installiert.
Also ich habe zwei Packs, die ich auf Pack1 und Pack2 mit den DIP Switchen konfiguriert habe (laut Anleitung für zwei Packs). Pack0 gibt es nur, wenn es ein einzelnes Pack existiert. In der Gui habe ich bis ID2 eingestellt, damit dürfte nur die ID 0 leer bleiben.
ID0 existiert nur, wenn man ein einzelnes Pack hat (alle DIP Switche auf 0)
ID1 ist das Master Pack (bei Verwendung von mehreren Packs, ansonsten leer bzw. nicht vorhanden)
ID2 ist das erste Slave Pack

Aber die ankommenden Werte sehen für mich nicht richtig aus.
Cycle sollte ich 1 stehen haben
Die Temperaturen sind um die 16°C im Keller
SoC ist jeweil um die 33%

Und hier die erhaltenen Daten im Anhang
[image]

Außerdem erhalte ich folgende Fehlermeldung, wenn ich Pack1 und Pack2 einstelle:
E (2023-04-30 12:24:58) SEPLOS_BMS: Timeout: Serial=2, u8_lRecvDataLen=255, u8_lRecvBytesCnt=0
E (2023-04-30 12:24:58) BSC_SERIAL: Error: device=2, reason=Cheksum wrong

Bei allen DIP Switchen auf 0 gehts dann, aber nur für den Master.

Ich hoffe du kannst mit meinem Quatsch was anfangen :-)

frankol, 2023-04-30:
Vielleicht wäre es sinnvoll die Anzahl der Packs in der GUI abzufragen, anstatt die IDs, denn ich muss bei der Originalsoftware auch die Anzahl der Packs auswählen, wenn ich eine RS485 Verbindung aufbaue. Ganz ohne Auswahl kriege ich einen TimeOut
[image]

[image]

shining-man, 2023-05-01:
Ok, das mit den Adressen habe ich verstanden. Ich habe das jetzt angepasst und frage nach Anzahl der Packs ab. Je nachdem ist meine erste ID dann 0 oder 1.
Ich lasse zusätzlich die Empfangenen Daten ins Log schreiben. Vielleicht kannst du mir das Log dann geben um zu sehen warum die Werte nicht stimmen.

frankol, 2023-05-01:
cool, danke. 
Die TX LED leuchtet jetzt leider nicht mehr auf. Hier ein Teil des Logs:
```
I (1970-01-01 00:00:00) LOG: Free Space total=1378241, used=5020, logSize=0
I (1970-01-01 00:00:00) MAIN: BSC V0.4.1
I (1970-01-01 00:00:00) MAIN: bootCounter=1
I (1970-01-01 00:00:00) MAIN: HW: 0
I (1970-01-01 00:00:00) MAIN: Free Heap: 203212
I (1970-01-01 00:00:00) WEB_SETTINGS: Free flash entries: 224
I (1970-01-01 00:00:04) MAIN: Free Heap: 198368
I (1970-01-01 00:00:04) MAIN: Init WLAN...
D (1970-01-01 00:00:04) MAIN: -> 'task_ConnectWiFi' runs on core 1
D (1970-01-01 00:00:04) MAIN: mConnectState=1
D (1970-01-01 00:00:04) MAIN: -> 'task_ble' runs on core 0
I (1970-01-01 00:00:04) MAIN: Init BLE...
I (1970-01-01 00:00:04) MAIN: [WiFi] status (a): 255
I (1970-01-01 00:00:04) MAIN: Verbindung zu SH
D (1970-01-01 00:00:04) MAIN: -> 'task_onewire' runs on core 1
D (1970-01-01 00:00:04) MAIN: -> 'task_bscSerial' runs on core 1
I (1970-01-01 00:00:04) BSC_SERIAL: initSerial SerialNr=0, funktionsTyp=0
D (1970-01-01 00:00:04) MAIN: -> 'task_alarmRules' runs on core 1
D (1970-01-01 00:00:04) MAIN: -> 'task_canbusTx' runs on core 1
I (1970-01-01 00:00:04) CAN: loadCanSettings(): u8_mBmsDatasource=9
I (1970-01-01 00:00:04) CAN: Init CAN ok
D (1970-01-01 00:00:04) MAIN: -> 'task_i2c' runs on core 1
D (1970-01-01 00:00:04) I2C: Master begin=1
I (1970-01-01 00:00:04) I2C: Display not found (2)
I (1970-01-01 00:00:04) I2C: Slave 0 (Adr=16) not found (2)
I (1970-01-01 00:00:04) I2C: Slave 1 (Adr=17) not found (2)
I (1970-01-01 00:00:04) I2C: Serial Ext. not found (2)
I (1970-01-01 00:00:04) MAIN: Free Heap: 144564
I (1970-01-01 00:00:04) BSC_SERIAL: initSerial SerialNr=1, funktionsTyp=0
I (1970-01-01 00:00:04) MAIN: TaskRunSate=126
I (1970-01-01 00:00:04) BSC_SERIAL: initSerial SerialNr=2, funktionsTyp=3
I (1970-01-01 00:00:04) BSC_SERIAL: setReadBmsFunktion SEPLOS
I (1970-01-01 00:00:04) BSC_SERIAL: initSerial SerialNr=3, funktionsTyp=3
I (1970-01-01 00:00:04) BSC_SERIAL: setReadBmsFunktion SEPLOS
I (1970-01-01 00:00:04) MAIN: [WiFi] event: 0
I (1970-01-01 00:00:04) BSC_SERIAL: initSerial SerialNr=4, funktionsTyp=0
I (1970-01-01 00:00:04) BSC_SERIAL: initSerial SerialNr=5, funktionsTyp=0
I (1970-01-01 00:00:04) BSC_SERIAL: initSerial SerialNr=6, funktionsTyp=0
I (1970-01-01 00:00:05) BSC_SERIAL: initSerial SerialNr=7, funktionsTyp=0
I (1970-01-01 00:00:05) BSC_SERIAL: initSerial SerialNr=8, funktionsTyp=0
I (1970-01-01 00:00:05) BSC_SERIAL: initSerial SerialNr=9, funktionsTyp=0
I (1970-01-01 00:00:05) BSC_SERIAL: initSerial SerialNr=10, funktionsTyp=0
D (10393) NIMBLE_NVS: ble_store_config_our_secs restored 0 bonds
D (10394) NIMBLE_NVS: ble_store_config_peer_secs restored 0 bonds
I (1970-01-01 00:00:05) MAIN: [WiFi] event: 2
D (10404) NIMBLE_NVS: ble_store_config_cccds restored 0 bonds
D (10406) NIMBLE_NVS: peer_dev_rec restored 0 records
I (1970-01-01 00:00:05) MAIN: Init BLE...ok
I (1970-01-01 00:00:05) MAIN: wlanConnectTimer=35240
I (1970-01-01 00:00:05) MAIN: wlanConnectTimer=35240
I (1970-01-01 00:00:05) MAIN: TaskRunSate=124
I (1970-01-01 00:00:05) MAIN: TaskRunSate=120
I (1970-01-01 00:00:05) MAIN: TaskRunSate=112
I (1970-01-01 00:00:06) SEPLOS_BMS: SeplosBms_readBmsData()
I (1970-01-01 00:00:06) SEPLOS_BMS: SeplosBms_readBmsData()
I (1970-01-01 00:00:06) MAIN: TaskRunSate=96
I (1970-01-01 00:00:06) MAIN: wlanConnectTimer=35240
I (1970-01-01 00:00:06) MAIN: TaskRunSate=32
I (1970-01-01 00:00:06) MAIN: TaskRunSate=0
I (1970-01-01 00:00:07) SEPLOS_BMS: SeplosBms_readBmsData()
I (1970-01-01 00:00:07) SEPLOS_BMS: SeplosBms_readBmsData()
I (1970-01-01 00:00:07) MAIN: wlanConnectTimer=35240
I (1970-01-01 00:00:08) SEPLOS_BMS: SeplosBms_readBmsData()
I (1970-01-01 00:00:08) MAIN: wlanConnectTimer=35240
I (1970-01-01 00:00:08) SEPLOS_BMS: SeplosBms_readBmsData()
I (1970-01-01 00:00:08) MAIN: [WiFi] event: 4
I (1970-01-01 00:00:09) MAIN: wlanConnectTimer=35240
I (1970-01-01 00:00:09) SEPLOS_BMS: SeplosBms_readBmsData()
I (1970-01-01 00:00:09) SEPLOS_BMS: SeplosBms_readBmsData()
I (1970-01-01 00:00:09) MAIN: [WiFi] event: 7
I (1970-01-01 00:00:10) MAIN: TaskRunSate=1
I (1970-01-01 00:00:10) MAIN: IP-Adresse = 192.168.200.35
I (1970-01-01 00:00:10) MAIN: [WiFi] status (b): 3
D (1970-01-01 00:00:10) MAIN: mConnectState=2
I (1970-01-01 00:00:10) SEPLOS_BMS: SeplosBms_readBmsData()
I (1970-01-01 00:00:10) SEPLOS_BMS: SeplosBms_readBmsData()
D (1970-01-01 00:00:11) MAIN: mConnectState=3
I (1970-01-01 00:00:11) SEPLOS_BMS: SeplosBms_readBmsData()
I (1970-01-01 00:00:11) SEPLOS_BMS: SeplosBms_readBmsData()
I (1970-01-01 00:00:12) TIME: Init NTP (IP): server=192.168.200.254
I (2023-05-01 12:38:45) MAIN: Time: 2023-05-01 12:38:45
D (2023-05-01 12:38:45) MAIN: mConnectState=5
I (2023-05-01 12:38:45) SEPLOS_BMS: SeplosBms_readBmsData()
I (2023-05-01 12:38:45) SEPLOS_BMS: SeplosBms_readBmsData()
D (2023-05-01 12:38:46) MAIN: mConnectState=6
I (2023-05-01 12:38:46) SEPLOS_BMS: SeplosBms_readBmsData()
I (2023-05-01 12:38:46) SEPLOS_BMS: SeplosBms_readBmsData()
D (2023-05-01 12:38:47) MAIN: mConnectState=7
I (2023-05-01 12:38:47) SEPLOS_BMS: SeplosBms_readBmsData()
I (2023-05-01 12:38:47) SEPLOS_BMS: SeplosBms_readBmsData()
D (2023-05-01 12:38:48) MAIN: mConnectState=8
I (2023-05-01 12:38:48) SEPLOS_BMS: SeplosBms_readBmsData()
I (2023-05-01 12:38:48) SEPLOS_BMS: SeplosBms_readBmsData()
I (2023-05-01 12:38:48) MAIN: TaskRunSate=0
D (2023-05-01 12:38:49) MAIN: mConnectState=9
I (2023-05-01 12:38:49) SEPLOS_BMS: SeplosBms_readBmsData()
I (2023-05-01 12:38:49) SEPLOS_BMS: SeplosBms_readBmsData()
```
Die letzte Zeile wiederholt sich ohne Anzeichen eines Lebenzeichens auf dem rs485 Adapters

shining-man, 2023-05-01:
Diesmal als fertiges bin-file:
[firmware.zip](https://github.com/shining-man/bsc_fw/files/11366748/firmware.zip)

frankol, 2023-05-01:
ja, Serial1 und ID0 genau das gleiche. Keine Reaktion auf dem rs485 Adapter. 
v40 hingegen funktioniert serial 1 mit id0 z.B.

shining-man, 2023-05-01:
Ist es mit der neuen Testversion (v0.4.1a) das Gleiche?

frankol, 2023-05-01:
ja, bei 0.4.1a ist es das gleiche. Keine Reaktion am Adapter.
`I (1970-01-01 00:00:00) LOG: Free Space total=1378241, used=5020, logSize=0
I (1970-01-01 00:00:00) MAIN: BSC V0.4.1a
I (1970-01-01 00:00:00) MAIN: bootCounter=1
I (1970-01-01 00:00:00) MAIN: HW: 0
I (1970-01-01 00:00:00) MAIN: Free Heap: 203868
I (1970-01-01 00:00:00) WEB_SETTINGS: Free flash entries: 224
I (1970-01-01 00:00:04) MAIN: Free Heap: 199024
I (1970-01-01 00:00:04) MQTT: MQTT: ip=192.168.200.12, port=1883
I (1970-01-01 00:00:04) MAIN: Init WLAN...
D (1970-01-01 00:00:04) MAIN: -> 'task_ConnectWiFi' runs onI (1970-01-01 00:00:04) MAIN: Init BLE...
I (1970-01-01 00:00:04) MAIN: Init BLE...
D (1970-01-01 00:00:04) MAIN: mConnectState=1
I (1970-01-01 00:00:04) MAIN: [WiFi] status (a): 255
I (1970-01-01 00:00:04) MAIN: Verbindung zu SH
D (1970-01-01 00:00:04) MAIN: -> 'task_onewire' runs on core 1
D (1970-01-01 00:00:04) MAIN: -> 'task_bscSerial' runs on core 1
D (1970-01-01 00:00:04) MAIN: -> 'task_alarmRules' runs on core 1
I (1970-01-01 00:00:04) BSC_SERIAL: initSerial SerialNr=0, funktionsTyp=0
D (1970-01-01 00:00:04) MAIN: -> 'task_canbusTx' runs on core 1
I (1970-01-01 00:00:04) CAN: loadCanSettings(): u8_mBmsDatasource=9
I (1970-01-01 00:00:04) CAN: Init CAN ok
D (1970-01-01 00:00:04) MAIN: -> 'task_i2c' runs on core 1
D (1970-01-01 00:00:04) I2C: Master begin=1
I (1970-01-01 00:00:04) I2C: Display not found (2)
I (1970-01-01 00:00:04) I2C: Slave 0 (Adr=16) not found (2)
I (1970-01-01 00:00:04) I2C: Slave 1 (Adr=17) not found (2)
I (1970-01-01 00:00:04) I2C: Serial Ext. not found (2)
I (1970-01-01 00:00:04) MAIN: Free Heap: 154108
I (1970-01-01 00:00:04) MAIN: TaskRunSate=126
I (1970-01-01 00:00:04) BSC_SERIAL: initSerial SerialNr=1, funktionsTyp=0
I (1970-01-01 00:00:04) BSC_SERIAL: initSerial SerialNr=2, funktionsTyp=3
I (1970-01-01 00:00:04) BSC_SERIAL: setReadBmsFunktion SEPLOS
I (1970-01-01 00:00:04) BSC_SERIAL: initSerial SerialNr=3, funktionsTyp=0
I (1970-01-01 00:00:04) MAIN: [WiFi] event: 0
I (1970-01-01 00:00:04) BSC_SERIAL: initSerial SerialNr=4, funktionsTyp=0
I (1970-01-01 00:00:05) BSC_SERIAL: initSerial SerialNr=5, funktionsTyp=0
I (1970-01-01 00:00:05) BSC_SERIAL: initSerial SerialNr=6, funktionsTyp=0
I (1970-01-01 00:00:05) BSC_SERIAL: initSerial SerialNr=7, funktionsTyp=0
I (1970-01-01 00:00:05) BSC_SERIAL: initSerial SerialNr=8, funktionsTyp=0
I (1970-01-01 00:00:05) BSC_SERIAL: initSerial SerialNr=9, funktionsTyp=0
I (1970-01-01 00:00:05) BSC_SERIAL: initSerial SerialNr=10, funktionsTyp=0
D (10511) NIMBLE_NVS: ble_store_config_our_secs restored 0 bonds
D (10511) NIMBLE_NVS: ble_store_config_peer_secs restored 0 bonds
I (1970-01-01 00:00:05) MAIN: [WiFi] event: 2
D (10519) NIMBLE_NVS: ble_store_config_cccds restored 0 bonds
D (10520) NIMBLE_NVS: peer_dev_rec restored 0 records
I (1970-01-01 00:00:05) MAIN: Init BLE...ok
I (1970-01-01 00:00:05) MAIN: wlanConnectTimer=35300
I (1970-01-01 00:00:05) MAIN: wlanConnectTimer=35300
I (1970-01-01 00:00:05) MAIN: TaskRunSate=124
I (1970-01-01 00:00:05) MAIN: TaskRunSate=120
I (1970-01-01 00:00:05) MAIN: TaskRunSate=112
I (1970-01-01 00:00:06) SEPLOS_BMS: SeplosBms_readBmsData() firstAdr=0, CountOfPacks=0
I (1970-01-01 00:00:06) MAIN: TaskRunSate=96
I (1970-01-01 00:00:06) MAIN: wlanConnectTimer=35300
I (1970-01-01 00:00:06) MAIN: TaskRunSate=32
I (1970-01-01 00:00:06) MAIN: TaskRunSate=0
I (1970-01-01 00:00:07) SEPLOS_BMS: SeplosBms_readBmsData() firstAdr=0, CountOfPacks=0
I (1970-01-01 00:00:07) MAIN: wlanConnectTimer=35300
I (1970-01-01 00:00:08) SEPLOS_BMS: SeplosBms_readBmsData() firstAdr=0, CountOfPacks=0
I (1970-01-01 00:00:08) MAIN: wlanConnectTimer=35300
I (1970-01-01 00:00:09) SEPLOS_BMS: SeplosBms_readBmsData() firstAdr=0, CountOfPacks=0
I (1970-01-01 00:00:09) MAIN: wlanConnectTimer=35300
I (1970-01-01 00:00:10) MAIN: TaskRunSate=1
I (1970-01-01 00:00:10) SEPLOS_BMS: SeplosBms_readBmsData() firstAdr=0, CountOfPacks=0
I (1970-01-01 00:00:10) MAIN: wlanConnectTimer=35300
I (1970-01-01 00:00:11) SEPLOS_BMS: SeplosBms_readBmsData() firstAdr=0, CountOfPacks=0
I (1970-01-01 00:00:11) MAIN: wlanConnectTimer=35300
I (1970-01-01 00:00:12) SEPLOS_BMS: SeplosBms_readBmsData() firstAdr=0, CountOfPacks=0
[... 14 lines cut ...]
D (2023-05-01 20:34:54) MAIN: mConnectState=5
I (2023-05-01 20:34:55) SEPLOS_BMS: SeplosBms_readBmsData() firstAdr=0, CountOfPacks=0
D (2023-05-01 20:34:55) MAIN: mConnectState=6
I (2023-05-01 20:34:56) SEPLOS_BMS: SeplosBms_readBmsData() firstAdr=0, CountOfPacks=0
I (2023-05-01 20:34:56) MAIN: TaskRunSate=0
D (2023-05-01 20:34:56) MAIN: mConnectState=7
I (2023-05-01 20:34:57) SEPLOS_BMS: SeplosBms_readBmsData() firstAdr=0, CountOfPacks=0
D (2023-05-01 20:34:57) MAIN: mConnectState=8
I (2023-05-01 20:34:58) SEPLOS_BMS: SeplosBms_readBmsData() firstAdr=0, CountOfPacks=0
D (2023-05-01 20:34:58) MAIN: mConnectState=9
D (2023-05-01 20:34:58) MQTT: smMqttConnectState=0
D (2023-05-01 20:34:58) MAIN: mConnectState=5
I (2023-05-01 20:34:59) SEPLOS_BMS: SeplosBms_readBmsData() firstAdr=0, CountOfPacks=0
D (2023-05-01 20:34:59) MAIN: mConnectState=6
I (2023-05-01 20:35:00) SEPLOS_BMS: SeplosBms_readBmsData() firstAdr=0, CountOfPacks=0
I (2023-05-01 20:35:00) MQTT: MQTT Broker connected
D (2023-05-01 20:35:00) MAIN: mConnectState=7
I (2023-05-01 20:35:01) SEPLOS_BMS: SeplosBms_readBmsData() firstAdr=0, CountOfPacks=0
D (2023-05-01 20:35:01) MAIN: mConnectState=8
I (2023-05-01 20:35:02) SEPLOS_BMS: SeplosBms_readBmsData() firstAdr=0, CountOfPacks=0
D (2023-05-01 20:35:02) MAIN: mConnectState=9
D (2023-05-01 20:35:02) MQTT: smMqttConnectState=1
I (2023-05-01 20:35:03) SEPLOS_BMS: SeplosBms_readBmsData() firstAdr=0, CountOfPacks=0
I (2023-05-01 20:35:04) SEPLOS_BMS: SeplosBms_readBmsData() firstAdr=0, CountOfPacks=0
I (2023-05-01 20:35:05) SEPLOS_BMS: SeplosBms_readBmsData() firstAdr=0, CountOfPacks=0`

shining-man, 2023-05-01:
[firmware.zip](https://github.com/shining-man/bsc_fw/files/11367616/firmware.zip)
Ein neuer Versuch

frankol, 2023-05-01:
[log (2).txt](https://github.com/shining-man/bsc_fw/files/11367721/log.2.txt)

gleiches Verhalten. keine Reaktion am rs485 Adapter. 

Sicherheitshalber nochmal die v40 installiert und da blinkt dann die TX Led.

frankol, 2023-05-02:
ausgezeichnet, es lebt, danke!

Was ist denn mit 3 und 4 gemeint?
Ist fast perfekt. Gibts noch die cellVoltage beim Pack2 irgendwie?
[image]

shining-man, 2023-05-03:
[firmware_with_log.zip](https://github.com/shining-man/bsc_fw/files/11379139/firmware_with_log.zip)
Anbei eine neue Version. Eine mit aktiven Log und eine ohne Esplos Logausgaben. Die ohne Log ist für den Fall, dass es funtionieren sollte, damit du nicht deinen Flash unnötig beschreibst.

Es war noch ein Fehler beim versenden der Daten per MQTT.
Der erste Pack müsste unter serial2 auftauchen und der zweite Pack dann bei serial3, u.s.w.

Wenn es jetzt funktionieren sollte, kannst du es dann auch noch einmal mit einem Pack (ID0) testen. Nur um zu sehen ob da auch noch alles funktioniert.

frankol, 2023-05-03:
genial. Funktioniert beides jetzt. Vielen Dank für die Mühe!

Eine kurze Rückfrage bzgl. Konfigurationsunterschieden zwischen dem BSC und dem BMS. Wenn das BMS die Ladeleistung drosseln möchte (weil irgendein Wert erreicht wurde) aber das BSC es nicht weitergibt, entsteht dadurch irgendein Problem?

Bei zwei Packs gehen ein paar Temperaturen verloren, ist aber unerheblich. Hier die Ausgabe bei zwei Packs:
[image]

und hier mit nur einem Pack:
[image]

Vielen Dank nochmal!

closed 2023-05-03 by frankol

shining-man, 2023-05-08:
@frankol 
Hast du eine Pinnelegung des RS485 am Seplos?
Auf welchem Pin ist RS485-A und auf welchem Pin ist RS485-B und wo GND?

frankol, 2023-05-09:
A liegt auf 2 und 7
B liegt auf 1 und 8

GND auf 3 und 6

alexseuf, 2023-06-17:
@frankol
welche Einstellugnen hast du denn im bsc Menü vorgenommen?
passen die foglenden:
[image]

alexseuf, 2023-06-17:
@frankol
Und mit welcher dip-schalter Stellung funktioniert das mit deiner 2x 10C-Seplos Hardware?
Pack 1 = 0 on; Pack 2 = 1 on
Pack 1 = 5 on; Pack 2 = 1 on ( Variante für can lt seplos Anleitung)
Pack 1 = 1 on; Pack 2 = 2 on (Variante für RS485 laut Seplos Anleitung)