shining-man/bsc_fw #51: JKBMS serial 1 - mqtt topics missing

juergen-miessmer, 2023-08-27:
Hello,
i have two JKBMS conectet to serial 0 an 1.
As long as serial 0 is set to "JK BMS", BSC does not send/update mqtt topics "Cycle" and "BalanceCapacity" for serial 1.
By selecting "nicht belegt" for serial 0, BSC is sending these topics.

Version is V0.4.16. Had done an erase and flashed comletely new for update.

Some minor things i noticed:
Mqtt topic "BalanceCapacity" is named "Cycle Capacity" in the JK app - maybe just a typo.
Topics "balancingActive" and "balancingCurrent" are available via blutooth, but not via serial.
Topics "Cycle" and "BalanceCapacity" are not available via blutooth.
Via blutooth 24 cell voltages are always transmited, even if there are less cells.
By changing "MQTT Device Name" and "MQTT Topic Name" in mqtt settings, BSC crashed and all settings where lost. Had to completely start from defaults, like a fresh clean flashed BSC.
If wifi is temporary not available, BSC falls back into accespoint mode, but then rests there for ever. I suggest to try to reconnect after some minutes, if no device is connected to the BSC-accespoint.

Regards and many thanks for your work.
Jürgen

shining-man, 2023-11-03:
Sorry for the late response and thank you for your feedback.

Regarding your first point: I have looked at why not all values come through and why it affects exactly the two values.
If you have two BMS on serial 0 and serial 1, do the values not come through for either of the two BMS, or only from one of them?

The JK-BMS does not send the same values on the two interfaces (serial+BT). For example, the "balancingCurrent" is not transmitted on the serial interface. 

In the Mqtt theme, I used the text "BalanceCapacity" for convenience, as I already had it. I change it to "Cycle Capacity".

Re: "By changing "MQTT Device Name" and "MQTT Topic Name" in mqtt settings, BSC crashed and all settings where lost.": That's not nice. I'll have a look to see if it's reproducible.

Re: "If wifi is temporary not available, BSC falls back into accespoint mode, but then rests there for ever.": You can set the time for how long it tries to reconnect. I don't have a real solution yet, but I will consider your suggestion.
There is one more thing I would like to point out here: As long as it is searching for the WLAN, it will not fetch any Bluetooth data. I mention this because it has already confused some people.

juergen-miessmer, 2023-11-07:
While both JK-BMS are set, the two values come through for serial 0, but not for serial 1.

I was able to reproduce the crash. 
If i change one value and click the "s" button, then change the second value and "s" - everythink works fine.
But by changing both values an afterword clicking both "s" buttons in a row - the crash happens.
I asume this is caused by clicking the second "s" while the first saving is still in progress. So maybe only because i forgot to click the "s" after changing the first value an then cklicked both....

Some Question to "WLAN connect Timeout":
What is the max. value to set ?
Setting the value to 0 means never falling back to accespoint mode, or to fall back without waiting?

shining-man, 2023-11-07:
"While both JK-BMS are set, the two values come through for serial 0, but not for serial 1." -> I can understand that. I've made a note of it to fix it.

I am currently working on making sure that there is only one save button per page. Then the problem will no longer exist.

0 means that you never fall into AP mode. The WLAN is searched for until it has found it. 
A maximum of 3600s can be set.

shining-man, 2023-12-23:
Issue is fixed with the latest release

closed 2023-12-23 by shining-man