shining-man/bsc_fw #53: ESP32 wird langsam

Elektrofix-OL, 2023-11-12:
Ich habe die Software auf eine NodeMCU mit 4GB-Speicher laufen. Nach dem Flashen war das System sehr schnell, aber seit dem er die Verbindung zu den zwei NEEY's aufgebaut hat, reagiert er sehr langsam.
Im IoBroker habe ich festgestellt, das alle 13-17 sec. neue Daten von NEEY geschrieben werden.
Ist es u.U. möglich die Abfragezeit zu verändern? Es rechen doch Abfragen alle 30-60 Sec.

shining-man, 2023-11-12:
Was heißt "sehr langsam"? Kannst du mir einmal das Logfile geben?

Elektrofix-OL, 2023-11-13:
I (1970-01-01 00:00:03) LOG: Free Space total=836081, used=251, logSize=0
I (1970-01-01 00:00:03) MAIN: BSC V0.4.16
I (1970-01-01 00:00:03) MAIN: bootCounter=1
I (1970-01-01 00:00:03) MAIN: HW: 0
I (1970-01-01 00:00:03) MAIN: Free Heap: 207000
I (1970-01-01 00:00:03) WEB_SETTINGS: Free flash entries: 629
I (1970-01-01 00:00:03) WEB_SETTINGS: readConfig: file not exist
I (1970-01-01 00:00:03) WEB_SETTINGS: writeConfig(): file not exist
I (1970-01-01 00:00:06) MAIN: Free Heap: 200064
I (1970-01-01 00:00:06) MAIN: Init WLAN...
I (1970-01-01 00:00:06) MAIN: Init BLE...
I (1970-01-01 00:00:06) MAIN: Wifi AP
I (1970-01-01 00:00:06) BSC_SERIAL: initSerial SerialNr=0, funktionsTyp=0
I (1970-01-01 00:00:06) BSC_SERIAL: initSerial SerialNr=1, funktionsTyp=0
I (1970-01-01 00:00:06) CAN: loadCanSettings(): dataSrcAdd=0, u8_mBmsDatasource=0, bmsConnectFilter=0, u8_mBmsDatasourceAdd=0
I (1970-01-01 00:00:06) CAN: TWAI START: ESP_OK
I (1970-01-01 00:00:06) I2C: Display not found (2)
I (1970-01-01 00:00:06) I2C: Slave 0 (Adr=16) not found (2)
I (1970-01-01 00:00:06) I2C: Slave 1 (Adr=17) not found (2)
I (1970-01-01 00:00:06) I2C: Serial Ext. not found (2)
I (1970-01-01 00:00:06) MAIN: Free Heap: 146844
I (1970-01-01 00:00:00) LOG: Free Space total=836081, used=8032, logSize=1206
I (1970-01-01 00:00:00) MAIN: BSC V0.4.16
I (1970-01-01 00:00:00) MAIN: bootCounter=1
I (1970-01-01 00:00:00) MAIN: HW: 0
I (1970-01-01 00:00:00) MAIN: Free Heap: 205864
I (1970-01-01 00:00:00) WEB_SETTINGS: Free flash entries: 351
I (1970-01-01 00:00:03) MAIN: Free Heap: 200128
I (1970-01-01 00:00:03) MAIN: Init WLAN...
I (1970-01-) MAIN: Init BLE...
I (1970-01-01 00:00:03) MAIN: Wifi AP
I (1970-01-01 00:00:03) BSC_SERIAL: initSerial SerialNr=0, funktionsTyp=0
I (1970-01-01 00:00:03) BSC_SERIAL: initSerial SerialNr=1, funktionsTyp=0
I (1970-01-01 00:00:03) CAN: loadCanSettings(): dataSrcAdd=0, u8_mBmsDatasource=0, bmsConnectFilter=0, u8_mBmsDatasourceAdd=0
I (1970-01-01 00:00:03) CAN: TWAI START: ESP_OK
I (1970-01-01 00:00:03) I2C: Display not found (2)
I (1970-01-01 00:00:03) I2C: Slave 0 (Adr=16) not found (2)
I (1970-01-01 00:00:03) I2C: Slave 1 (Adr=17) not found (2)
I (1970-01-01 00:00:03) I2C: Serial Ext. not found (2)
I (1970-01-01 00:00:03) MAIN: Free Heap: 146920
I (1970-01-01 00:00:03) BSC_SERIAL: initSerial SerialNr=2, funktionsTyp=0
I (1970-01-01 00:00:03) BSC_SERIAL: initSerial SerialNr=3, funktionsTyp=0
I (1970-01-01 00:00:03) BSC_SERIAL: initSerial SerialNr=4, funktionsTyp=0
I (1970-01-01 00:00:03) BSC_SERIAL: initSerial SerialNr=5, funktionsTyp=0
I (1970-01-01 00:00:03) BSC_SERIAL: initSerial SerialNr=6, funktionsTyp=0
I (1970-01-01 00:00:03) BSC_SERIAL: initSerial SerialNr=7, funktionsTyp=0
I (1970-01-01 00:00:03) BSC_SERIAL: initSerial SerialNr=8, funktionsTyp=0
I (1970-01-01 00:00:03) BSC_SERIAL: initSerial SerialNr=9, funktionsTyp=0
I (1970-01-01 00:00:03) BSC_SERIAL: initSerial SerialNr=10, funktionsTyp=0
I (1970-01-01 00:00:03) MAIN: [WiFi] event: 0
I (1970-01-01 00:00:04) MAIN: [WiFi] event: 10
I (1970-01-01 00:00:04) MAIN: [WiFi] event: 11
I (1970-01-01 00:00:04) MAIN: [WiFi] event: 10
I (1970-01-01 00:00:04) MAIN: Init BLE...ok
I (1970-01-01 00:00:20) MAIN: [WiFi] event: 12
I (1970-01-01 00:00:20) MAIN: [WiFi] event: 14
I (1970-01-01 00:00:00) LOG: Free Space total=836081, used=10291, logSize=3264
I (1970-01-01 00:00:00) MAIN: BSC V0.4.16
I (1970-01-01 00:00:00) MAIN: bootCounter=1
I (1970-01-01 00:00:00) MAIN: HW: 0
I (1970-01-01 00:00:00) MAIN: Free Heap: 205632
I (1970-01-01 00:00:00) WEB_SETTINGS: Free flash entries: 264
I (1970-01-01 00:00:03) MAIN: Free Heap: 199236
I (1970-01-01 00:00:03) MAIN: Init WLAN...
I () MAIN: Init BLE...
I (1970-01-01 00:00:03) MAIN: Verbindung zu iot
I (1970-01-01 00:00:03) CAN: loadCanSettings(): dataSrcAdd=0, u8_mBmsDatasource=0, bmsConnectFilter=0, u8_mBmsDatasourceAdd=0
I (1970-01-01 00:00:03) CAN: TWAI START: ESP_OK
I (1970-01-01 00:00:03) BSC_SERIAL: initSerial SerialNr=0, funktionsTyp=0
I (1970-01-01 00:00:03) I2C: Display not found (2)
[... 385 lines cut ...]
I (2023-11-12 14:47:07) BLE_HANDLER: onDisconnect() dev=1, mac=3c:a5:51:85:e8:4c
I (2023-11-12 14:47:07) BLE_HANDLER: onDisconnect() dev=0, mac=00:e4:4c:21:d9:cd
I (2023-11-12 14:47:09) BLE_HANDLER: onConnect() 3c:a5:51:85:e8:4c
I (2023-11-12 14:47:24) BLE_HANDLER: onConnect() 00:e4:4c:21:d9:cd
I (2023-11-12 15:26:30) BLE_HANDLER: onDisconnect() dev=0, mac=00:e4:4c:21:d9:cd
I (2023-11-12 15:26:34) BLE_HANDLER: onConnect() 00:e4:4c:21:d9:cd
I (2023-11-12 15:51:13) BLE_HANDLER: onDisconnect() dev=0, mac=00:e4:4c:21:d9:cd
I (2023-11-12 15:51:15) BLE_HANDLER: onConnect() 00:e4:4c:21:d9:cd
I (2023-11-12 18:56:57) BLE_HANDLER: onDisconnect() dev=1, mac=3c:a5:51:85:e8:4c
I (2023-11-12 18:56:59) BLE_HANDLER: onConnect() 3c:a5:51:85:e8:4c
I (2023-11-12 19:13:44) BLE_HANDLER: onDisconnect() dev=0, mac=00:e4:4c:21:d9:cd
I (2023-11-12 19:13:46) BLE_HANDLER: onConnect() 00:e4:4c:21:d9:cd
I (2023-11-12 19:26:15) BLE_HANDLER: onDisconnect() dev=1, mac=3c:a5:51:85:e8:4c
I (2023-11-12 19:26:19) BLE_HANDLER: onConnect() 3c:a5:51:85:e8:4c
I (2023-11-12 19:41:16) BLE_HANDLER: onDisconnect() dev=0, mac=00:e4:4c:21:d9:cd
I (2023-11-12 19:41:20) BLE_HANDLER: onConnect() 00:e4:4c:21:d9:cd
I (2023-11-12 20:14:15) BLE_HANDLER: onDisconnect() dev=1, mac=3c:a5:51:85:e8:4c
I (2023-11-12 20:14:21) BLE_HANDLER: onConnect() 3c:a5:51:85:e8:4c
I (2023-11-12 20:18:47) BLE_HANDLER: onDisconnect() dev=0, mac=00:e4:4c:21:d9:cd
W (2023-11-12 20:18:50) BLE_HANDLER: Reconnect failed: dev=0
I (2023-11-12 20:18:52) BLE_HANDLER: onConnect() 00:e4:4c:21:d9:cd
I (2023-11-12 21:42:47) BLE_HANDLER: onDisconnect() dev=1, mac=3c:a5:51:85:e8:4c
I (2023-11-12 21:42:51) BLE_HANDLER: onConnect() 3c:a5:51:85:e8:4c
I (2023-11-12 22:37:26) BLE_HANDLER: onDisconnect() dev=1, mac=3c:a5:51:85:e8:4c
I (2023-11-12 22:37:26) BLE_HANDLER: onDisconnect() dev=0, mac=00:e4:4c:21:d9:cd

Elektrofix-OL, 2023-11-13:
Sorry konnte ich nur mit de Tablet kopieren.

shining-man, 2023-11-13:
Da System ist so langsam, da der BSC dauernt neu startet. 
Ich vermute du hast eine NodeMCU (o.ä.) mit einem schlechten Netzteil. Das ist ein typisches Problem vieler NodeMCU User.
Was verwendest du für Hardware?
Mein Tipp: Ordentliches Netzteil und ein zusätzlicher Elko und Kerko direkt am Stromversorgungspin des ESP32.

Elektrofix-OL, 2023-11-14:
Der ESP ist von [AZ-Delivery](https://www.amazon.de/AZDelivery-Development-unverlötet-inklusive-Nachfolger/dp/B08BZGC22Q?pd_rd_w=ccQ4z&content-id=amzn1.sym.6a985e9e-92ee-4610-844d-feb18aa20830&pf_rd_p=6a985e9e-92ee-4610-844d-feb18aa20830&pf_rd_r=2WCFTYX7CNKHZYVCF5QC&pd_rd_wg=CblSn&pd_rd_r=94011b04-9843-48d4-bb1d-43a3d67be8fd&pd_rd_i=B08BZGC22Q&ref_=pd_bap_d_grid_rp_0_1_ec_t&th=1) und der [DCDC-Wander](https://www.amazon.de/Converter-Stabilizer-Module-Galvanic-Isolated/dp/B0995QZQW9/ref=rvi_sccl_9/260-1165039-7068033?pd_rd_w=VaXFQ&content-id=amzn1.sym.13dbab83-f61c-4000-b9ab-184f02ce8fa2&pf_rd_p=13dbab83-f61c-4000-b9ab-184f02ce8fa2&pf_rd_r=7DFMNH8ENPN5W40BFHD6&pd_rd_wg=Dkyw4&pd_rd_r=d0d952a1-c553-42ee-b905-a28192ab2a70&pd_rd_i=B0995QZQW9&psc=1) ein Noname.

Elektrofix-OL, 2023-11-15:
Heute habe ich den RS485 Umsetzter bekommen und versucht die Seplos auf Serial 2 einzubinden. Ich habe zwei Packs.
- Ich habe es mit dem Netzteil aus 48V versucht
- Ich habe einen Stützkondensator an den 5 Volt von 100uF angelötet
- Ich habe den Konverter an 3,3V und 5V betrieben
- Ich habe auch ein Appel Netzteil 1A und 2,5A (womit auch ein Pi läuft) probiert
Die Einstellungen Serial 2 Seplos
- Auch Serial 3 auf Selplos Ein und ausgeschalte
- Automatische Adressconierung Ein und ausgeschaltet
- Anzahl der Pack von 1 auf 2 und wieder auf 1
- Die Dippschalter verstellt (#5 ein und aus)
- Bluetooth ein und ausgeschaltet
Nicht bring das BSC dazu meinen zweiten Pack anzuzeigen.

Hier das Log:
E (2023-11-15 19:02:59) BSC_SERIAL: Error: device=2, reason=Checksum wrong
E (2023-11-15 19:03:00) SEPLOS_BMS: Timeout: Serial=2, u8_lRecvDataLen=255, u8_lRecvBytesCnt=0
E (2023-11-15 19:03:00) BSC_SERIAL: Error: device=2, reason=Checksum wrong
E (2023-11-15 19:03:01) SEPLOS_BMS: Timeout: Serial=2, u8_lRecvDataLen=255, u8_lRecvBytesCnt=0
E (2023-11-15 19:03:01) BSC_SERIAL: Error: device=2, reason=Checksum wrong
W (2023-11-15 19:03:02) BLE_HANDLER: Failed to connect, deleted client; dev=1
E (2023-11-15 19:03:02) SEPLOS_BMS: Timeout: Serial=2, u8_lRecvDataLen=255, u8_lRecvBytesCnt=0
E (2023-11-15 19:03:02) BSC_SERIAL: Error: device=2, reason=Checksum wrong
E (2023-11-15 19:03:04) SEPLOS_BMS: Timeout: Serial=2, u8_lRecvDataLen=255, u8_lRecvBytesCnt=0
E (2023-11-15 19:03:04) BSC_SERIAL: Error: device=2, reason=Checksum wrong
I (2023-11-15 19:03:04) BLE_HANDLER: onConnect() 3c:a5:51:85:e8:4c
E (2023-11-15 19:03:05) SEPLOS_BMS: Timeout: Serial=2, u8_lRecvDataLen=255, u8_lRecvBytesCnt=0
E (2023-11-15 19:03:05) BSC_SERIAL: Error: device=2, reason=Checksum wrong
E (2023-11-15 19:03:06) SEPLOS_BMS: Timeout: Serial=2, u8_lRecvDataLen=255, u8_lRecvBytesCnt=0
E (2023-11-15 19:03:06) BSC_SERIAL: Error: device=2, reason=Checksum wrong
E (2023-11-15 19:03:07) SEPLOS_BMS: Timeout: Serial=2, u8_lRecvDataLen=255, u8_lRecvBytesCnt=0
E (2023-11-15 19:03:07) BSC_SERIAL: Error: device=2, reason=Checksum wrong
E (2023-11-15 19:03:09) SEPLOS_BMS: Timeout: Serial=2, u8_lRecvDataLen=255, u8_lRecvBytesCnt=0
E (2023-11-15 19:03:09) BSC_SERIAL: Error: device=2, reason=Checksum wrong
E (2023-11-15 19:03:10) SEPLOS_BMS: Timeout: Serial=2, u8_lRecvDataLen=255, u8_lRecvBytesCnt=0
E (2023-11-15 19:03:10) BSC_SERIAL: Error: device=2, reason=Checksum wrong
E (2023-11-15 19:03:11) SEPLOS_BMS: Timeout: Serial=2, u8_lRecvDataLen=255, u8_lRecvBytesCnt=0
E (2023-11-15 19:03:11) BSC_SERIAL: Error: device=2, reason=Checksum wrong
E (2023-11-15 19:03:12) SEPLOS_BMS: Timeout: Serial=2, u8_lRecvDataLen=255, u8_lRecvBytesCnt=0
E (2023-11-15 19:03:12) BSC_SERIAL: Error: device=2, reason=Checksum wrong
E (2023-11-15 19:03:14) SEPLOS_BMS: Timeout: Serial=2, u8_lRecvDataLen=255, u8_lRecvBytesCnt=0
E (2023-11-15 19:03:14) BSC_SERIAL: Error: device=2, reason=Checksum wrong
E (2023-11-15 19:03:15) SEPLOS_BMS: Timeout: Serial=2, u8_lRecvDataLen=255, u8_lRecvBytesCnt=0
E (2023-11-15 19:03:15) BSC_SERIAL: Error: device=2, reason=Checksum wrong
E (2023-11-15 19:03:16) SEPLOS_BMS: Timeout: Serial=2, u8_lRecvDataLen=255, u8_lRecvBytesCnt=0
E (2023-11-15 19:03:16) BSC_SERIAL: Error: device=2, reason=Checksum wrong
E (2023-11-15 19:03:17) SEPLOS_BMS: Timeout: Serial=2, u8_lRecvDataLen=255, u8_lRecvBytesCnt=0
E (2023-11-15 19:03:17) BSC_SERIAL: Error: device=2, reason=Checksum wrong
E (2023-11-15 19:03:19) SEPLOS_BMS: Timeout: Serial=2, u8_lRecvDataLen=255, u8_lRecvBytesCnt=0
E (2023-11-15 19:03:19) BSC_SERIAL: Error: device=2, reason=Checksum wrong
E (2023-11-15 19:03:20) SEPLOS_BMS: Timeout: Serial=2, u8_lRecvDataLen=255, u8_lRecvBytesCnt=0
E (2023-11-15 19:03:20) BSC_SERIAL: Error: device=2, reason=Checksum wrong
E (2023-11-15 19:03:21) SEPLOS_BMS: Timeout: Serial=2, u8_lRecvDataLen=255, u8_lRecvBytesCnt=0
E (2023-11-15 19:03:21) BSC_SERIAL: Error: device=2, reason=Checksum wrong
E (2023-11-15 19:03:22) SEPLOS_BMS: Timeout: Serial=2, u8_lRecvDataLen=255, u8_lRecvBytesCnt=0
E (2023-11-15 19:03:22) BSC_SERIAL: Error: device=2, reason=Checksum wrong
E (2023-11-15 19:03:24) SEPLOS_BMS: Timeout: Serial=2, u8_lRecvDataLen=255, u8_lRecvBytesCnt=0
E (2023-11-15 19:03:24) BSC_SERIAL: Error: device=2, reason=Checksum wrong
E (2023-11-15 19:03:25) SEPLOS_BMS: Timeout: Serial=2, u8_lRecvDataLen=255, u8_lRecvBytesCnt=0
E (2023-11-15 19:03:25) BSC_SERIAL: Error: device=2, reason=Checksum wrong
E (2023-11-15 19:03:26) SEPLOS_BMS: Timeout: Serial=2, u8_lRecvDataLen=255, u8_lRecvBytesCnt=0
E (2023-11-15 19:03:26) BSC_SERIAL: Error: device=2, reason=Checksum wrong
E (2023-11-15 19:03:27) SEPLOS_BMS: Timeout: Serial=2, u8_lRecvDataLen=255, u8_lRecvBytesCnt=0
E (2023-11-15 19:03:27) BSC_SERIAL: Error: device=2, reason=Checksum wrong
E (2023-11-15 19:03:29) SEPLOS_BMS: Timeout: Serial=2, u8_lRecvDataLen=255, u8_lRecvBytesCnt=0
E (2023-11-15 19:03:29) BSC_SERIAL: Error: device=2, reason=Checksum wrong
E (2023-11-15 19:03:30) SEPLOS_BMS: Timeout: Serial=2, u8_lRecvDataLen=255, u8_lRecvBytesCnt=0
E (2023-11-15 19:03:30) BSC_SERIAL: Error: device=2, reason=Checksum wrong
I (2023-11-15 19:03:30) MQTT: MQTT Broker connected
E (2023-11-15 19:03:31) SEPLOS_BMS: Timeout: Serial=2, u8_lRecvDataLen=255, u8_lRecvBytesCnt=0
E (2023-11-15 19:03:31) BSC_SERIAL: Error: device=2, reason=Checksum wrong
[... 1291 lines cut ...]
E (1970-01-01 01:00:18) BSC_SERIAL: Error: device=2, reason=Checksum wrong
[... cut ...]
E (1970-01-01 01:00:30) SEPLOS_BMS: Timeout: Serial=2, u8_lRecvDataLen=255, u8_lRecvBytesCnt=0
E (1970-01-01 01:00:30) BSC_SERIAL: Error: device=2, reason=Checksum wrong
E (1970-01-01 01:00:31) SEPLOS_BMS: Timeout: Serial=2, u8_lRecvDataLen=255, u8_lRecvBytesCnt=0
E (1970-01-01 01:00:31) BSC_SERIAL: Error: device=2, reason=Checksum wrong
E (1970-01-01 01:00:33) SEPLOS_BMS: Timeout: Serial=2, u8_lRecvDataLen=255, u8_lRecvBytesCnt=0
E (1970-01-01 01:00:33) BSC_SERIAL: Error: device=2, reason=Checksum wrong
I (1970-01-01 01:00:34) BLE_HANDLER: onConnect() 00:e4:4c:21:d9:cd

shining-man, 2023-11-16:
Hast du dazu den wiki Eintrag gelesen? 
https://github.com/shining-man/bsc_fw/wiki/Use-Cases#seplos-bms-anschlussm%C3%B6glichkeiten

`Nicht bring das BSC dazu meinen zweiten Pack anzuzeigen.` -> D.h. des ersten Pack zeigt er an? Die Kommunikation funktioniert also prinzipiell?

Die Logs bitte als Datei anhängen, da sonst die Post unendlich lang werden.

Elektrofix-OL, 2023-11-16:
Den Wiki Eintrag habe ich gelesen, damit funktioniert es auch. Aber dann geht der SunnyIsland auf Störung, weil das Pack0 nicht mehr da ist, sondern nur Pack1 und Pack2. Das BSC benutz Adressen, die nach der Seplos-Lehre nicht gültig sind, denn das Pack0 ist zwingend erforderlich für die Kommunikation mit dem SI.
Ich habe auch versucht das Pack1 wieder zu Pack0 mit CAN umzustellen, dann friert das BSC ein 
[Log.txt](https://github.com/shining-man/bsc_fw/files/13379782/Log.txt)
Wenn das BSC im Multipackbetrieb nur ab Pack1 und nicht ab Pack0 funktioniert, dann kann ich es nur für das Auslesen der NEEY's benutzen, das Seplos bleibt mir dann verwehrt

Elektrofix-OL, 2023-11-17:
Das war gerade das, was ich nicht wollte. Das BSC sollte nur Überwachen nicht steuern.
Halten wir fest:
- Eine NodeMCU muss mit einem Kondensator gestützt werden (oder ein sehr gutes Netzteil), sonst kommt es zu Abstürzen
- Neey's lassen sich über ein NodeMCU per BT problemlos auslesen
- das Auslesen von einem Seplos-Pack ist möglich
- das Auslesen mehrere Seplos-Packs geht nicht, es sei denn das BSC übernimmt die Steuerung und regelt auch den CAN
- das Auslesen mehrere Seplos-Packs geht nur wie im [Wiki](https://github.com/shining-man/bsc_fw/wiki/Use-Cases#seplos-bms-anschlussm%C3%B6glichkeiten) beschrieben, andere Adresseingabe an dem Seplos-Packs lässt das NodeMCU abstürzen.

Das Projekt BSC ist fantastisch und Möglichkeiten sind vielfältig. Mach weiter so!

closed 2023-11-17 by Elektrofix-OL