shining-man/bsc_fw #55: Zeitserver wird scheinbar nur bei Bootup abgerufen

tueftler0, 2023-11-18:
Ich habe gerade gesehen, dass der Zeitserver scheinbar nur im Bootup abgerufen wird.
- Dann könnte z.B. die Zeit weg laufen
- Weiterhin bleibt es auf 1970 stehen, wenn der interne Zeitserver gerade nicht erreichbar war

Ich denke das sollte man öfters triggern.

pepe-el-pepe, 2023-12-10:
Der esp32 hat scheinbar eine interne RTC ((https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/system_time.html)) die über resets hinweg die Zeit behält. Den zu nutzen wäre doch eine Idee damit die Zeit nicht erst nach dem booten und NTP Server finden korrekt ist. Eine manuelle Eingabe von Zeit und Datum wäre für alle ohne INET oder NTP Server Möglichkeit noch eine feine Option.

tueftler0, 2024-01-14:
Mit der v0.5.4 wird der NTP nach einer Stunde abgerufen.
Somit geschlossen

closed 2024-01-14 by tueftler0