shining-man/bsc_fw #56: Log delete erzeugt kein neues

tueftler0, 2023-11-19:
Es scheint so, als ob nach einem Entfernen des Logs kein neues erzeugt wird und neu ankommende Infos dadurch unsichtbar bleiben.
Erst ein Reboot behebt dies.

do13, 2023-11-19:
Sicher? Wie lange hast du gewartet?

tueftler0, 2023-11-19:
Ich versuche seid 12 Stunden den Seplos-Fehler zu loggen.
In diesen 12h gab es nicht einen einzigen Eintrag im Log.
Laut Discord ist das bekannt. Ich kann aber auch nicht 100% sagen ob es wirklich so ist!
Gegen checken kann ich es grade nicht.

nsc2001, 2023-11-20:
Kann ich bestätigen, ist bei mir auch so.

do13, 2023-11-20:
@tueftler0 Du hast Recht, die folgende Funktion kann tatsächlich zu solchen Fällen führen:

`void btnSystemDeleteLog()
{
  SPIFFS.remove("/log.txt");
  SPIFFS.remove("/log1.txt");
  if(SPIFFS.exists("/log.txt")) //Wenn Log-Datei immer noch vorhanden
  {
    SPIFFS.format();
    delay(100);
    webSettingsSystem.writeConfig();
  }
  BSC_LOGI(TAG, "Logfiles deleted");
}
`
Theoretisch kann durch einen Spannungsausfall auch die Konfiguration verloren gehen ....
Ich war mir eigentlich ziemlich sicher das logfile auch schon gelöscht zu haben und es wurde neu angelegt. Entweder hat sich in der Zwischenzeit etwas geändert oder es war Zufall. Aber ja, das ist tatsächlich noch ein Problem hier.

shining-man, 2023-11-20:
Das Logfile wird aktuell nicht neu angelegt. Das war schon immer so und war ursprünglich bewusst so, da das Löschen nur zum Testen war.
Ich werde es anpassen und ein neues File anlegen.

tueftler0, 2023-12-20:
Super, funktioniert nun auch. rev0.4.19

closed 2023-12-20 by tueftler0