shining-man/bsc_fw #66: Second include of `LITTLEFS.h` in main.cpp, may result in build error

meikjaeckle, 2024-01-08:
[`LITTLEFS.h`](https://github.com/shining-man/bsc_fw/blob/dc3831b71255e6299909bf8ec7ddc8e67ae14239/src/main.cpp#L38C1-L38C1) is already included in [line 34](https://github.com/shining-man/bsc_fw/blob/dc3831b71255e6299909bf8ec7ddc8e67ae14239/src/main.cpp#L34C7-L34C7)

Results in compile error if LITTLEFS lib is not available.

shining-man, 2024-01-09:
You must replace LITTLEFS.h with LittleFS.h

meikjaeckle, 2024-01-09:
Thanks, that's a good catch, I'm building on Linux.
What I wanted to mention is that the header is once included inside the `#ifdef USE_LittleFS` and once just afterwards. I think this is just a left over from the last change. Not a big issue.

```
#ifdef USE_LittleFS
  #define SPIFFS LittleFS
  #include <LITTLEFS.h> 
#else
  #include <SPIFFS.h>
#endif
#include <LITTLEFS.h>
```

meikjaeckle, 2024-01-11:
Fixed with version 0.5.3

closed 2024-01-11 by shining-man