shining-man/bsc_fw #77: BSC Boottime wird vor NTP fest gesetzt

tueftler0, 2024-02-04:
Nach Reboot beim Update wird die Boottime im Dashboard scheinbar gesetzt, ohne auf eine NTP-Zeit zu warten.
Evtl. braucht es hier noch einen Timeout.

[image]

KarlMeier1933, 2024-02-08:
So ist es bei mir auch!

Ingo16KausA, 2024-02-12:
Hallo
das Problem habe ich schon bei der  V5.4.0
Boottime 1970-01-01

tueftler0, 2024-03-02:
Hat sich hier etwas geändert?
Bekomme ich gerade nicht mehr reproduziert.

shining-man, 2024-03-02:
Nein, da hat sich nichts geändert.

KarlMeier1933, 2024-03-02:
Ich konnte es beheben indem ich eine IP-Adresse, anstatt einer Domain-Adresse eingegeben hab. Sämtliche ntp-Adressen haben nicht funktioniert. Mit der IP-Adresse der Fritzbox hats dann geklappt.

tueftler0, 2024-03-14:
> Ich konnte es beheben indem ich eine IP-Adresse, anstatt einer Domain-Adresse eingegeben hab...
Ich habe bisher nur die direkte IP der Fritzbox genutzt. Damit klappte es nicht immer.