shining-man/bsc_fw #81: Seplos BVMS V3.0 - Keine Kommunikation

DarthPride, 2024-02-11:
Hallo zusammen,

auch wenn es imn Discord schon steht, hier noch mal der Übersicht halber:

Das BSC bekommt keine Verbindung zu einem Seplos BMS V3.0
Die Hardware wurde kontrolliert, sie funktioniert.

Vermutlich liegt es an einem geänderten Protokoll, da das Seplos BMS 2.0 auch nicht mit dem 3.0 kommunizieren kann.

Ich habe den Vertriebler, der mir das 3.0 verkauft hat schon angeschrieben ob er eine Beschreibung der Schnittstelle hat, aber in China is Neujahr. Die Antwort wird also noch dauern.

shining-man, 2024-02-11:
[Seplos-48v-100a-bms-rs485-protocol.pdf](https://github.com/shining-man/bsc_fw/files/14231481/Seplos-48v-100a-bms-rs485-protocol.pdf)

shining-man, 2024-02-11:
[SEPLOS-3.0RS485BMS Modbus RTU Protocol-1.pdf](https://github.com/shining-man/bsc_fw/files/14231583/SEPLOS-3.0RS485BMS.Modbus.RTU.Protocol-1.pdf)

tueftler0, 2024-03-02:
@DarthPride Es gibt eine neue Betaversion im Discord. Wenn Du diese testen könntest wäre toll.
Damit die Integrationsarbeit nicht umsonst war, Danke.

DarthPride, 2024-03-03:
Doofe Frage: muss ich weiterhin einen TTL auf RS485 Konverter dazwischen hängen?

shining-man, 2024-03-03:
Ja, den brauchst du weiterhin.

DarthPride, 2024-03-03:
Dann kommt nichts an BSC an. Wobei ich mir ehrlich gesagt nicht sicher bin, ob meine Hardware nicht einen Weg hat.
Ich muss mir mal einen neuen ESP32 und einen neuen Konverter besorgen, dass kann aber etwas dauern.

shining-man, 2024-05-26:
Das Seplos V3 ist in der Version v0.6.0 enthalten.

closed 2024-05-26 by shining-man