shining-man/bsc_fw #86: Wunsch: SoC Überwachung auch direkt von BMS auslesen

lugiiii, 2024-02-16:
Ich habe heute meine Schaltwerte für den SoC eingegeben. Leider ohne funktion. Ich denke das die SoC-Werte nur direkt von einen MPPT gelesen werden? Es wäre schön, wenn diese auch direkt vom BMS gelesen werden könnten. Also ich habe nur ein JK-BMS und wollte anhand von diesen Werten den Trigger auslösen lassen. Ist das möglich bzw. möglich diese noch in ein Softwareupdate einzubringen?
Danke schon im Voraus!

shining-man, 2024-02-17:
Welche MPPT's meinst du? Es stehen doch gar keine MPPT's zur Verfügung.
Wenn dein BMS das einen SoC liefert, dann steht der auch im System zur Verfügung. Das Problem könnte an der Stelle sein, dass die momentane Implementierung den SoC nutzt, der zum Wechselrichter gesandt wird. Wenn du keinen Wechselrichter angeschlossen hast, dann funktioniert auch die SoC Regelung nicht. Deshalb ist sie aktuell auch im Inverter Menü.
Hast du einen Wechselrichter angeschlossen?

lugiiii, 2024-02-17:
Danke für die schnelle Antwort. Wenn ich in das Menü "Inverter" gehe, sehe ich viele Dinge über Ladestrom usw., daher meine Vermutung, dass du hier den MPPT ansprechen willst. Wenn du natürlich Energie vom Akku zum Wechselrichter meinst, würde ich das Entladstrom nennen. Ist halt immer eine Sache des Blickwinkels, von wo man drauf schaut. (ich schaue auf den Akku) Aber vielleicht habe ich da ja auch was falsch verstanden. Schreib doch einfach statt Inverter --> Wechselrichter, für mich wäre es dann eindeutiger. Ich habe als Wechselrichter einen Growatt der eigentlich für ein Balkonkraftwerk gedacht war und nur am Akku hängt, also keine Kommunikation mit dem BSC oder BMS macht, nur mit dem SDM630.  (Ich bin mir nicht sicher, ob sich das auch über den Growatt realisieren liese?)
Für mich wäre es am einfachsten, genau die gleich Menüpunkte wie unter Inverter, sprich "Trigger bei SoC" für das BMS zu haben und hier dann der Wert (SoC) vom BMS gelesen werden würde.
Wäre schön, wenn du das einpflegen könntest!

shining-man, 2024-02-17:
Wechselrichter (deutsch) == Inverter (englisch)
Es geht hierbei um den Ladestrom vom Wechselrichter in die Batterie. Der Wechselrichter muss dazu über CAN mit dem BSC verbunden sein.
Ich notiere es mir einmal.

lugiiii, 2024-02-17:
Genau, mir ist der Usecase hierzu nicht bekannt, da es bei meiner Konfiguration so ja nicht funktionieren würde. Darum war ich wohl auch etwas verwirrt.
Ich finde es immer noch super, dass du das Projekt durchgezogen hast und würde mich sehr freuen, wenn du das einpflegst könntest. Danke!

tueftler0, 2024-02-18:
Ich finde die Namensgebung ganz okay.
Bei mir läuft es mit einem Growatt Inverter (inkl MPPTs intern) zuverlässig.