shining-man/bsc_fw #94: Stromanzeige falsch, Neey geht aus, Serial Port 3 absturz ESP32

siggi07, 2024-02-26:
1.) Der Ist Strom wir  doppelt  so groß angezeigt, 1 Pack angeschlossen  "Seplos BMS V16.6"
2.) Neey Balance werden  Daten gesendet wenn AUS abgespeichert ist und man im Serialmenue speichern wählt
3.) Wenn man Serial 3 auswählt bei BMS hängt sich das System auf und man kommt nicht mehr dran, so das man den ESP32 komplett löschen muss.
 
Ich hoffe das Hilft , 

Gruss Siggi

siggi07, 2024-02-26:
Zu 1.) Der Strom wird im BSC korrekt angezeigt im Victronportal doppelt so groß

siggi07, 2024-02-26:
Zu 2.) Dann ist mir auch noch aufgefallen das aus irgend welchen Gründen die Werte 2.6 Volt für ein und 2.5Volt für aus eingetragen werden und ich weiß nicht wo die her kommen.

shining-man, 2024-02-27:
zu 1) Ich vermute einen Konfigurationsfehler in dem das BMS doppel ausgewählt wurde. Und zwar als Masterdatenquelle und in den zusätzlich Datenquellen.
[image]
Das ist falsch.

shining-man, 2024-02-27:
zu 2) Mit der Beschreibung kann ich nicht so vie Anfangen. Was heißt wen "AUS" gespeichert ist?

shining-man, 2024-02-27:
zu 3) Hast du ein serial Extension. Serial 3 hat nur eine Funktion, wenn auch eine serial Extension vorhanden ist. Dennoch darf es sich nicht "aufhängen". Das werde ich mir anschauen.

siggi07, 2024-02-27:
Konfigurationsfehler gucke ich heute Abend nach der Arbeit. Wirst wohl Recht haben. Serial 3 habe ich nicht ist mir nur passiert weil ich fälschlicherweise 3 ausgewählt habe.