shining-man/bsc_fw #99: Names of Neey settings

pepe-el-pepe, 2024-03-13:
hello 👋  

found something today. Shouldn't it be **buzzer** instead of **balancer**?

**Start Voltage** also sounds a bit weird. In the original Neey app it is called **Startup DifVolt**. In my humble opinion, that makes more sense or am I misunderstanding something here? 

[image]

[image]

shining-man, 2024-03-13:
Nein, Balancer On ist richtig. Das hat nichts mit dem Buzzer zu tun. Die Buzzer Option ist im BSC nicht abgebildet.

shining-man, 2024-03-13:
"Start Voltage" ist vielleicht nicht ganz glücklich gewählt. Da hast du recht.

pepe-el-pepe, 2024-03-13:
Danke für die Richtigstellung. Das heißt, das Balancer-on/off-Setting ist gleichzusetzen mit dem Schalter „Equalizing“ von der Neey-App „Hauptseite“?
Nach meinem Empfinden ist die Position ein wenig verwirrend. Die Reihenfolge aller Einstellungen von der Neey-App ist im BSC genauso zu finden, außer die des Buzzers.
Wäre es eine Option, die Einstellung ganz oben zu platzieren, vielleicht mit ein wenig Abstand zu den anderen Werten. Dann wäre ich niemals auf die Idee mit dem Buzzer gekommen.

closed 2024-05-07 by pepe-el-pepe