shining-man/bsc_fw #100: Zusätzlicher Startwert bei "Entladestrom Zellspannungsabhängig drosseln"

KarlMeier1933, 2024-03-19:
Eine absolut super Funktion! Richtig genial fände ich es noch, wenn nicht der allgemeine Maximal-Entladestrom der Startwert für die Reduzierung wäre, sondern wenn man zusätzlich noch einen Entladestrom-Wert eingeben könnte, welcher gesetzt wird, wenn die Drosselung beginnt und von diesem Wert wird dann bis zum Minimal-Entladestrom reduziert.

pepe-el-pepe, 2024-04-17:
Hallo KarlMeier1933,
mich würde es interessieren warum du diese Funktion brauchen könntest. Bitte erhelle mich :)
Ich freu mich immer wenn der Akku bis mindest Entladegrenze oder min. SoC entladen wird ohne das der Strom gedrosselt wird

KarlMeier1933, 2024-04-17:
Der Hintergrund ist der, dass ich viel feiner regeln kann, das hilft mir das Ganze noch weiter zu optimieren und noch mehr herauszuholen, ohne meine Sicherheitseinstellungen verändern zu müssen. Ich möchte nämlich dass Vor-Alarme/Warnungen und Alarme tatsächlich nur dann auslösen, wenn sich die Zellen oder der ganze Pack bei einer Fehlfunktion in diesem Bereich bewegen.
Mit der Drosselung kann ich große Lasten schonmal sehr gut einbremsen, damit mir diese nicht durch nen hohen Entladestrom die Zellen in einen Alarmbereich ziehen. Sobald die Last aber weg ist, steigt die Spannung wieder an und die Drosselung setzt den maximalen Entladestrom wieder entsprechend der Packspannung nach oben, bis zur nächsten Lastspitze. Gerade bei nem Backofen ist das ja ein ständiges hin und her. Und um das zu verhindern würde so ein Startwert für die Drosselung Sinn machen.

Also zum Beispiel:
Entladestrom normal - 200 A
Ab einem eingestellten Spannungswert zB 50V springt der Entladestrom auf 50A und wird von nun an bis 46,5V auf 0 reduziert.