shining-man/bsc_fw #101: Set minimum charging current for SOC dependent charge current limiting

tanelvakker, 2024-03-31:
Currently it allows the charge current to go down to 0. 
[image]
Would it make sense to define minimum charging current as well? I would still allow at least 1A current so the balancer can do it's work, but keep the SOC up.

shining-man, 2024-04-01:
It's a good idea.
I'll add the setting for a minimum charging current.

tueftler0, 2024-05-07:
@tanelvakker Integrated now. Please test the newest beta. Thank you

tueftler0, 2024-05-26:
@shining-man No reply since three weeks from TO. Maybe the issue can be closed now -> Released

tanelvakker, 2024-05-26:
It works fine. Can be closed

tueftler0, 2024-05-26:
Nice to hear that! You can close the issue yourself by clicking the button at the bottom.

closed 2024-05-26 by tanelvakker