shining-man/bsc_fw #115: Pylontech BMS Unterstützung ?

adcrafter27, 2024-05-22:
Hallo, Ich wollte nachfragen ob dieses System auch Pylontech Batterien unterstützt.
Ich habe ein etwas speziellen mix zwar habe ich 6x Pylontech und 3 selbstgebaute 15S mit JK BMS
Daher währe schön wenn alle BMS über dieses System laufen würden 
Das hängt momentan alles direkt am Cerbo GX
Ich würde aber gerne ein wenig die Pylontech von der Lade Leistung beschränken über dieses System.

adcrafter27, 2024-05-30:
Hallo, Danke das du dir das vlt man ansiehst. 
Dokumente habe ich so auf die schnelle nicht gefunden, aber ich habe das hier im Einsatz
https://github.com/irekzielinski/Pylontech-Battery-Monitoring
Damit kann man die Pylontech auch auslesen. Vielleicht hilft das was weiter

shining-man, 2024-05-30:
Wenn braucht es schon ein Protocol. Alles andere ist mir zu Fehlerträchtig und Aufwendig.
Finden tut man schon was auf die schnelle, aber ob es das richtig ist, weiß ich nicht. Das müsstest du sagen/überprüfen. 
https://github.com/Frankkkkk/python-pylontech/blob/master/RS485-protocol-pylon-low-voltage-V3.3-20180821.pdf

shining-man, 2024-06-02:
@adcrafter27 
Ich denke das Protocol könnte schon passen, wenn du dich über den RS485 Port mit dem Pylontech verbindest, was ja auch ok wäre. Als Alternative gibt es noch irgend einen Consolenport, aber ich denke, dass das nicht das richtige ist.
Wäre das das was du willst?

adcrafter27, 2024-06-02:
Hallo, Danke das du dich damit beschäftigst.
Ich habe von den ganzen Protokollen leider nicht viel Ahnung.
Ich würde gerne die werte von JK Bms und Pylontech Bms über dein System zusammenfassen und eben gegeben falls von mir angepasste werte  zbs. maximalen Ladestrom  usw. an das Victron System übergeben. 
Ich hoffe auch das es bald eine "fertige" Hardware zu kaufen gibt. 
Momentan habe ich dein System noch nicht getestet da mit die Hardware und der Pylontech BMS support fehlt.

Der B/RS485 Port ist bei meinen Pylontech noch frei angeschlossen ist der Cerbo über A/CAN
Wenn ich dann die Pylontech Batterien an dein System anschließen würde, muss ich jede Batterie einzeln über B/RS485 anschließen oder verwendet er die interne Kommunikation der Batterien

shining-man, 2024-06-02:
Wie sind denn die einzelnen Pylontechs untereinander verbunden?
Kannst du mir da irgendwie eine Übersicht/Zeichnung machen?

adcrafter27, 2024-06-02:
Die Pylontech besitzen einen Link port 0 und 1, es wird einfach ein normales Netzwerk Kabel von port 1 zu der nächsten Batterie Port 0 gesteckt und dann wieder von 1 zu 0 usw

Der cerbo hängt nur an der "Master" Batterie an A/Can

Ich habe US2000C und us3000C gemischt
Habe aber auch noch Systeme nur mit US5000 die ich verwalte

Hier nochmal die Bedienungsanleitung von Pylontech
https://en.pylontech.com.cn/download.aspx?id=326

Seite 8 sind die Anschlüsse zu sehen und Seite 19 wie sie untereinander verbunden werden

shining-man, 2024-06-03:
Es könnte reichen, wenn du vom Master den Port A/CAN, mit dem du jetzt zum cerbo gehst, dann zum BSC gehst. Dafür darfst du dann aber nur Pin 7 und 8 nehmen. 
Wenn das nicht ausrreicht um an alle Daten zu kommen, dann musst du, wie auf Seite 26 für den Multigroup-Mode beschrieben, alle Pylontech miteinander verbinden.
[image]

ich habe noch gelesen, dass bei einer Mischung von 2000 und 3000, der Master ein 2000 sein muss, damit es funktioniert.
Ich denke das alles läuft auf einen Versuch hinaus. Ich kann es implementieren, dann kannst du es testen. Wie und ob es dann am Schluss funktioniert kann ich aber nicht versprechen.
Sag einfach bescheid ob du bereit bist es zu probieren, auch wenn es am Schluss nichts wird?

adcrafter27, 2024-06-28:
Hallo, sorry das ich mich erst jetzt melde.
Ok das mit der Belegung sollte kein Problem sein einfach ein Kabel zerschneiden.
ja das mit der Mischung von 2000 und 3000 hatte ich auch gelesen kann man ja testen ob es anders auch klappt dann weiß man das auch.
Ein versuch ist es wert, wenn's nicht klapp ist es auch kein Weltuntergang.
Was alles brauche ich an Hardware um das zu testen, ein "original" BSC Modul habe ich leider nicht bekommen.

shining-man, 2024-06-30:
Bis der orginale BSC wieder verfügbar ist, dauert es noch ein bisschen.
Entweder warten oder du kannst zum Testen solange ein LILYGO T-CAN485 nehmen. Das hat ein RS485 und CAN an Board. Aber aufpassen, es ist nicht galvanisch getrennt!

Wenn du Testbereit bis, sag bescheid, dann bekommst du einen Teststand.

adcrafter27, 2024-07-07:
Danke, ich werde noch warten bis das Originale BSC wieder verfügbar ist, dann kann ich es ordentlich machen. Auch wenn das mit den pylontech nicht klappen sollte werde ich das BSC trotzdem verwenden.

BennoB666, 2025-03-07:
ANbei der Link zum aktuellsten Protokoll was ich finden konnte. Version ist mittlerweile bei 3.x
[https://github.com/Interster/PylonTechBattery/blob/master/PYLON%20LFP%20Battery%20communication%20protocol%20-%20RS232%20V2.8%2020161216.pdf](url)

Fuendiger, 2025-03-12:
Dein Link selber funktioniert nicht. Wenn man den Linktext kopiert, dann kommt man du einem alten Protokoll.
Ich denke das aktuellste ist das von mir [hier](https://github.com/shining-man/bsc_fw/issues/115#issuecomment-2140110692) verlinkte.

tueftler0, 2025-09-05:
Für Pylontech hat die SingleDeviceExtension ein RS232-Anschluss mit drauf.
Damit kann man dann mit dem Console-Port reden und alle angeschlossenen Packs auslesen.
Ob das derzeit im BSC so unterstützt wird weiß ich jedoch gerade nicht.

shining-man, 2025-09-06:
@jensbehrens Ich schaue es mir noch einmal an. Es ist schon eine weile her. Da sich damals keiner mehr gemeldet hat, habe ich es nicht mehr weiter verfolgt.

jensbehrens, 2025-09-06:
Danke dir erstmal! 👍 
Ich habe eine US2000C und zwei US5000.
Habe am RS485B und CANA probiert, sowohl an 2000er, als auch 5000er. An den Link Ports auch… Auch einzeln hat sich nix getan. Immer checksum Fehler…
Nutze das Lilygo T-Connect.

Habe mir die Insider Firmware auch mal gecloned /gebaut und gesehen (mit meinen ehrlich gesagt etwas eingerosteten Programmierkenntnissen), dass scheinbar die Seplos readBMSData Funktionen genutzt werden. Ist das so korrekt?

jensbehrens, 2025-09-06:
Habe mir von hier mal den tx geholt: https://github.com/martiby/ess/blob/master/pylontech.py

Aber tut iwie auch nicht:

IMP I (2025-09-06 14:30:00) SEPLOS_BMS: sendBytes: 0x20 0x2 0x46 0x42 0xe0 0x2 0x2 0xfd 0x33 
IMP I (2025-09-06 14:30:00) SEPLOS_BMS: Timeout: RecvBytes=0:

shining-man, 2025-09-06:
> Habe mir die Insider Firmware auch mal gecloned /gebaut und gesehen (mit meinen ehrlich gesagt etwas eingerosteten Programmierkenntnissen), dass scheinbar die Seplos readBMSData Funktionen genutzt werden. Ist das so korrekt?

So weit ich mich erinnere ist das richtig. Nur die Baudrate war, gleaube ich, anders. Ich schau mir das noch einmal an.

jensbehrens, 2025-09-07:
Ja, Zellspannungen kommen und stimmen.

Der FET-State kann aber noch nicht korrekt gelesen werden...(entladen/laden ist erlaubt, laut CAN Verbindung passen CCL,DCL):

[image]

shining-man, 2025-09-10:
@jensbehrens Da fehlen die Debugausgaben (D). Nur warum? Wenn `SEPLOS_DEBUG` gesetzt ist, dann setzt er das Loglevel auf Debug. Komsich.
Zur Not in der SeplosBms.cpp die `BSC_LOGD` zu `BSC_LOGI` machen.

Das mit dem SoC musste zumindest nach der Spec. falsch sein ;-)

jensbehrens, 2025-09-16:
Hab's die BSC_LOGDs in BSC_LOGI umbenannt:
```
parseMessage: serialDev=1
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: Number of cells: 15
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: Number of temperature sensors: 5
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: SoC Abweichung > 5; SoC_alt=52, SoC_neu=51
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: Dev=1, RecvBytes=126
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: 0x32 0x30 0x30 0x32 0x34 0x36 0x30 0x30 0x43 0x30 0x36 0x45 0x30 0x30 0x30 0x32 0x30 0x46 0x30 0x43 
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: 0x44 0x38 0x30 0x43 0x44 0x41 0x30 0x43 0x44 0x41 0x30 0x43 0x44 0x41 0x30 0x43 0x44 0x37 0x30 0x43 
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: 0x44 0x39 0x30 0x43 0x44 0x39 0x30 0x43 0x44 0x37 0x30 0x43 0x44 0x38 0x30 0x43 0x44 0x39 0x30 0x43 
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: 0x44 0x39 0x30 0x43 0x44 0x38 0x30 0x43 0x44 0x39 0x30 0x43 0x44 0x42 0x30 0x43 0x44 0x43 0x30 0x35 
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: 0x30 0x42 0x43 0x39 0x30 0x42 0x42 0x34 0x30 0x42 0x42 0x30 0x30 0x42 0x41 0x45 0x30 0x42 0x42 0x41 
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: 0x46 0x46 0x46 0x38 0x43 0x30 0x42 0x38 0x32 0x41 0x41 0x31 0x30 0x32 0x43 0x33 0x35 0x30 0x30 0x32 
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: 0x30 0x35 0x45 0x34 0x32 0x43 
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: crc=64817
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: sendBytes: 0x20 0x2 0x46 0x44 0xe0 0x2 0x2 0xfd 0x31 
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: RecvBytes=82
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: 0x32 0x30 0x30 0x32 0x34 0x36 0x30 0x30 0x41 0x30 0x34 0x32 0x30 0x30 0x30 0x32 0x30 0x46 0x30 0x30 
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x35 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x45 0x34 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x46 0x31 
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: 0x30 0x35 
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: parseMessage: serialDev=1
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: SeplosBms_readBmsData() devNr=1, readFromAdr=3, BmsDataAdr=2, CountOfPacks=0
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: crc=64817
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: sendBytes: 0x20 0x3 0x46 0x42 0xe0 0x2 0x3 0xfd 0x31 
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: RecvBytes=138
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: 0x32 0x30 0x30 0x33 0x34 0x36 0x30 0x30 0x46 0x30 0x37 0x41 0x30 0x30 0x30 0x33 0x30 0x46 0x30 0x43 
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: 0x44 0x38 0x30 0x43 0x44 0x42 0x30 0x43 0x44 0x39 0x30 0x43 0x44 0x41 0x30 0x43 0x44 0x41 0x30 0x43 
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: 0x44 0x39 0x30 0x43 0x44 0x41 0x30 0x43 0x44 0x38 0x30 0x43 0x44 0x39 0x30 0x43 0x44 0x39 0x30 0x43 
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: 0x44 0x39 0x30 0x43 0x44 0x39 0x30 0x43 0x44 0x39 0x30 0x43 0x44 0x42 0x30 0x43 0x44 0x42 0x30 0x35 
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: 0x30 0x42 0x42 0x42 0x30 0x42 0x41 0x46 0x30 0x42 0x41 0x41 0x30 0x42 0x41 0x43 0x30 0x42 0x42 0x34 
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: 0x46 0x46 0x46 0x32 0x43 0x30 0x42 0x45 0x46 0x46 0x46 0x46 0x30 0x34 0x46 0x46 0x46 0x46 0x30 0x31 
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: 0x45 0x33 0x30 0x30 0x35 0x43 0x46 0x38 0x30 0x31 0x38 0x36 0x41 0x30 0x45 0x30 0x45 0x32 
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: parseMessage: serialDev=2
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: Number of cells: 15
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: Number of temperature sensors: 5
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: crc=64815
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: sendBytes: 0x20 0x3 0x46 0x44 0xe0 0x2 0x3 0xfd 0x2f 
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: RecvBytes=82
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: 0x32 0x30 0x30 0x33 0x34 0x36 0x30 0x30 0x41 0x30 0x34 0x32 0x30 0x30 0x30 0x33 0x30 0x46 0x30 0x30 
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x35 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x45 0x34 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x46 0x31 
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: 0x30 0x33 
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: parseMessage: serialDev=2
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: SeplosBms_readBmsData() devNr=1, readFromAdr=4, BmsDataAdr=3, CountOfPacks=0
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: crc=64815
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: sendBytes: 0x20 0x4 0x46 0x42 0xe0 0x2 0x4 0xfd 0x2f 
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: RecvBytes=138
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: 0x32 0x30 0x30 0x34 0x34 0x36 0x30 0x30 0x46 0x30 0x37 0x41 0x30 0x30 0x30 0x34 0x30 0x46 0x30 0x43 
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: 0x44 0x41 0x30 0x43 0x44 0x43 0x30 0x43 0x44 0x43 0x30 0x43 0x44 0x42 0x30 0x43 0x44 0x41 0x30 0x43 
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: 0x44 0x36 0x30 0x43 0x44 0x41 0x30 0x43 0x44 0x38 0x30 0x43 0x44 0x38 0x30 0x43 0x44 0x42 0x30 0x43 
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: 0x44 0x42 0x30 0x43 0x44 0x39 0x30 0x43 0x44 0x39 0x30 0x43 0x44 0x38 0x30 0x43 0x44 0x36 0x30 0x35 
[... cut ...]
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: 0x32 0x30 0x30 0x34 0x34 0x36 0x30 0x30 0x41 0x30 0x34 0x32 0x30 0x30 0x30 0x34 0x30 0x46 0x30 0x30 
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x35 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x45 0x34 0x30 0x30 0x30 0x30 0x30 0x30 0x30 0x46 0x31 
IMP I (2025-09-16 15:03:15) SEPLOS_BMS: 0x30 0x31 
```

shining-man, 2025-09-29:
@jensbehrens Ich habe Änderungen vorgenommen. Kannst du bitte den aktuellen Teststand T0.9.0_T1 testen?

jensbehrens, 2025-10-06:
@shining-man hab's probiert, bekomme aber direkt crc checksum fehler. Kann es sein, dass die baudrate noch auf 9600 steht? Ich habe noch nicht in deine Änderungen geguckt, muss ich gestehen. Werde in den nächsten Tagen das Tag bei mir nochmal auschecken und dann prüfen... gebe Dir dann nochmal Feedback.

shining-man, 2025-10-06:
Hast du das letzte mal an der Baudrate was geändert?

shining-man, 2025-11-05:
Der SoC von den US5000 ist richtig?

jensbehrens, 2025-11-10:
jetzt sieht es gut aus! Vielen Dank! :-)

shining-man, 2025-11-15:
@jensbehrens 
In der Liste mit den Unterstützen BMS fehlen noch ein paar Daten. Kannst du mir da evtl. kurz helfen?
- Welche Adresse muss an den Pylontechs eingestellt werden, wenn man nur einen Akku hat und was wird eingestellt wenn man mehrere hat?
- An welcher Schnittstelle wird das BSC am Pylontech angeschlossen?

jensbehrens, 2025-11-15:
@shining-man klar... Jumper sind aktuell auf wie auf dem bild eingestellt, die 3 module sind via link port miteinander verbunden und bsc hängt am RS485 (bei mir aktuell die US2000).

[image]

Die Devices sind wie folg konfiguriert (Serial 0 aktuell JK Inv. BMS, Serial 1 Pylontech): 

[image]

Allerdings habe ich gestern noch eine Unstimmigkeit gefunden.... Ich habe die Akkus zum Testen auf 100% laden wollen. Aber nachdem alle auf 95% waren, habe ich festgestellt, dass sie keine Ladung mehr aufnahmen. Der zus. Akku (Gobel DIY) war zu diesem Zeitpunkt bereits bei 100% und die Pylons via CAN im VRM haben dort auch 100% angezeigt. Also scheint hier eine Differenz von 5% vorzuliegen.