shining-man/bsc_fw #118: Keine Bluetooth Daten für JK Inverter BMS

a-wolter, 2024-05-31:
Hallo,

ich habe das aktuelle (V0.6.0) auf einen ESP32 geflasht und soweit konfiguriert.
Die JK Inverter BMS (JK-PB2A16S20P) wurden auf der Bluetoothseite erkannt und die Adresse per Copy übernommen.
Gerätetyp JK-BMS [TEST] wurde für BT Device 0 und BT Device 1 ausgewählt.

Allerdings erhalte ich auch nach einem Reboot keine Daten, jedoch quittiert das BMS auch mit einem Beep die Verbindung, ich bin hier nicht sicher, ob es beide BMS gleichzeitig sind, oder nur eines, welches den Beep ausgibt. 

Das Logfile sieht für mich i.O. aus. (Bluetooth Adressen wurden hier anonymisiert):

I (1970-01-01 00:00:01) LOG: Free Space total=836081, used=42168, logSize=0
I (1970-01-01 00:00:01) MAIN: BSC V0.6.0
I (1970-01-01 00:00:01) MAIN: bootCounter=1
I (1970-01-01 00:00:01) MAIN: HW: 0
I (1970-01-01 00:00:01) MAIN: Free Heap: 197056
I (1970-01-01 00:00:01) WEB_SETTINGS: Free flash entries: 560
I (1970-01-01 00:00:01) WEB_SETTINGS: Kein Parameterfile vorhanden
I (1970-01-01 00:00:01) WEB_SETTINGS: writeConfig(): file not exist
I (1970-01-01 00:00:07) MAIN: Free Heap: 188808
I (1970-01-01 00:00:07) MAIN: Hostname: bsc
I (1970-01-01 00:00:07) MAIN: Init WLAN...
**I (1970-01-01 00:00:07) MAIN: Init BLE...**
I (1970-01-01 00:00:07) BSC_SERIAL: initSerial SerialNr=0, funktionsTyp=0
I (1970-01-01 00:00:07) BSC_SERIAL: initSerial SerialNr=1, funktionsTyp=0
I (1970-01-01 00:00:07) MAIN: Open Wifi AP
I (1970-01-01 00:00:07) Inverter: Load inverter settings(): dataSrc=0, dataSrcAdd=0
I (1970-01-01 00:00:07) Canbus: TWAI START: ESP_OK
I (1970-01-01 00:00:07) I2C: Display not found (2)
I (1970-01-01 00:00:07) I2C: Slave 0 (Adr=16) not found (2)
I (1970-01-01 00:00:07) I2C: Slave 1 (Adr=17) not found (2)
I (1970-01-01 00:00:07) I2C: Serial Ext. not found (2)
I (1970-01-01 00:00:07) BSC_SERIAL: initSerial SerialNr=2, funktionsTyp=0
I (1970-01-01 00:00:07) MAIN: Free Heap: 136060
I (1970-01-01 00:00:07) BSC_SERIAL: initSerial SerialNr=3, funktionsTyp=0
I (1970-01-01 00:00:07) BSC_SERIAL: initSerial SerialNr=4, funktionsTyp=0
I (1970-01-01 00:00:07) BSC_SERIAL: initSerial SerialNr=5, funktionsTyp=0
I (1970-01-01 00:00:07) BSC_SERIAL: initSerial SerialNr=6, funktionsTyp=0
I (1970-01-01 00:00:07) BSC_SERIAL: initSerial SerialNr=7, funktionsTyp=0
I (1970-01-01 00:00:07) BSC_SERIAL: initSerial SerialNr=8, funktionsTyp=0
I (1970-01-01 00:00:07) BSC_SERIAL: initSerial SerialNr=9, funktionsTyp=0
I (1970-01-01 00:00:07) BSC_SERIAL: initSerial SerialNr=10, funktionsTyp=0
I (1970-01-01 00:00:07) MAIN: WIFI ready (0)
I (1970-01-01 00:00:07) MAIN: WIFI AP start (10)
I (1970-01-01 00:00:07) MAIN: WIFI AP stop (11)
I (1970-01-01 00:00:07) MAIN: WIFI AP start (10)
**I (1970-01-01 00:00:07) MAIN: Init BLE...ok**
I (1970-01-01 00:00:26) MAIN: WIFI event: 12
I (1970-01-01 00:00:28) MAIN: WIFI event: 14
I (1970-01-01 00:04:10) ALARM: Trigger 1, value 0 - 
I (1970-01-01 00:04:10) ALARM: Trigger 2, value 0 - 
I (1970-01-01 00:04:10) ALARM: Trigger 3, value 0 - 
I (1970-01-01 00:04:10) ALARM: Trigger 4, value 0 - 
I (1970-01-01 00:04:10) ALARM: Trigger 5, value 0 - 
I (1970-01-01 00:04:10) ALARM: Trigger 6, value 0 - 
I (1970-01-01 00:04:10) ALARM: Trigger 7, value 0 - 
I (1970-01-01 00:04:10) ALARM: Trigger 8, value 0 - 
I (1970-01-01 00:04:10) ALARM: Trigger 9, value 0 - 
I (1970-01-01 00:04:10) ALARM: Trigger 10, value 0 - 
I (1970-01-01 00:04:21) ALARM: Trigger 1, value 0 - 
I (1970-01-01 00:04:21) ALARM: Trigger 2, value 0 - 
I (1970-01-01 00:04:21) ALARM: Trigger 3, value 0 - 
I (1970-01-01 00:04:21) ALARM: Trigger 4, value 0 - 
I (1970-01-01 00:04:21) ALARM: Trigger 5, value 0 - 
I (1970-01-01 00:04:21) ALARM: Trigger 6, value 0 - 
I (1970-01-01 00:04:21) ALARM: Trigger 7, value 0 - 
I (1970-01-01 00:04:21) ALARM: Trigger 8, value 0 - 
I (1970-01-01 00:04:21) ALARM: Trigger 9, value 0 - 
I (1970-01-01 00:04:21) ALARM: Trigger 10, value 0 - 
I (1970-01-01 00:08:13) ALARM: Trigger 1, value 0 - 
I (1970-01-01 00:08:13) ALARM: Trigger 2, value 0 - 
[... 75 lines cut ...]
I (2024-05-31 14:29:00) ALARM: Trigger 1, value 0 - 
I (2024-05-31 14:29:00) ALARM: Trigger 2, value 0 - 
I (2024-05-31 14:29:00) ALARM: Trigger 3, value 0 - 
I (2024-05-31 14:29:00) ALARM: Trigger 4, value 0 - 
I (2024-05-31 14:29:00) ALARM: Trigger 5, value 0 - 
I (2024-05-31 14:29:00) ALARM: Trigger 6, value 0 - 
I (2024-05-31 14:29:00) ALARM: Trigger 7, value 0 - 
I (2024-05-31 14:29:00) ALARM: Trigger 8, value 0 - 
I (2024-05-31 14:29:00) ALARM: Trigger 9, value 0 - 
I (2024-05-31 14:29:00) ALARM: Trigger 10, value 0 - 
I (2024-05-31 14:29:04) ALARM: Trigger 1, value 0 - 
I (2024-05-31 14:29:04) ALARM: Trigger 2, value 0 - 
I (2024-05-31 14:29:04) ALARM: Trigger 3, value 0 - 
I (2024-05-31 14:29:04) ALARM: Trigger 4, value 0 - 
I (2024-05-31 14:29:04) ALARM: Trigger 5, value 0 - 
I (2024-05-31 14:29:04) ALARM: Trigger 6, value 0 - 
I (2024-05-31 14:29:04) ALARM: Trigger 7, value 0 - 
I (2024-05-31 14:29:04) ALARM: Trigger 8, value 0 - 
I (2024-05-31 14:29:04) ALARM: Trigger 9, value 0 - 
I (2024-05-31 14:29:04) ALARM: Trigger 10, value 0 - 
I (2024-05-31 14:30:44) Inverter: Load inverter settings(): dataSrc=0, dataSrcAdd=0
I (2024-05-31 14:30:46) Inverter: Load inverter settings(): dataSrc=0, dataSrcAdd=0
I (2024-05-31 14:30:48) Inverter: Load inverter settings(): dataSrc=0, dataSrcAdd=0

Habe ich etwas übersehen?

tanelvakker, 2024-06-04:
You can use the new ones over RS485 first port.
UART1 Protocol No.: 0 (4G-GPS Remote module). Normal JK-BMS in BSC settings
Dip switch address needs to be 0

a-wolter, 2024-06-04:
[register_map_1029228503.pdf](https://github.com/user-attachments/files/15569458/register_map_1029228503.pdf)

Evtl hilft das

shining-man, 2024-06-06:
Wenn das für das JK Inverter BMS ist, dann schaut es auf den ersten Blick brauchbar aus. 
Vielleicht findet noch jemand ein direkt Englisches Dokument, das schöner (lesbarer) ist.
Immer diese chinesischen Dokumente. Warum machen die das nicht einfach auf Englisch?

D3R-ST3FAN, 2024-06-16:
Es gibt einen [Fork](https://github.com/txubelaxu/esphome-jk-bms) vom [esphome-jk-bms](https://github.com/txubelaxu/esphome-jk-bms) wo die Unterstützung für das JK Inverter BMS via RS485 und BT funktioniert.
Ich bin jetzt nicht so der Entwickler aber das [hier](https://github.com/txubelaxu/esphome-jk-bms/blob/main/components/jk_rs485_bms/jk_rs485_bms.cpp) aus dem Fork](https://github.com/txubelaxu/esphome-jk-bms)
sieht interessant aus.

shining-man, 2024-06-17:
Ich bin dabei, das ganze nach dem Protokoll zu implementieren. Das ist mir immer lieber, als es irgendwo abzuschreiben.
Gerade wenn es ans Fehler suchen geht, ist es einfacher. Dennoch Danke für die Infos.

MeisterQ, 2024-06-19:
> Ich bin dabei, das ganze nach dem Protokoll zu implementieren. Das ist mir immer lieber, als es irgendwo abzuschreiben. Gerade wenn es ans Fehler suchen geht, ist es einfacher. Dennoch Danke für die Infos.

Sehe ich das richtig, dass das BMS Modbus RTU spricht? Zumindest laut dem Übersetzen Dokument oben wird von Holding Registern und Function Codes gesprochen.

Woher nimmst du deine Infos zum Implementieren? Hast du da weitere Dokumente?

tanelvakker, 2024-06-19:
@shining-man do you have new jk-bms hardware to test this on? I could send you one if needed

shining-man, 2024-06-19:
@MeisterQ Im Dokument steht eigentlich alles drin. Mehr Infos brauche ich nicht.

@tanelvakker Thanks for the offer. I don't have one, but I don't rent equipment anymore. If something breaks, that's a big problem.

MeisterQ, 2024-06-19:
> @MeisterQ Im Dokument steht eigentlich alles drin. Mehr Infos brauche ich nicht.
> 
> @tanelvakker Thanks for the offer. I don't have one, but I don't rent equipment anymore. If something breaks, that's a big problem.

Welches Dokument genau?

shining-man, 2024-06-19:
@MeisterQ Das oben angehängte: https://github.com/shining-man/bsc_fw/issues/118#issuecomment-2148244985

MeisterQ, 2024-06-19:
@shining-man Ich habe mal einen Terminal Mitschnitt laufen lassen wenn man mit dem JK BMS Monitor die Livedaten abfragt:

Vom Tool wird gesendet: 0x01 0x10 0x16 0x20 0x00 0x01 0x02 0x00 0x00 0xD6 0xF1

Als Antwort kommt: 85 170 235 144 2 5 238 12 243 12 243 12 238 12 235 12 246 12 244 12 247 12 240 12 250 12 241 12 239 12 250 12 245 12 241 12 246 12 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 255 255 0 0 243 12 15 0 9 4 51 0 53 0 64 0 67 0 78 0 88 0 103 0 116 0 121 0 113 0 104 0 101 0 91 0 69 0 65 0 53 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 215 0 0 0 0 0 43 207 0 0 168 47 7 0 62 33 0 0 213 0 217 0 0 0 8 0 0 0 0 68 99 234 2 0 192 69 4 0 1 0 0 0 238 119 4 0 100 0 0 0 27 214 3 0 1 1 0 0 0 0 0 0 0 0 0 0 0 0 0 0 255 0 1 0 0 0 157 3 23 0 0 0 136 31 64 64 0 0 0 0 183 20 0 0 0 1 0 1 0 6 0 0 193 47 13 0 0 0 0 0 215 0 212 0 218 0 157 3 60 108 103 8 50 0 0 0 128 81 1 0 0 0 0 0 0 0 0 0 0 0 0 0 0 254 255 127 220 47 1 1 176 7 0 0 0 19 1 16 22 32 0 1 4 75

Nur halt in HEX. Hatte es gerade nur in Dezimal die Antwort.

Vielleicht hilft das auch noch.

Hast du schon ein File online wo das Auslesen drin ist?

shining-man, 2024-06-30:
@MeisterQ Danke für den Mittschnitt. Ist das das einzige Telegramm, dass die Software zum BMS sendet? 
In der Antwort müssten dann alles drin sein. Zellspannungen und Errors.

MeisterQ, 2024-06-30:
> @MeisterQ Danke für den Mittschnitt. Ist das das einzige Telegramm, dass die Software zum BMS sendet?
> 
> Die Antwort die zurück kommt, schaut für mich nicht nach einer Modbus Antwort aus. Kannst du mit einem Modbus Tool (z.B. Modbus Poll) einmal versuchen eine Abfrage entsprechend dem Protocoll zu machen? Das sehen wir ob eine Antwort kommt.

Ja das ist wenn man auf der "Livedaten" Seite vom BMS ist. Ich denke das ist das "Standard GPS Protokoll" was man am BMS für den 485 Port auswählen kann.

Für Modbus muss man Protokoll 1 (Modbus V1.0) Auswählen.

Das habe ich mal mit qmodmaster versucht auszulesen.

Das ist das Ergebnis:

[image]

shining-man, 2024-07-01:
Hier wäre eine erste Testversion zum prinzipiellen Funktionstest: 
[firmware_V0.6.1_T2.bin.zip](https://github.com/user-attachments/files/16047715/firmware_V0.6.1_T2.bin.zip)

MeisterQ, 2024-07-01:
> Hier wäre eine erste Testversion zum prinzipiellen Funktionstest: [firmware_V0.6.1_T2.bin.zip](https://github.com/user-attachments/files/16047715/firmware_V0.6.1_T2.bin.zip)

Ich kann es leider nicht testen, da meine Hardware nicht zu den Pins vom BSC passt.
Aber @a-wolter vielleicht?

shining-man, 2024-07-20:
@MeisterQ Welchen Port vom JK hast du für die Kommunikation genutzt, und was hast du dazu im BMS eingestellt?

Ximerox, 2024-07-20:
> Hier wäre eine erste Testversion zum prinzipiellen Funktionstest: [firmware_V0.6.1_T2.bin.zip](https://github.com/user-attachments/files/16047715/firmware_V0.6.1_T2.bin.zip)

Hab die Firmware getestet. Aber nun erhalte ich gar keine Daten mehr vom BMS (JK Inverter BMS)

MeisterQ, 2024-07-22:
> @MeisterQ Welchen Port vom JK hast du für die Kommunikation genutzt, und was hast du dazu im BMS eingestellt?

Auf dem Kommunikationsboard ganz rechts die 2 RS485 Ports sind dafür da. Zusätzlich muss das BMS auf ID 1 stehen, also der erste Dip Schalter auf Ein, ODER je nach BMS was du auslesen willst, die Modbus UID auf die Zahl stellen, worauf das BMS mit den DIP konfiguriert ist.

closed 2024-12-31 by shining-man