shining-man/bsc_fw #120: bsc/trigger/ wird mit MQTT nicht mehr geliefert

tueftler0, 2024-06-01:
Normalerweise wurde bisher alle ca 30s alle Topics via MQTT übertragen, auch wenn sich keine Wertänderung ergab (glaube ich zumindest).
Mir fehlen derzeit alle Trigger-Übertragungen seid dem neuen Release.
In der RestAPI sind sie sichtbar.

shining-man, 2024-06-01:
Die Trigger wurde auch vorher nicht zyklisch übertragen. Hier hat sich nichts geänert.

tueftler0, 2024-06-01:
Danke fürs Feedback.
Okay, dann wäre es toll, wenn man dies in Zukunft mal mit integrieren würde.
HomeAssistant beispielsweise hätte gerne alle x Minuten eine Message, ansonsten gehen die States auf Unbekannt:
[image]

shining-man, 2024-06-02:
Ansich finde ich es nicht schön etwas einzubauen nur um irgendwelche System mit etwas zu befriedigen, was sie morgen vielleicht schon wieder ändern. 
Macht es wirklich Sinn Nachrichten zyklisch zu versenden, auch wenn sie sich nicht geändert haben? Wäre es nicht Sinnvoller man würde den QoS erhöhen?
Ansich kommt es auf die 10 Nachrichten nicht an, aber ich wollte genrell den Traffic senken und andere Werte auch nur bei Änderung senden, das würde dann vmtl. an mehr Stellen zu diesem Verhalten führen. Gut, dass wir auf das Thema zu sprechen kommen.

tueftler0, 2024-06-02:
Ich kann Dich da schon verstehen. Jedoch kann es ja auch gut sein, dass der MQTT-Broker einmal neu gestartet wird oder ein Repeater ausfällt (usw) und in dieser Zeit ein Trigger oder sonstiger Wert sich ändert.
Das würde dann bis zur nächsten Änderung nie auffallen.
Daher finde ich es schon sinnvoll, wenn man alle Werte alle 30 Sekunden oder so einmal rüber sendet.
Auch könnte man in dieser Funktion das HA-Discovery integrieren, da hier ja scheinbar beim Anlegen immer alle via MQTT möglichen Daten vorliegen müssen. Das hatte glaub mal NSC heraus gefunden.

shining-man, 2024-06-02:
HA-Discovery wird vmtl. nie den Weg in den BSC finden. Das ist einfach zu viel Overhead. 
Einen ausfallenden Repeater könnten man mit QoS 1 auch feststellen, denn dann muss die Nachricht mindestens einmal zugestellt werden.
Zum Neustart: Es macht mehr Sinn alle Nachrichten erneut zusenden, wenn es zu einem reconnect kommt.

Ansich möchte ich die Flut der Nachrichten etwas ein dämmen. Ich möchte auf ein dynamisches Konzept umsteigen, bei dem es keine festen Sendezeiten mehr gibt, sondern nur bei Änderung und Gewichtung der Werte gesendet wird. 
Wenn das BPN einmal kommen sollte, dann gibt es pro BMS fast 40 Temperaturwerte mehr. Wenn ich die alle zyklisch sende, dann gibt es keinen 30s Intervall mehr, sondern ich muss langsamer werden. Daher nehme ich jetzt ungern Werte in das zyklische Senden auf.

tueftler0, 2024-06-02:
HA-Discovery könnte man ja auch einmalig über das Webfrontend, oder bei jedem Neustart triggern - Aber nun ja erst einmal egal ;)

QoS 1 wäre definitiv sinnvoll, wenn ein Senden aller Werte umgangen werden soll.
Alle Werte nur bei einem Reconnect senden wäre schon eine Lösung.
Was man genau gegen dieses "Unbekannt" machen kann werde ich mal nachforschen.
Blöd ist eben, wenn man nach true/false triggert und dann ein unknown bekommt :)

tueftler0, 2024-06-03:
Korrektur der Aussage:
HomeAssistant führt den Trigger (Sensor Component) als "unavailable" wenn bis zu dem Zeitpunkt über MQTT kein Wert übermittelt wurde.
Grundsätzlich okay, wenn der State ja nicht sicher anzunehmen ist.
Daher ist Deine Idee mit dem Senden aller Datenpunkte bei reconnect wahrscheinlich passend.

Ein Timeout für den Sensor ist zwar konfigurierbar, aber standardmäßig deaktiv.

tueftler0, 2024-09-16:
Zusätzlich zum Senden bei Reconnect bräuchte man noch eine Möglichkeit manuell das Senden zu aktivieren.
Wenn man z.B. nur alle YAML-Dateien neu einlesen lässt, oder Sensoren neu benennt, dann wäre es gut einmalig alles auf den aktuellen Stand setzen zu können.
Vorstellbar wäre z.B. ein Trigger (VTrigger?) über MQTT oder Tastendruck in der GUI.

shining-man, 2024-10-01:
Ich werde jetzt vorerst einmal die Trigger mit in dem zyklischen Sendeintervall aufnehmen. Damit sollte sich das Problem und die Notwedigkeit des Buttons erledigt haben?

a-wolter, 2024-10-13:
> Ich werde jetzt vorerst einmal die Trigger mit in dem zyklischen Sendeintervall aufnehmen. Damit sollte sich das Problem und die Notwedigkeit des Buttons erledigt haben?

Ich denke so würde es passen. Es kann nämlich tatsächlich ein Problem sein, da der mqtt Status wie von @tueftler0 erwähnt, nicht immer passt. 

Beispiel: Trigger wurde aktiv, mqtt reconnect und mqtt Status hat sich nicht verändert obwohl der trigger low gegangen ist in der Zwischenzeit:

[image]
[image]

So ist er nun über Tage als high im mqtt (BMS Verbindungsverlust)

closed 2024-12-31 by shining-man