shining-man/bsc_fw #123: Warnings aus Seplos werden als Error an Victron weitergeleitet

SladeTheLeveller, 2024-06-21:
Das Seplos sendet bei einer Schwelle x eine Warnung, über diese Warnungsschwelle muss es 30 Minuten bleiben, um 100% SOC anzuzeigen. Die Warning wird als Error im BSC interpretiert, sowohl in MQTT als auch im Victron selber und führt damit zum Ladungsabbruch.

shining-man, 2024-06-25:
Wie heißt die Warnung?
Kann diese Warnung im Seplos eingestellt werden, oder ist die fest?

shining-man, 2024-06-29:
"High voltage alarm for total voltage" werte ich als Warnung und 
"Overvoltage protection for total voltage" als Error. Da sehe ich erst einmal keinen Fehler.
Ums das Problem eingrenzen zu können braucht es mehr Infos:
- Welchen Fehler meldet der Wechselrichter?
- Was kommt über MQTT als Fehler?
- Welche Fehler zeigt das BMS selber an?

RRonGit, 2024-06-30:
Ich habe mal versucht die Meldungen zusammen zu tragen, weis nicht ob das so reicht.
Die Meldungen werden ab einer Zellspannung >3.5V oder ab einer Gesamtspannung >56V angezeigt (ist gem. Seplos der "Over voltage warning" Bereich.

**_WR:_**
[image]

**_MQTT:_**
[image]

[image]

_**BMS:**_
[image]

_**BSC:**_
[image]

shining-man, 2024-06-30:
@RRonGit Danke für die Infos. Hab jetzt den Fehler gefunden.

RRonGit, 2024-07-02:
@shining-man wird die entsprechende Fehlerbehebung im nächsten Release drinnen sein?

shining-man, 2024-07-02:
> @shining-man wird die entsprechende Fehlerbehebung im nächsten Release drinnen sein?

Ja. Aber das neue Release wird noch dauern.
Im Branch [new_features_24-2](https://github.com/shining-man/bsc_fw/tree/new_features_24-2) ist es mit drin. Der ist aber nicht stable.

RRonGit, 2024-07-03:
@shining-man Ok, alles klar. Freue mich jetzt schon aufs neue Release und auf jedes weitere. Super Projekt, danke für deine tolle Arbeit.

closed 2024-12-31 by shining-man