shining-man/bsc_fw #137: Probleme mit 2x NEEY

peterfroehlich, 2024-07-21:
**Setup:** 
2x Seplos BMS (nicht mit BSC) verbunden mit je einem NEEY 4A Balancer
Beide sind im BSC (auf ESP32-wroom Dev Board) auf BT ID 0 und 1 eingetragen. 

**Problem:** 
Der Neey auf ID 0 funktioniert wunderbar, aber der auf ID 1 wird in der Übersicht als "c" angezeigt, reported aber keine Daten. Im Log sieht es so aus als ob die Verbindung immer wieder auf und abgebaut wird. 

Beide NEEY sind über die Android App konfigurierbar, verhalten sich aber unterscheidlich wenn man sich direkt per Bluetooth paired. #1 verlangt einen Pin, #2 verlangt keinen. 

Ich hab schon versucht über die App den Pin auf 123456 zu ändern, aber das hat nichts geändert. 

**Log:** 
> I (2024-07-21 22:35:01) BLE_HANDLER: Device connected; dev=1
> I (2024-07-21 22:35:06) BLE_HANDLER: No cyclical data from dev 1
> I (2024-07-21 22:35:06) BLE_HANDLER: Device disconnected single, dev=2b:80:03:f5:7b:c9
> I (2024-07-21 22:35:07) BLE_HANDLER: No cyclical data from dev 1
> I (2024-07-21 22:35:07) BLE_HANDLER: Device disconnected single, dev=2b:80:03:f5:7b:c9
> I (2024-07-21 22:35:08) BLE_HANDLER: onDisconnect() dev=1, mac=2b:80:03:f5:7b:c9
> W (2024-07-21 22:35:11) BLE_HANDLER: Reconnect failed: mac=2b:80:03:f5:7b:c9
> I (2024-07-21 22:35:13) BLE_HANDLER: onConnect() 2b:80:03:f5:7b:c9
> I (2024-07-21 22:35:22) BLE_HANDLER: Device connected; dev=1
> I (2024-07-21 22:35:27) BLE_HANDLER: No cyclical data from dev 1
> I (2024-07-21 22:35:27) BLE_HANDLER: Device disconnected single, dev=2b:80:03:f5:7b:c9

shining-man, 2024-07-22:
Kann es sein, dass es zwei Unterschiedliche NEEYs sind? Ein GW-24S4EB und ein EK-24S4EB?
Für den EK-24S4EB gibt es bisher nur eine Testversion.

peterfroehlich, 2024-07-22:
Mist, mir war nicht bewusst das es das es da Unterschiede gibt. Ich habe die Versionsstände in der App mal ausgelesen, lustigerweise melden sich beide als EK-24S4EB. 

Der Funktionierende:
Hardware Version: HW-3.2.0
Software Version: ZH-1.3.2
App Version: V1.2.9
Prod Date: 20230829

Der nicht funktionierende:
Hardware Version: HW-3.2.0
Software Version: ZH-2.0.4
App Version: V1.2.27
Prod Date: 20240220
 
 Ich habe das 0.6.0 release drauf. 
 Welcher Branch ist denn die besagte "Testversion"? Ich teste gerne mit. :) 

Hier noch die Ansicht vom "Funktionierenden": 
[image]

tueftler0, 2024-07-22:
Hallo.
Die Teststände für z.B. den neuen Neey-Dialekt findest Du im Discord-Channel unter /firmware/Teststände.
Die Bezeichnung des vorliegenden Neeys findest Du abgedruckt auf der Leiterplatte. Ob man die Hardware-Version in der App zurate ziehen kann weiß ich nicht. Da hatte ich bei den Tests nicht drauf geschaut.

peterfroehlich, 2024-07-22:
Jetzt hatte ich gerade "new_features_24-2" gebaut, ist das der letzte? 
Auf jeden Fall schönen Dank für die Hilfe und das Projekt im Ganzen!  <3

peterfroehlich, 2024-07-22:
Also ich habe die selbstgebaute firmware mal ausprobiert, damit funktioniert die Kommunikation. Die Werte werden korrekt ausgelesen soweit ich das sehe:

[image]

Die Balancer Konfiguration nicht: 

[image]

Logfile ist sauber, beim Konfiguration auslesen kam das hier :

> I (2024-07-22 13:57:17) MAIN: StartReadDataFromNeey
> I (2024-07-22 13:57:17) BLE_HANDLER: readDataFromNeey()
> I (2024-07-22 13:57:18) NEEY: S; dev=1, p=0
> I (2024-07-22 13:57:18) NEEY: S; dev=1, p=1
> I (2024-07-22 13:57:18) NEEY: S; dev=1, p=2
> I (2024-07-22 13:57:18) NEEY: S; dev=1, p=3

shining-man, 2024-07-22:
"new_features_24-2" ist der aktuelle new Feature Branch. Wichtig ist, dass nicht unbedingt alles stable seins muss. Aktuell sollte es zwar keine großen Prbleme mit dem Branch geben, aber stable ist immer nur der main branch.
Das lesen und schreiben der Settings muss ich erst noch machen, daher kommen so komische Meldungen im Log.

Schatemuhl, 2024-10-03:
Hallo an alle
Ich habe mich heute mal mit dem BSC beschäftigt, und wollte vorerst nur die NEEY Ballancer auslesen. 
Nun habe ich festgestellt das die EK-24S4EB nocht nicht mit der Version V 0.6.0 gehen.
Wie bekomme ich diese Testversion von der hier die Rede ist auf den ESP ?   Danke für Eure Antworten.

Das ist echt ein super Projekt, werde auf jeden Fall Sponsor.

tueftler0, 2024-10-03:
Testversionen findest Du im Discord-Chat
Dort kommst du hierüber hin: https://discord.gg/WRUhegRPQn