shining-man/bsc_fw #143: Auswahl BMS Temperatur Sensor

a-wolter, 2024-07-30:
Laut WebUI kann ich bei Auswahl "BMS" nur 0-2 als Temp. Sensor festlegen. Ich konnte problemlos "4" eintragen, der Wert welcher an VenusOS gesendet wird ändert sich allerdings nicht.
Zudem ist nicht ganz klar, welcher Sensor aus welchem Pack nun übertragen wird. Ich vermute es ist ein Wert aus dem Pack mit der niedrigsten ID, da die Temperatur passen würde.

Mit z.B. 5 JK Inverter BMS im Daisy-Chain an Serial 2 sollten min. auch 5 Sensoren zur Verfügung stehen.
Evtl. sogar mehr, da über die JK App pro Pack 5 Sensoren angezeigt werden

[image]

shining-man, 2024-07-31:
Werte kannst du von 0-63 eintragen. Aber für BMS ist nur 0-2 gültig. Dies hat den Grund, dass in der internen Datenstruktur des BSC platz für 3 Temperaturen ist. Wenn ein BMS mehr als drei Temperaturen liefert, dann nehme ich immer die ersten drei.
Aus diesen drei Temperaturen wird dann auch die Min./Max.-Temp genommen. Ich vermute, dass eine der Temperaturen 0 ist und daher ist die Min.-Temp auch 0. Das wäre von daher erst einmal richtig.

Man muss jetzt schauen, welche drei Temperaturen sinn ergeben.

a-wolter, 2024-07-31:
Das kann ich bestätigen 
[image]

a-wolter, 2024-07-31:
Hier der Vergleich von JK App zu /restapi

[image]
[image]

a-wolter, 2024-08-07:
Funktioniert ab der 0.6.1_T7

closed 2024-08-07 by a-wolter