shining-man/bsc_fw #155: BSC kommt durcheinander bei Verbindungsverslust zum MQTT Server, Alarm auf Victron

ChrisX159, 2024-08-30:
Schaltet man bei aktiver Verbindung BSC zu MQTT den Accesspoint aus oder trennt man den MQTT Server vom Strom, so fällt das BSC kurz aus.
=> Keine Daten mehr an Victron => Anlage geht auf Störung

BSC fängt sich jedoch wieder.

Problematisch ist das ganze bei Schwacher WLAN-Verbindung, da sich das ganze ständig wiederholt

https://discordapp.com/channels/1037083515755434005/1082587204561412116/1254873135996600492

net2use, 2024-09-25:
Hallo zusammen,

in Discord konnte ich dem ganzen nicht mehr folgen...
Daher auf diesem Wege, ich konnte das selbe Problem feststellen jedoch den Auslöser MQTT nicht verifzieren.
Im Programmcode konnte ich jedoch die Stelle fixieren, welche anschließend zu dem Ausschalten des Victron führt:
File: InverterBattery.cpp
`
Function:
  void InverterBattery::getBatteryVoltage(Inverter &inverter, Inverter::inverterData_s &inverterData)
  {
    if((millis()-getBmsLastDataMillis(inverterData.u8_bmsDatasource))<CAN_BMS_COMMUNICATION_TIMEOUT)
    {
      inverter.inverterDataSemaphoreTake();
      inverterData.batteryVoltage = (int16_t)(getBmsTotalVoltage(inverterData.u8_bmsDatasource)*100.0f);
      inverter.inverterDataSemaphoreGive();
      return;
    }
    else //Wenn Masterquelle offline, dann nächstes BMS nehmen das online ist
    {
      for(uint8_t i=0;i<SERIAL_BMS_DEVICES_COUNT;i++)
      {
        //Wenn BMS ausgewählt und die letzten 5000ms Daten kamen
        if(((inverterData.u16_bmsDatasourceAdd>>i)&0x01) && ((millis()-getBmsLastDataMillis(BMSDATA_FIRST_DEV_SERIAL+i))<CAN_BMS_COMMUNICATION_TIMEOUT))
        {
          inverter.inverterDataSemaphoreTake();
          inverterData.batteryVoltage = (int16_t)(getBmsTotalVoltage(BT_DEVICES_COUNT+i)*100.0f);
          inverter.inverterDataSemaphoreGive();
          return;
        }
      }
    }
    inverter.inverterDataSemaphoreTake();
    **inverterData.batteryVoltage = 0;**
    inverter.inverterDataSemaphoreGive();
  }

`

In meinem Fall findet eine Mittelwertbildung von zwei Packs statt. Beim Victron kommen dann 60V + 0V => 30V an, dies führt dazu, dass er eine Low-Bat detectiert.
Die Ursache, warum er jedoch in diesem Moment keine gültige Quelle findet und in den Fallback-Return fällt habe ich noch nicht gefunden.

Gibt es hierzu vll. schon eine BugFix-Ansatz? Ich sitze biszu zwei mal am Tag im dunkeln, wenn ich den BSC produktiv schalte.

shining-man, 2024-09-26:
@net2use Wie hast du das verifiziert?

Die Mittelwertbildung betrifft nur den SoC und nicht die Batteriespannung.
Die Funktion ermittelt die Batteriespannung. Dazu wird zuerst geschaut ob das Masterpack online ist, wenn ja wird diese Spannung genommen. Ist das Masterpack nicht online wird geschaut, ob ein anderes Pack online ist, wen ja wird die Spannung von dem nächsten das online ist, genommen. Ist kein Pack online, wird eine Spannung von 0V gesendet, was richtig ist, denn es ist ja kein Pack mehr online.
Ob dein Pack immer wieder offline ist siehst du am "valid" per MQTT. In der restapi sieht man es auch.

Das Problem beim MQTT ist, dass eine System Lib blockierend, dadurch hängt der CAN Task und es wird ein paar Sekunden nichts an den Wechselrichter gesendet, was Victron dazu veranlasst, dass dieser auf Fehler geht, was ja auch wieder richtig ist.
Einen Lösungsansatz gibt es, der aber noch nicht stabil läuft.

net2use, 2024-09-26:
@shining-man 
Vielen Dank für die Aufklärung der Funktionsweise und Ablauf - das erklärt weshalb ich keine Funktion zur MIttelung finden konnte.
Bezüglich Verifizierung: Mein MQTT und BSC sind mittels Firewall voneinander getrennt, ich hab dementsprechend Pakete von BSC zum MQTT Server einfach mal für ein paar Minuten blockiert. Dies führte zu keinem Abbruch. Möglicherweise jedoch könnten WLAN Abbruche als solches ein Problem sein, dies habe ich nicht verifziert. Auch habe ich nicht die Pakete von MQTT zum BSC blockiert, da aber TCP als Protokoll verwendet wird, sollte die Sperrung in eine Richtung ausreichend sein.

Aber wenn es in der Lib zu einem blockierenden verhalten kommt (aus welchem Grund auch immer), könnte das zum Problem führen, da ja das Alter der Daten abgefragt wird millis() - DATEN_ALTER > TIMEOUT => Dann schickt er automatisch die 0V und 0V sind für den Victron bedeutend für eine sofortige Abschaltung. Jedoch nicht in erster Linie das Problem, dass keine Daten an Victron gesendet werden, denn hier hat (mein) Victron Inverter einen etwas längeren Timeout.

Wäre das eine plausible Erklärung?
Falls ja, welche Möglichkeiten gäbe es das Verhalten zu verbessern/beheben?

net2use, 2024-10-01:
Kurzes Feedback von meiner Seite:
Ich habe die Konstante CAN_BMS_COMMUNICATION_TIMEOUT temporär von 5000 auf 15000 hoch gesetzt, seit dem gab es keine Probleme mehr. Parallel dazu habe ich die Zeitdifferenz geloggt, sobald die bisherigen 5000ms überschritten wurde.
Dies passierte mehrmals am Tag und der Mittelwert lag hier bei etwa 8000ms. Es gab auch einen Ausreiser mit 11000ms. Auffällig oft ist es gekommen, wenn ich die WiFi Verbindung künstlich verschlechtert habe oder sogar zum abreisen brachte. In wie fern hier MQTT mit auslöser war konnte ich nicht beurteilen.
Aufgrund welcher Grundlage wurde der Timeout auf 5000ms definiert?

shining-man, 2024-10-02:
Eine richtige Grundlage gibt es da nicht. Die Zeit habe ich irgendwann einmal so definiert. Wenn von einem BMS 5s keine Daten kommen, dann ist dies eine lange Zeit. 
Ich kann dir einmal eine Version machen, die bei Hängern eines Tasks den Taskstate mitloggt, dann sieht man vielleicht leichter, ob und welcher Task das Problem ist. 
Hast du Timeouts an der seriellen Schnittstelle?

net2use, 2024-10-03:
Ich habe noch eine Verständnisfrage im Quelltext, in dem oben geposteten Code werden im selben Context zwei verschiedene Konstanten verwendet:
getBmsLastDataMillis(**BMSDATA_FIRST_DEV_SERIAL**+i)
getBmsTotalVoltage(**BT_DEVICES_COUNT**+i)

Was ist der Unterschied zwischen den beiden?

shining-man, 2024-10-04:
> @shining-man Danke für die Rückinfo. Die Timeouts kommen sporadisch von zwei seriellen Schnittstellen. Aber nur selten gleichzeitig. Ich kann die Version gerne mal probieren.

Welche Version nutzt du eigentlich? Für welche Version brauchst du das loggen der Taskstates?
Nutzt du mehr als zwei Schnittstellen? Was heißt sporadische Timeouts? Wenn deine Timeouts länger den 5s sind, dann ist auch klar warum du die Probleme hast.

shining-man, 2024-10-04:
> Ich habe noch eine Verständnisfrage im Quelltext, in dem oben geposteten Code werden im selben Context zwei verschiedene Konstanten verwendet: getBmsLastDataMillis(**BMSDATA_FIRST_DEV_SERIAL**+i) getBmsTotalVoltage(**BT_DEVICES_COUNT**+i)
> 
> Was ist der Unterschied zwischen den beiden?

Zur Zeit keiner

closed 2024-12-31 by shining-man