shining-man/bsc_fw #156: History über CAN versenden - Seplos E10

Saharel001, 2024-09-01:
Wie bereits im Discord angesprochen, wird die History die vom BMS kommt nicht via CAN übermittelt. Bedeutet das mir die geladenen und entladenen kWh fehlen. Wenn das BMS direkt an dem Wechselrichter bzw. die Steuereinheit von Victron angeschlossen ist (Cerbo GX) habe ich die Werte zur Verfügung. Für mich wichtig für die Auswertung.

Saharel001, 2025-02-12:
Wird das irgendwann noch kommen?

shining-man, 2025-02-23:
Um das ganze sauber und durchgängig zu haben, müsste ich das für alle BMS nachziehen. 
Dies ist ein ziemlich großer Aufwand, da ich alle Protokolle durchgehen muss, wer wie die Daten sendet. Sendet sie überhaupt die Mehrzahl der BMS?
Daher steht das eher hinten auf der Liste. Denkbar wäre so etwas per MQTT zu versenden. Das kann ich auch für ein einzelnes BMS machen. Würde das überhaupt helfen?

shining-man, 2025-05-02:
Ich habe mir das Protokoll vom Seplos einmal angschaut. Die gewünschten Werte liefert das BMS nicht.
Das einzige was ich gefunden habe ist die "remaining capacity", aber das hilft nur bedingt für die geladene/entladene Energie (kWh).
Daraus könnte ich zwar mit der Spannung berechen was aktuell an kWh in der Batterie ist, aber ich weis immer noch nicht was geladen/entladen wurde. Das müsste ich dann selber über die Zeit ausrechen, aber ob das wirklich genau wird?
Vielleicht hat jemand mehr Informationen, wo die Werte herkommen?

tueftler0, 2025-05-04:
> Vielleicht hat jemand mehr Informationen, wo die Werte herkommen?
Sehe ich genau so - sehe da auch nichts. Glaube das JK-Inverter-BMS macht das.
Evtl sendet das Seplos aber ja auch mehr Infos über CAN, als über RS485