shining-man/bsc_fw #157: Problem mit Sonderzeichen "Tilde" im WLAN-Passwort

toptec01, 2024-09-09:
Hallo,

mein BSC verbindet sich nicht mit meinem Accesspoint der das im Titel genannte Zeichen in Passwort hat.

Gibt es dort Einschränkungen in der Firmware? Andere ESP32 verbinden sich ohne Probleme.

BR toptec

shining-man, 2024-09-17:
Das Problem mit dem Tildezeichen kommt daher, da ich es als Steuerzeichen in meinem Config-File nehme. Das lässt sich jetzt auch erst einmal nicht so schnell ändernd. Alle anderen Sonderzeichen sollten theoretisch funktionieren.

tueftler0, 2024-09-17:
Habe ich im Wiki mal temporär so aufgenommen:
https://github.com/shining-man/bsc_fw/wiki/03-Software#system

Fuendiger, 2025-03-30:
Das ~ kan nach wie vor nicht genutzt werden, aber es kann jetzt zumindest nicht mehr eingegeben werden bzw. wird als Fehler angezeigt.