shining-man/bsc_fw #167: Error durch Low Voltage Protection blockiert laden und entladen dauerhaft

Saharel001, 2024-10-03:
Wenn das BSC in den Error Zustand (durch Low Voltage Protection) geht, ist ein laden z.B. meiner Seplos Akkus nicht mehr möglich. Hier würde eine Möglichkeit den Fehler zu quittieren und zu ignorieren helfen um z.B. das Pack wieder laden zu können.

Der Fehler geht in dem Fall auch nicht mehr selbstständig weg da die Releasespannung nicht wieder erreicht wird ohne Ladung.

shining-man, 2024-10-03:
Vom Prinzip her ist es erst einmal gewollt, dass ich bei einem Error vom BMS alles auf 0 fahre. Sicher ist sicher. Normalerweise sollte es dazu gar nicht erst kommen.
Jetzt braucht es eine schlaue Möglichkeit um aus der Nummer wieder raus zu kommen. Eine Möglichkeit wäre ein Button im WebUI der für eine gewisse Zeit, das Laden zulässt. 
Mal sehen ob mir was einfällt was aber dennoch sicher ist.

SlapJackNpNp, 2024-10-03:
würd bei Low Voltage nicht eine Rampe sinn machen, die das Entladen nivht blockiert sondern langsam runter / hochfährt? Also z.B. bei 46V noch 5A in akku  bei 46.2 dürfen 15A geladen werden etc. Dann muss man auch nicht erst ans BSC und ne Taste in der Oberfläche drücken.

shining-man, 2024-10-04:
> würd bei Low Voltage nicht eine Rampe sinn machen, die das Entladen nivht blockiert sondern langsam runter / hochfährt? Also z.B. bei 46V noch 5A in akku bei 46.2 dürfen 15A geladen werden etc. Dann muss man auch nicht erst ans BSC und ne Taste in der Oberfläche drücken.

Das kannst du mit dem BSC alles machen bevor der Alarm vom BMS kommt. Du hast so viele möglichkeiten das alles zu regeln. Doch wenn das BMS einen Alarm/Fehler meldet, dann bedeutet das für mich das etwas schief läuft und da will ich das abgeschaltet wird. Normalerweiße sollte das BMS gar nicht erst in den Fehlerzustand kommen. Um aus dem Zustand rauszukommen muss ein eingreifen Notwendig sein. Ab dann sollte nichts mehr automatisch gehen. Keiner weiß doch was da so schief läuft.

tueftler0, 2024-10-04:
@shining-man 
Im Grunde ist das nur ein "Problem" bei Leuten, die ein FET-BMS besitzen.
Nur diese können bei einer low voltage protection mit einem Strom in Richtung Batterie, diese auch wieder aufladen.
Wenn ein Schütz auslöst muss man eh von Hand ran.

Was man machen könnte ist zu unterscheiden wo die Zellen sich befinden und was der Grund für die Protection war.
Das ist aber schon komplex dann, denn es gibt ja etliche Protection Events.
Beispiel "Unterspannung" -> Dann wäre es per GUI-Tastendruck ja okay, wenn man eine Ladung frei geben könnte, aber eine Entladung wäre nicht okay. Anders herum bei einer OVP.
Ich denke so komplex sollte man es nicht treiben ;)

shining-man, 2024-10-06:
@tueftler0 So komplex ist das eigentlich nicht. Wenn das das BMS Unterspannung meldet, dann kann ein Laden freigeschaltet werden.

Eigentlich sollte es zu dieser Situtaion gar nicht kommen. Dazu ist der BSC da und es kann eigentlich alles eigestellt werden um genau solche Situation zu vermeiden. Aber was tun, wenn man in solcher Situtaion ist. Wenn man dann das Laden kurz freigeben kann, dann wäre das evtl. schon hilfreich.
Ob es dir in so einer Situtaion wirklich hilft, stelle ich aber dennoch in Frage. Denn soetwas passiert, wenn keine Sonne da ist. Was bringt dir ein Laden pber den Wechselrichter? In so einem Fall muss ein Netzteil her, da braucht man auch nichts freischalten.

Wenn das Feature wirklich notwendig ist, dann gerne hier diskutieren.

tueftler0, 2024-10-06:
Klar, normal sollte man nicht in das Problem kommen. Aber wie ich gestern auch gesehen habe, entladen die Umrichter den Akku ganz gemächlich trotz alles auf 0A gesetzt. Das bekommt das BMS nicht mit und irgendwann kommt man dann trotz 10% SoC in die UVP.
Die Inverter haben Ladegeräte und könnten dann vom Netz laden und das Problem damit beseitigen.

Also ich persönlich finde die Idee nicht schlecht.

Saharel001, 2024-10-23:
> Vom Prinzip her ist es erst einmal gewollt, dass ich bei einem Error vom BMS alles auf 0 fahre. Sicher ist sicher. Normalerweise sollte es dazu gar nicht erst kommen. Jetzt braucht es eine schlaue [...]

Das passt für mich, damit kann jeder wenn er das "Problem" gelöst hat, das Pack wieder laden.

closed 2024-12-31 by shining-man