shining-man/bsc_fw #172: Alle 6 Temperaturen vom Seplos BMS V2 per API übertragen

Schatemuhl, 2024-10-14:
Ich würde es für sehr sinnvoll halten die 6 Temperaturen vom Seplos V2 per API zu übertragen.
Der Sensor 1-4 ist für die Zellen
Der Sensor 5 ist für die Temperatur im Gehäuse
Der Sensor 6 ist die Temperatur auf der Power Seite , also von den MosFet's

Sensor 5 und 6 sind nach meiner Meinung sehr wichtig, und sollten dann auch eien Rolle bei der Notabschaltung spielen.

[image]

shining-man, 2024-10-14:
Link zur Discord Diskussion: https://discord.com/channels/1037083515755434005/1295426778290196552

shining-man, 2024-12-31:
The change/extension is included in version 0.7.0.

closed 2024-12-31 by shining-man

Schatemuhl, 2024-12-31:
@shining-man 
Super, alle 6 Temperaturen kommen nun über die API.  Danke.