shining-man/bsc_fw #177: BMS Gobel Power

Maai-0815, 2024-10-18:
Hallo,
ich habe 2 Akkus von Globel Power, die ich gerne mit dem Tool verbinden möchte.
Leider bekomme ich eine Fehlermeldung.
Die Akkus haben das BMS GP-PC 100. 
Ich weiss, das das Tool für GP-PC 200 (Test) entwickelt wurde, aber ich verwende die gleiche Software wie die PC 200 als BMSTool für die Parametierung.
Aus diesem Grund dachte ich mir, das es funktionieren könnte.
Folgende Fehlermeldung erhalte ich:

E (2024-10-18 18:54:09) GOBEL_BMS_PC200: Timeout: Serial=2, u8_lRecvDataLen=255, u8_lRecvBytesCnt=0
E (2024-10-18 18:54:09) GOBEL_BMS_PC200: Timeout: Serial=2, u8_lRecvDataLen=255, u8_lRecvBytesCnt=0
E (2024-10-18 18:54:09) BSC_SERIAL: ERROR: device=3, reason=Checksum wrong

Liegt das Problem in der Software oder mache ich einen anderen Fehler?
Bin gespannt.
Marc

shining-man, 2024-10-18:
Mit welcher FW-Version vom BSC hast du es getestet? 
Hast du die richtige Adersse am BMS eingestellt?
Bekommst du von beiden BMS einen Tiemout, oder nur von einem?

Hast du ein Protokoll zu dem PC100? Dann könnte ich das einmal vergelichen mit dem PC200.

Maai-0815, 2024-10-18:
Ich habe das komplette Log-file angehängt.
[LogFile-BSC-20241018.txt](https://github.com/user-attachments/files/17440396/LogFile-BSC-20241018.txt)
Ich bin mir nicht sicher, was genau du mit richtiger Adresse meinst ....
Die Protokolle in der Config des BMS-Tools kann ich nicht lesen.
Ich habe in meinem BMS-Tool das Protokoll für Victron CanBus ausgewählt. Eine andere Einstellung kenne ich nicht.
Im Monitoring wird dieser Code für beide Akkus ausgegeben:
Host BMS S/N	Host Pack S/N	Host VER
216062635300078F'	GP-SR3-PC100L-240302'	P16S200A-21606-2.06

Hilft dir das weiter?

Maai-0815, 2024-10-19:
Ach so okay, ja die Dips habe ich gesetzt und nochmals kontrolliert.
Als einziges Dokument hierzu habe ich das folgende pdf bekommen.
[GP-SR3-PC100-Product Specification.pdf](https://github.com/user-attachments/files/17444875/GP-SR3-PC100-Product.Specification.pdf)

Maai-0815, 2024-10-21:
Hast du eine Idee woran es vielleicht noch liegen könnte, wenn du das Logfile anschaust?

shining-man, 2024-10-22:
> Ach so okay, ja die Dips habe ich gesetzt und nochmals kontrolliert. Als einziges Dokument hierzu habe ich das folgende pdf bekommen. [GP-SR3-PC100-Product Specification.pdf](https://github.com/user-attachments/files/17444875/GP-SR3-PC100-Product.Specification.pdf)

Das hilft nichts. Es bräuchte zumindest die Bezeichnung vom verbauten BMS um dann zu schauen ob man irgendwo ein Protokoll dazu findet. Evtl. hat schon jemand andere ein Tool o.ä. geschrieben um das BMS auszulesen. Evtl. kann man dort nach dem Protokoll fragen.

shining-man, 2024-10-22:
> Hast habt ihr eine Idee woran es vielleicht noch liegen könnte, wenn ihr euch das Logfile anschaust?

Das Log sagt da nicht viel. Er kann nicht mit dem BMS kommunizieren.
Ursachen kann es viele haben.
- Im OC100 ist ein anderes BMS. Das wäre die mindest Info die notwendig ist. Welches BMS steck im Pack. Zur not das pack zerlegen und schauen was auf der PCB steht. Irgendwie muss sich die Info ja finden lassen. Dann braucht es ein Protokoll dazu.
Falsche Adressen DIPs. Das erste BMS muss häufig Adresse 1 haben, zumindest bei den meisten Herstellern.
- Falsche Verkabelung. Verwendest du einen orginal BSC-HW oder ein NodeMCU?

Maai-0815, 2024-10-22:
Die Adresse an den Dips ist okay und laut Hersteller einegstellt. ( bei mir 1 an 1 und 2 an 2)
Ich habe zum Ausprobieren diese Komponenten verwendet: 
[image]
Auf die BMS habe ich gerade ein Software Update gemacht.

shining-man, 2026-01-17:
Ist hier noch etwas offen? Wird hier noch Unterstützung gebraucht?