shining-man/bsc_fw #178: Eigenes Profil für NEEY 10A active balancer (EK-24S10EB)

Svol0, 2024-10-19:
Der EK-24S10EB von NEEY unterstützt einen maximalen Balancestrom von 10A.
Aktuell (v0.7.0_9t) kann unter Bluetooth Schnittstellen nur der EK-24S4EB ausgewählt werden, welcher unter Bluetooth Device bei „**Max. Balance Current**“ „_nur_“ 8A als oberes Limit zulässt.
Da mit der EK-24S4EB Einstellung alle anderen Funktionen in Ordnung sind (lesen und schreiben der Einstellwerte an den NEEY) wäre es schön, wenn der EK-S10EB einen eigenen Eintrag in der Auswahl bei der Bluetooth Schnittstelle bekommt mit einem „**Max. Balance Current**“ von 10A als oberes Limit.

Gruß Svol0

shining-man, 2024-10-19:
Funktioniert beim EK-S10EB das Lesen der Einstellungen, wenn du den EK-24S4EB auswählst? Denn da gibt es beim EK-24S4EB noch Probleme.

Svol0, 2024-10-19:
Leider kann ich dieses noch nicht zu 100% bestätigen. Was ich eben auf die Schnelle testen konnte, war, das wenn ich am Neey direkt den Balancestrom auf einen Wert größer 8A stelle, scheint er sich nicht mehr mit dem BSC zu verbinden. Das kann aber eventuell daran liegen, dass ich zu ungeduldig war. Habe den Wert dann auf 7A reduziert und das BSC rebootet. Nach dem sich beide Neey wieder mit dem BSC verbunden hatten, wurde in der NEEY-Ansicht der geänderte Balancestrom Wert bei dem einen Balancer korrekt eingelesen, ohne dass ich den Read-Button betätigen musste.

[image]

Ausgiebige Tests kann ich leider erst ab dem 29.10. machen. Vorher habe ich keinen direkten Zugriff auf die Neey's.

shining-man, 2024-10-22:
Das ist sehr interessant und schaut gar nicht so schlecht aus. Ein extra Eintrag ist nicht das Problem.
Wenn du Testbereit bist, kannst du mal das Lesen und Schreiben der Settings genauer testen.

shining-man, 2024-10-28:
Das lesen der Einstellungen vom NEEY funktioniert leider nicht bei allen NEEYs automatisch. Manchmal braucht es da den Button.
Wenn man die Werte immer über diesen Weg einstellt, dann ist es eingetlich kein Problem, da dann die Daten passen. Es ist aktuell so, dass die Daten zuerst im BSC gespeichert sein müssen (S-Button), dass ich sie dann übertragen kann. 

Ein/Aus von "Balancer On" sollte dem Schiebeschalter im NEEY entsprechen. Wenn du Trigger x einstellst, dann muss der Trigger auch den entsprechenden Status haben. Ich schaue mir das noch einmal an. Da ich selber keinen 10A habe, müßte ich die evtl. eine FW-Version machen, die mir ein paar mehr Infos loggt. Wäre das ein Problem? Kannst du auch seriell loggen?

Svol0, 2024-10-28:
Ich hoffe, dass der Adapter für das serielle Loggen heute eintrifft. Werde dann Serial 0 frei machen zum loggen. Kannst mir gerne eine angepasste Version zukommen lassen (eventuell über Discord; ist der gleiche Benutzername ).

Zu „Balancer On“:
Habe es nicht mit den Triggern getestet, sondern mit der Auswahl „Ein“ und „Aus“. Damit konnte ich keine Veränderung der Einstellung am Neey erkennen. Werde das aber heute Abend noch mal testen. Dann auch mal mit Trigger.

Svol0, 2024-11-01:
@shining-man Leider konnte ich das mit den Triggern noch nicht weiter testen. Der 2te Neey hatte sich mit gleichem Fehlerbild verabschiedet. (4 x kurzes Piepen, per Bluetooth zwar als Device in der Enerkey App angezeigt, aber eine Verbindung nicht möglich.)
Daraufhin den anderen noch mal wieder unter Spannung gesetzt und siehe da, alles wieder in Ordnung. (War für ca. 4 Tage spannungslos) Dann den 2ten ausgebaut, Gehäuse geöffnet und die Kondensatoren händisch mit einem Widerstand entladen (unter 0,3V). Erneut angeschlossen und auch dieser arbeitet wieder wie er soll.
Im nachhinein muss ich dazu sagen, dass in beiden Fällen als das Problem aufgetreten ist, zuvor Werte per BSC in die Neey geschrieben wurden. Ich habe das beim ersten Neey nicht weiter in Verbindung gebracht, aber jetzt meine ich schon, dass der Ablauf gleich war.
Da der Akku jetzt hauptsächlich im unteren Bereich betrieben wird (dunkle Jahreszeit), wollte ich bei einem Akkupack ein Lowlevel Balancing durchführen. Dazu habe ich die Werte für "Sleep Voltage"  auf 2,550V und "Equalisation Voltage" auf 2,650V gesetzt und übertragen. Anschließend wurde der Neey nicht meht als connected angezeigt. Da ich erst knapp eine Woche später vor Ort war und dann das weiter oben beschrieben Fehlerbild vorfand hatte ich dieses nicht mit einander in Verbindung gebracht.
Bei dem 2ten Neey war es jetzt ähnlich. Ich wollte auch dort die Werte übertragen und anschließend wurde der Neey nicht mehr als connected angezeigt. Abends war ich dann vor Ort und hörte schon das piepen. Ob es jetzt wirklich mit dem schreiben der Werte zu tun hat, kann ich noch nicht bestätigen. Habe heute mehrfach versucht, dass ganze zu reproduzieren, ist mir aber nicht gelungen. (Habe dazu von der v0.7.0_t10a wieder auf die v0.7.0_t9 gewechselt, da diese bei beiden Fällen im Einsatz war).
Über Nacht läuft jetzt aber wieder die v0.7.0_t10a.
Ich gebe bescheid, wenn ich weitere Erkenntnisse habe. Solltest Du weitere Informationen benötigen, die hier noch fehlen, lass es mich bitte wissen.

shining-man, 2024-11-02:
Verstehe ich das richtig: Manchmal funtkioniert das schreiben der Werte, aber ab und zu verabschiedet der NEEY sich ins Nirwana?
Läuft bei dir die T10a genauso wie die T9?

Am besten wäre es, wenn man die orginale Kommunikation zwischen App und NEEY einmal belauschen würde. Es gibt zwei Ansätze:
- NEEY öffnen und entweder mit einem LogicAnalyser oder mit einem USB-UART Kabel die Kommunikation zwischen BT-Modul und Controller belauschen.
- Die BT-Kommunikation Sniffen

Svol0, 2024-11-02:
Es scheint zumindest so, dass der Neey sich gelegentlich nach dem Schreiben verabschiedet. In beiden Fällen konnte ich nicht sofort vor Ort sein und habe zuerst nur mitbekommen, dass nach dem Schreiben der Neey nicht mehr als connected angezeigt wurde und auch ein reboot des BSC keine Besserung brachte. Es ist halt bisher erst 2 mal vorgekommen und ich konnte es leider noch nicht reproduzieren.
Seit heute Nachmittag läuft das BSC wieder mit der v0.7.0_t9.
Der Grund dafür ist, dass es mit dem Neey an BT0 immer wieder zum Wechsel zwischen connected und disconnected kam und der Neey an BT1 sich nicht connected hat (so weit ich sehen konnte). Werde dieses morgen nochmal mit der 10a versuchen zu testen.
Hinzu kam, dass das Seplos BMS an Serial 1 nicht mehr angezeigt wurde. Habe daraufhin die Kabelverbindung zwischen beiden BMS umgesteckt, um zu kontrollieren, ob es eventuell am BMS liegt. Aber es wurde trotzdem nur das BMS an Serial 2 angezeigt. Auch nach einem reboot des BSC wurde das BMS an Serial 1 nicht angezeigt. Daraufhin habe ich das BSC spannungslos gemacht und wieder eingeschaltet. Nach dem Bootvorgang wurden beide BMS wieder angezeigt. 
Das Logfile bis kurz vor dem Rebootversuch habe ich angehängt.

Was mich bei den beiden Neey etwas stutzig macht ist, dass auf dem Gehäuse 10A (und wurden auch als 10A-Version verkauft) steht, auf der Platine aber EK-24S8EB aufgedruckt ist. Die Software dieses aber auch als EK-24S10EB anzeigt. (Screenshot ist von gestern Abend, nachdem ich den Neey „entladen“ und dann testweise wieder mit Netzteil in Betrieb genommen hatte). Daher sind keine Zellspannungen im Hintergrund sichtbar.

Zum Lauschangriff:
Das USB-UART Kabel wäre wahrscheinlich das gleiche, welches ich zur Zeit für das Serial Log verwende, oder?
Falls Du die entsprechenden Möglichkeiten hast, kann ich Dir sonst auch gerne einen der beiden Neey Leihweise inkl. Netzteil zur Verfügung stellen. Dadurch ist ein Betrieb ohne angeschlossenen Speicher möglich.

[v0.7.0_t10a_20241102.log](https://github.com/user-attachments/files/17608067/v0.7.0_t10a_20241102.log)

Neey im Fehlerzustand:
https://github.com/user-attachments/assets/36745bc2-d2b5-4ae2-adce-0162c58c6ff2

[image]
[image]

[image]

Svol0, 2024-11-02:
So wie es aussieht, handelt es sich bei dem verbauten Bluetooth Modul um ein JDY-18 (auch wenn EnerKey die Chipbezeichnungen entfernt).
Habe mir schon die Doku dazu besorgt.

Svol0, 2024-11-14:
@shining-man
Das Verabschieden der 10A NEEY scheint eine andere Ursache zu haben. Bin gerade über diesen Beitrag gestolpert:
https://akkudoktor.net/t/neey-10-a-piept-und-blinkt-rot-keine-bt-verbindung/19160
Den oben beschriebenen Zustand konnte ich bis jetzt auch nicht durch das Schreiben der NEEY Parameter über den BSC erreichen. Es war wohl purer Zufall, dass das Problem auftauchte, nachdem ich Parameter geändert hatte. 

Wenn ein stabiler Connect vom BSC zu den NEEY besteht, funktioniert das Lesen und Schreiben ohne Probleme.
Verwende noch die v0.7.0_T9 mit serial log.
Dort kommt es vor, dass die NEEY nach anfänglichem stabilen Connect nur noch als Disconnected angezeigt werden. Es hilft dann nur  ein Reboot des BSC um wieder einen Connect zu bekommen. Habe leider erst wieder Mitte nächster Woche Zugriff auf den Raspberry PI, der hoffentlich das Logfile weiterhin aufzeichnet und werde dir das Logfile dann zukommen lassen.

shining-man, 2024-11-15:
Interesant ist, dass er das BT-Modul (JDY-18) der alten Generation hat. Daher funktioniert auch das lesen/schreiben der Einstellungen. die neuen NEEYs haben ein anderes BT-Modul.
Bluetooth ist ein großes Problem und aktuell eine Groß-Baustelle.

Beim Schreiben funktioniert aber das On/Off noch nicht, oder hat sich das mittlerweile auch gelöst?

"Das USB-UART Kabel wäre wahrscheinlich das gleiche, welches ich zur Zeit für das Serial Log verwende, oder?" -> Ja