shining-man/bsc_fw #186: Zustand der Trigger merken und beim Reboot wieder setzten

Schatemuhl, 2024-10-31:
Wenn ein Trigger den Zustand HIGH hat und der BSC einen Neustart macht (gewollt oder nicht gewollt) sollten die Trigger nach dem Reboot wieder die vorherigen Zustände haben.

shining-man, 2024-10-31:
Wenn der Grund für den Trigger noch ansteht, dann haben sie auch wieder den alten Stand. 
Hier wäre der genaue Fall wo das nicht so ist interessant.

Schatemuhl, 2024-10-31:
Wenn der Trigger per vtrigger (MQTT) gesetzt wurde.

shining-man, 2024-10-31:
Die Trigger haben auch nach dem reboot den richtigen Zustand, da zyklisch alle Alarmrules abgearbeitet werden und die Trigger wider gesetzt werden. Da der vTrigger aber von außerhalb kommt, ist dieser nach einem Reboot weg und daher ist der Trigger wieder auf Low. Nach einem reboot muss also dafür gesorgt werden, dass der vTrigger über MQTT wieder gesetzt wird.

Schatemuhl, 2024-10-31:
das macht in meinen Augen keinen Sinn, das ich von extern schauen muss ob der BSC einen Neustart gemacht hat und ich dann erneut den vtrigger senden muss.
wir gehen mal davon aus das ich aus irgendwelchen Gründen, an meiner VISU den Button betätige "nicht Laden" oder " nicht entladen" dann will ich auch das ich mich darauf verlassen kann. Ist es nicht Möglich die Zustände der vtrigger in einen Speicher Bereich zu schreiben der nicht beim Start gelöscht wird ?

Schatemuhl, 2024-10-31:
Habe aber gerade das Problem gelöst in dem ich die vtrigger als "retain" sende, dann bekommt der BSC es gleich nach dem anmelden am MQTT mit und setzt diese sofort. damit kann ich leben.

shining-man, 2024-10-31:
Das aktuelle Release (v0.6.0) macht meinen Wissens keinen Reboot. Dies darf auch nicht sein. Der BSC muss stabil laufen und darauf muss verlass sein. 
Möglich ist es schon den vTrigger zu speichern. Doch sind hier meines erachten schon ein paar Gedanken vorher notwendig. Macht es wirklich Sinn den zu speichern? Ich persönlich würde erst einmal davon ausgehen, dass, wenn ich den BSC neu starte, alles Low ist und dann sich alles nach dem aktuellen Zustand wieder einstellt. Wenn ich z.B. die Anlage zu Wartungszwecken abschalte, und der vTrigger vorher High war, dann will ich nicht, dass er hinterher immer noch High ist. Ich denke das ist auch eine Betrachtungssache. Ich bin immer bereit Features zu erweitern, aber gewissen Dinge müssen auch erst Diskutiert werden.

shining-man, 2024-10-31:
Was man vielleicht noch machen könnte ist, dass man 5 Remanente vTrigger macht und 5 normale? So könnte jeder die benutzen, die er gerade braucht. Vielleicht haben auch andere Ideen dazu?

Schatemuhl, 2024-10-31:
Ja damit könnte ich auch gut leben.
Mal sehen was der Rest so meint.
Ich danke Dir auf jeden Fall für dein offenes Ohr.

tueftler0, 2024-11-01:
Oder man definiert eigene vTrigger, denen man dann wahlweise per Remanent-Checkbox diese Möglichkeit verleihen kann.
Wobei man das auch bei den normalen Triggern integrieren könnte.

shining-man, 2024-11-01:
Die normalen Trigger brauchen nicht remanent sein, da sie sekündlich den Zustand der Alamrules etc. auswerten. Die sind immer aktuell, auch nach einem reboot.

Schatemuhl, 2024-12-06:
Gibt es hier schon Ansätze wie man gewisse trigger remanent setzen kann, wenn es der MQTT Client nicht kann ?
Jetzt in der Winterzeit schalte ich den Entladestrom gern per Trigger auf "0", problem ist aber wenn der BSC neu startet ist der Trigger wieder false und zieht die Batterie leer. 
Mein MQTT Client kann momentan die retain Funktion noch nicht, deswegen die Frage ob da was in Planung ist.

shining-man, 2024-12-07:
Der Ansatz wird der RTC-Speicher sein. Der bleibt erhalten, so lange keine Spannung weggenommen wird. Bei der Lösung über den RTC-Speicher, bleibt gegenüber dem Speichern im Flash, die Gefahr, das ein sehr schwerwiegender Kernel Panic evtl. den RTC Speicher beschädigt und die remanenz dann nicht gegeben ist. Dies sehe ich aber erst einmal nicht als Problem. Schreibe ich es direkt in dem Flash (z.B: NVS), kann es unter umständen zu einer höheren Schreiblast im Flash kommen, das will ich unterbinden. 

Neben den vTriggern wird es dann sowas wie rTrigger geben, die dann remanent sind, oder ich mache es per Checkbox.

tueftler0, 2024-12-30:
@Schatemuhl Kannst du einmal testen

shining-man, 2024-12-31:
The change/extension is included in version 0.7.0.

closed 2024-12-31 by shining-man

Schatemuhl, 2024-12-31:
@tueftler0  ja, hab die Version gerade installiert, teste und werde berichten.