shining-man/bsc_fw #188: Victron Smart Shunt - Falsche Sklierung der Werte über MQTT

TheSmartGerman, 2024-11-17:
In der LiveView übersicht werden die richtigen Werte angezeigt, über MQTT wird der Wert falsch übermittelt. Es muss um Faktor 100 geteilt werden. Umrechung A / mA wäre Faktor 1000

Im Vergleich zum BMS sollte jeweils die geleiche Skalierung übermittelt werden. 

[image]
[image]
[image]
[image]

shining-man, 2024-11-18:
Es gibt zu dem Problem schon das Issue #174

closed 2024-11-18 by shining-man

TheSmartGerman, 2024-11-18:
ups, sorry übersehen... alles klar!