shining-man/bsc_fw #216: Erweiterung der Funktion Quelle Gesamtspannung und Quelle Gesamtstrom auf mehrere DataDevices

D3R-ST3FAN, 2025-01-14:
Analog zur neuen Funktion Quelle Gesamtspannung und Quelle Gesamtstrom wäre es sinnvoll hier ebenfalls optional aus mehreren Datanquellen/DataDevices auswählen kann.

Ähnlich ist das ja bereits bei der SOC Funktion umgesetzt.

Zusätzlich wäre ebenfalls eine Value Handling für Gesamtstrom (Minimalwert, Mittelwert, Maximalwert + neu Summe)
bzw. Gesamtspannung (Minimalwert, Mittelwert, Maximalwert) notwendig.

Anwendungsbeispiel: mehrere Batteriepacks mit jeweils einem Smartshunt.

D3R-ST3FAN, 2025-01-19:
Funktioniert jetzt mit der v0.7.3_t2 wie gewünscht. Vielen Dank.
Kurze Rückfrage: 
Die "neuen" Werte für Gesamtspannung und Gesamtstrom werden die nur zum Inverter weitergeleitet oder auch für andere Strom/Spannungsabhängige Funktionen benutzt?

shining-man, 2025-01-19:
Der Strom wird noch für den ChargeCurrentCutOff genutzt. 
Die Spannung wird im Bereich des Autobalancer genutzt.

D3R-ST3FAN, 2025-03-20:
Danke für die Umsetzung

closed 2025-03-20 by D3R-ST3FAN