shining-man/bsc_fw #221: Trigger für 0% SOC bei Wechselrichter & Laderegelung

D3R-ST3FAN, 2025-01-21:
Analog zu den Einstellungen bei Wechselrichter & Laderegelung

"SOC auf 100 " bei --> Trigger

solte es eine gleiche Möglichkeit für

"SOC auf 0 " bei --> Trigger

geben. 
Das wäre dann für mich konsistent.

D3R-ST3FAN, 2025-01-24:
Beides ;-)

Ist für mich irgendwie inkonsequent wenn es einen fürs 100 % triggern gibt.

Usecase, wäre per Trigger ein Nachladen zu veranlassen (Strompreis...whatever).
Sicherlich ähnlich, wie der Usecase für 100%.

shining-man, 2025-01-24:
Ich sehe da eigentlich nichts Inkonsequentes oder inkonsistentes. Nur weil die eine Richtung vorhanden ist, muss doch die andere nicht auch vorhanden sein. Es muss auch Sinn ergeben und benötigt werden. Ich sehe es eher als konsequent auf unnötiges zu verzichten ;-)

Mit einem Trigger auf einen SoC von 100%, wird dem Wechselrichter signalisiert, dass er nicht mehr weiter laden braucht, da voll.
Ein Trigger auf 0% birgt bei Fehlkonfiguration die Gefahr, dass der Wechselrichter laden soll, obwohl er vielleicht schon bei 100% ist. Daher gibt es nur Trigger auf SoC 100% und auf Lade/Entladestrom 0A.

D3R-ST3FAN, 2025-12-24:
Ich halte es zwar immer noch für Sinvoll, da ja die anderen Schutzmechanismen ja weiter greifen sollten, aber aus Hygenegründen kann das dann hier wohl zu.

D3R-ST3FAN, 2025-12-25:
Vielleicht interessiert es ja wirklich niemanden.
Ich habe leider mein Smartmeter noch nicht, sonst könnte ich es jetzt schon (sinnvoll) nutzen.

Aber evtl. lesen hier auch nicht so viele mit.
Konnte man bei Discord nicht irgendwie einen Kanal mit den Github issues einblenden, der automatisch postet?

Dann ürden das vielleicht mehr leute lesen.

shining-man, 2025-12-25:
> Konnte man bei Discord nicht irgendwie einen Kanal mit den Github issues einblenden, der automatisch postet?

Das ist keine schlechte Idee. Ich schaue einmal wie gut sich sowas umsetzen lässt.

closed 2025-12-27 by D3R-ST3FAN