shining-man/bsc_fw #224: BSC not calculating Total Current correctly

sticilface, 2025-02-05:
[image]

From this picture it shows the MQTT messages from the BSC.  
It correctly shows the current from the 2 packs... 
But it does not add them correctly for total current.  

-62.36 + -24.89 does not equal -61.60... 

Using 2 JKBMSs PB2A16S20P

Thoughts...

shining-man, 2025-02-05:
Can you take a screenshot of the corresponding inverter settings?
And preferably the output of the complete /restapi

D3R-ST3FAN, 2025-02-05:
Assuming you are using the latest sponsor support version.
Have you checked current aggregation method and also selected both devices for current ?

sticilface, 2025-02-05:
> Can you take a screenshot of the corresponding inverter settings? And preferably the output of the complete /restapi

Happy to give the output of the rest API... how is this done?

sticilface, 2025-02-05:
> Assuming you are using the latest sponsor support version. Have you checked current aggregation method and also selected both devices for current ?

I one-time sponsored, not reoccurring which I don't believe give me access to the sponsor version.

D3R-ST3FAN, 2025-02-05:
> > Can you take a screenshot of the corresponding inverter settings? And preferably the output of the complete /restapi
> 
> Happy to give the output of the rest API... how is this done?

very simple and might be usefull in the future.
http://ip-of-your-bsc/restapi

Great you fixed it.

shining-man, 2025-02-05:
The master source should not be selected again under "+ Data sources." If it is selected twice, the software ignores the duplicate selection.

closed 2025-02-05 by sticilface