shining-man/bsc_fw #225: Laderegelung zur Akkuschonung

Ximerox, 2025-02-19:
**Ziel:** Akku nur bis definierten SoC laden (Schonung) und trotzdem in definierten Abständen auf 100% zum Balancen

**_INFO:_** Teilweise ist dies bereits möglich (über "Ladestrom reduzieren - SoC"), aber dies wird in der aktuellen offiziellen Firmware nicht durch die Einstellungen zum Balancen übersteuert, so dass es nicht zum Balancen kommt. Unter Victron-Systemen mit Laderegler funktioniert dies auch nicht, da die Laderegler Spannungsgeführt sind und trotz einem CCL von 0A weiterhin in den Akku laden.

**Lösungsansatz:** 
- Bereich "Ladestrom reduziere - SoC" ergänzen um: [de/aktivieren] von "Ladespannung anpassen". Diese Funktion müsste dann langsam und schrittweise die Ladespannung so reduzieren/anpassen, dass möglichst wenig in den Akku geht (0A geht nicht)

**Logik** _(von mir in NodeRed im Cerbo erprobt):_
Werte:
- Gesamtleistung am Akku anliegend - davon die jeweils letzten 10 Werte gemittelt
- Aktueller SoC
- Aktuelle setpoint_cv
- Korridor: 600W (passt bei mir, evtl muss dieser Wert einstellbar sein)

If SoC >= eingestellter Ziel-SoC UND autobal_state <=1 {
  Wenn Leistung > Korridor setze setpoint_cv - 0,1V
  Wenn Leistung < -Korridor setze setpoint_cv + 0,1V
}
Else {
   mach das was das BSC sonst vorhat :)
}

warte 15s und prüfe erneut (wenn man die Schleife zu schnell laufen lässt, dann gibts ne Katastrophe (ich kann berichten :)))

tueftler0, 2025-02-19:
Evtl könnte es auch eine einzelne Definition des gewollten SoC geben.
Denn die von Dir nun genutzte Funktion ist ja dazu da, am Ende des Bereiches den Ladenstrom zu reduzieren und erst wieder bei x% zu zu lassen.
Wenn ich nun z.B. auf 80% immer laden möchte, dann möchte ich auch kurz vor diesem Punk den Ladestrom reduzieren und nicht abrupt ab schalten. Auch möchte ich, dass erst bei 75% z.B. nachgeladen wird.
Daher denke ich, wäre eine einzelne Definition evtl auch eine Idee.
Grundsätzlich ist Dein Vorhaben aber natürlich wirklich sinnvoll und würde ich auch begrüßen/nutzen.

Fuendiger, 2025-03-15:
Es gibt hierfür jetzt die Funktion "Spannungsregelung zur Ladestrombegrenzung". Damit wird ab einem bestimmten SoC die Spannung so ausgeregelt, dass der Strom sich in dem eingestellten Korridor befindet.  
https://bsc-org.github.io/bsc/settings_inverter/#spannungsregelung-zur-ladestrombegrenzung

closed 2025-03-15 by Fuendiger