shining-man/bsc_fw #230: Log C/G Infos in S/R ändern

tueftler0, 2025-02-25:
Es wäre allgemein gängiger die Fehlerzusammenfassung mit dem Prefix **C**oming/**G**oing in **S**et/**R**eset abzuändern.

shining-man, 2025-02-25:
S/R finde ich in dem Zusammenhang nicht gängiger und irgendwie auch nicht logisch. Denn einen Fehler setze ich nicht, der tritt auf. Daher finde ich ein C/G eigentlich logischer, da es genau das beschreibt und so kenne ich das auch aus anderen Systemen.  
Man könnte noch F/O machen. F für Failure kommt und O für OK/over. Oder F/R für Failure/Resolved.

tueftler0, 2025-02-25:
Okay. F/R wäre auch passend.
Es ist mit einem Buchstaben irgendwie nicht so übersichtlich.
Evtl könnte man auch darüber nachdenken z.B. "Fail" / "Res" zu verwenden.

shining-man, 2025-03-14:
In der T0.8.0_T7 geändert auf:
- IMP (Impuls) für eine einmalige Meldung
- SET für einen kommenden Fehler
- RES für einen gehenden Fehler

closed 2025-03-14 by shining-man