shining-man/bsc_fw #232: Seplos BMS V16 (10E) keine Entladung möglich

RRonGit, 2025-03-06:
Der BSC läuft mit der Hardware 2.5 und der Firmware v0.7.4
Der BSC liest zwei Seplos Akkus aus. Seplos BMS V16 (10E) Software Version: 16.6

Habe folgendes festgestellt:

Wenn das Seplos BMS ein Überspannung registriert, lässt der BSC keine Entladung mehr zu obwohl das Seplos BMS den Entladeschalter freigegeben hat. 

Screenshots dazu:

_**_BSC Restapi:_**_

{
  "system": {
    "fw_version": "V0.7.4",
    "fw_add": "",
    "hw_version": "2",
    "name": "bsc01"
  },
  "trigger": {
    "0": 0,
    "1": 0,
    "2": 1,
    "3": 0,
    "4": 0,
    "5": 0,
    "6": 0,
    "7": 0,
    "8": 0,
    "9": 0
  },
  "inverter": {
    "current": 0,
    "voltage": 54.66,
    "soc": 100,
    "cv_state": 1,
    "setpoint_cv": 55.2,
    "setpoint_cc": 0,
    "setpoint_dcc": 0,
    "cc_cellVoltage": 0,
    "cc_soc": 0,
    "cc_cellDrift": 0,
    "cc_cutOff": 0,
    "dcc_cellVoltage": 0,
    "autobal_state": 0
  },
  "data_device": [
    {
      "name": "Seplos_1",
      "en": 1,
      "valid": 1,
      "nr": 0,
      "cells": 16,
      "totalVolt": 54.71,
      "totalCurr": 0,
      "soc": 100,
      "maxCellDiff": 3,
      "maxCellVolt": 3421,
      "minCellVolt": 3418,
      "maxCellVoltNr": 1,
      "minCellVoltNr": 2,
      "balOn": 0,
      "FetState": 2,
      "bmsErr": 4,
      "cell_voltage": [3419, 3421, 3418, 3418, 3419, 3419, 3419, 3418, 3418, 3419, 3419, 3419, 3421, 3420, 3420, 3420],
      "temperature": [20.6, 20.1, 19.7, 20.8, 22.7, 18.4]
    },
    {
      "name": "Seplos_2",
      "en": 1,
      "valid": 1,
      "nr": 1,
      "cells": 16,
      "totalVolt": 54.66,
      "totalCurr": 0,
      "soc": 100,
      "maxCellDiff": 5,
      "maxCellVolt": 3418,
      "minCellVolt": 3413,
      "maxCellVoltNr": 14,
      "minCellVoltNr": 9,
      "balOn": 0,
      "FetState": 2,
      "bmsErr": 4,
      "cell_voltage": [3416, 3417, 3414, 3417, 3417, 3416, 3417, 3417, 3416, 3413, 3417, 3417, 3416, 3417, 3418, 3418],
      "temperature": [20.7, 20.4, 20.29, 21.5, 23.2, 18.9]
    },

**_BSC:_**

[image]

[image]

**_Seplos BMS:_**

[image]

**_Trigger:_**

[image]

[image]

[image]

[image]

Als ich die Gesamtspannung unter 54V brach gingen die Seplos Warnungen weg und der BSC lies wieder Ladung und Entladung zu.

Gerade bei Überspannung wäre es aber gut, dass das Entladen noch gehen würde, damit die Spannung runterkäme und die Warnungen im Seplos BMS gelöscht werden.

Fuendiger, 2025-03-06:
Das Thema wurde in der Vergangenheit schon öfters diskutiert.
Der Grundgedanke des BSC ist hierbei immer die Sicherheit. Wenn das BMS einen Fehler meldet, dann stimmt etwas nicht mehr und das Laden/Entladen wird unterbunden. Denn in dem Moment weiß keiner was los ist. Es sollte nie dazu kommen, dass das BMS einen Fehler meldet, denn das läst sich alles über die Laderegelungen des BSC machen.
Daher wäre die Empfehlung erst einmal zu betrachten, warum es überhaupt zu der Überspannung kommen konnte?
Sind die Regelungen im BSC passend eingestellt?

RRonGit, 2025-03-06:
@Fuendiger Danke für deine Info.

Die Überspannung wurde bewusst provoziert um zu schauen wie sich das System verhält. Die Regelungen / Settings im BSC passen, das System läuft schon fast ein Jahr ohne das es jemals zu einer Abschaltung durch das BMS gekommen ist.  Wenn es mal zu einer Abschaltung durch das BMS kommt, ist es halt relativ aufwendig das System wieder zum Entladen zu bringen. Da wäre in diesem Fall ein Button im BSC hilfreich den man betätigen könnte um die Entladung kontrolliert frei zu geben.

Fuendiger, 2025-03-07:
So einen Button gibt es für die andere Richtung:  
[image]
Den könnte man erweitern, dass er beide Fälle abdeckt.

RRonGit, 2025-03-15:
Die Lösung mit der Erweiterung / Ergänzung des bestehenden Button würde das Problem lösen. So könnte man manuell / kontrolliert den Akku entladen.

Fuendiger, 2025-03-30:
@RRonGit Der Button ist in dem aktuellen Teststand (Insider) enthalten. 
Mit Verzögerung kommt es dann auch in die freie Version.

closed 2025-03-30 by Fuendiger