shining-man/bsc_fw #233: BSC Address Conflict with PACE BMS Batteries on RS485 (No DIP Switches to Set Addresses)

martinarva, 2025-03-11:
### **Issue: BSC Address Conflict with PACE BMS Batteries on RS485 (No DIP Switches to Set Addresses)**  

#### **Description**  
I'm trying to connect a **BSC (Battery Safety Controller)** to my **Hysone 4x10kWh batteries with PACE BMS** over **RS485**, but I’m encountering an addressing issue due to the lack of DIP switches to manually configure addresses.  

- Each battery **defaults to Address 1** when powered on.  
- When **daisy-chained**, they **correctly auto-assign addresses**:  
  - **Battery 1 → Address 1**  
  - **Battery 2 → Address 2**  
  - **Battery 3 → Address 3**  
  - **Battery 4 → Address 4**  
- **BSC also defaults to Address 1**, which conflicts with Battery 1 and causes data corruption.  
- **I can't manually change the battery addresses** because the PACE BMS doesn’t have DIP switches or documented commands for address reassignment.  
- **If I disconnect the communication cables between batteries**, each battery will likely remain at Address 1, making it impossible to communicate with all batteries using a single RS485 interface.  
- **I tried connecting only Battery 1 to BSC**, expecting it to take Address 2 dynamically, but it remained at Address 1.  

#### **Expected Behavior**  
BSC should be able to communicate with all batteries in a daisy-chain setup **without address conflicts**.  

#### **Potential Solutions (Need Confirmation)**  

1. **Can BSC be set to Address 0 (Master Mode)?**  
   - If BSC could act as **Address 0**, it might be able to poll **Addresses 1-4** dynamically.  
   - Not sure if the current firmware supports this.  

2. **Can BSC force itself as the Master?**  
   - There is no clear documentation on how PACE BMS determines the master.  
   - Some BMS systems use **UP/DOWN port connections** to set address hierarchy—does PACE BMS follow this logic?  

3. **Is there a way to manually override battery addresses?**  
   - Since there are no DIP switches, is there a software method to set addresses?  

4. **Any known workarounds to avoid multiple RS485 interfaces?**  
   - If I disconnect battery communication cables, all batteries would stay at Address 1, making a single RS485 interface ineffective.  

Would appreciate any insights or possible firmware modifications to resolve this issue.

Fuendiger, 2025-03-12:
The BSC does not have an address. The protocol structure does not provide for that at all. What are you basing your assumption on that it would require an address? I am not aware of any protocol that describes it that way.

The structure of the request is as follows (with example data):

```
7E (SOI)  
32 35 (**VER, that is version 25H, V2.5**)  
30 30 (**ADR**)  
34 36 (CID1, 46H)  
34 32 (CID2, 42H)  
45 30 30 32 (LENGTH, E002, LENID is 002H, DATAINFO length is 2, LCHKSUM is EH)  
46 46  INFO FFH  
46 44 30 36 (CHKSUM, FD06)  
0D (EOI)  
```

"ADR" is the address of the battery from which I am retrieving the data.

It is important that you use RS485B; otherwise, the response will follow an undocumented protocol. You can see that it is returned as Protocol V2.5.

What's not entirely clear to me is the "INFO" byte. I also transmit the address there, which works. Perhaps one could also transmit a 0xFF here.

What I can offer is a version that logs serially (but not into the logfile) and outputs the response from the BMS, if it responds at all.
Do you have the ability to perform serial logging? (https://bsc-org.github.io/bsc/troubleshooting/#serial-debugging)

martinarva, 2025-03-12:
> The BSC does not have an address. The protocol structure does not provide for that at all. What are you basing your assumption on that it would require an address? I am not aware of any protocol [...]

I don't have the official HW yet and using the Lilygo T-CAN485 for testing at the moment.

Here is my current data
Serial  2 (TCAN default) and adresses are 1,2,3,4

[image]

time to time some of the values does make sense, but hey rotate and change all the time

five minutes of the logs as well

[four batteries defined.txt](https://github.com/user-attachments/files/19206257/four.batteries.defined.txt)

When i define only one battery on address 1 then i get a stable data:

[image]

Also the logs are clean now. So when using only one battery adress 1 then it works.

So what i'm thinking is that the master battery on adress 1 is polling the other batteries for all the data already and BSC is doing the same but does not recognize the answers that slave batteries are giving to the master battery.

But to confirm this we should log what happens in the serial. How i can to this with the Lilygo T-CAN485?

shining-man, 2025-03-12:
> But to confirm this we should log what happens in the serial. How i can to this with the Lilygo T-CAN485?

You need to swap the DDEBUG_ON_FS and DDEBUG_ON_HW_SERIAL switches under build_flags and build_unflags.
  
> EDIT: One idea. If the master battery is polling the other batteries anyway for basically all the data needed for the BSC, would it be possible for BSC just to listen the serial feed and pick up whats needed?

This is a problem, since the BSC only listens for a response after it has sent. This isn't easily changed.

Fuendiger, 2025-03-12:
> But to confirm this we should log what happens in the serial. How i can to this with the Lilygo T-CAN485?

You need to swap the DDEBUG_ON_FS and DDEBUG_ON_HW_SERIAL switches under build_flags and build_unflags.

> EDIT: One idea. If the master battery is polling the other batteries anyway for basically all the data needed for the BSC, would it be possible for BSC just to listen the serial feed and pick up whats needed?

This is a problem, since the BSC only listens for a response after it has sent. This isn't easily changed.

martinarva, 2025-03-13:
> > But to confirm this we should log what happens in the serial. How i can to this with the Lilygo T-CAN485?
> 
> You need to swap the DDEBUG_ON_FS and DDEBUG_ON_HW_SERIAL switches under build_flags and build_unflags.
> 

Here are the serial logs:

1 battery on adress 1 defined - 

[one battery on addres 1 defined.txt](https://github.com/user-attachments/files/19223846/one.battery.on.addres.1.defined.txt)

4 batteries on adresses 1,2,3,4 defined - 

[4 batteries defined.txt](https://github.com/user-attachments/files/19223850/4.batteries.defined.txt)

Fuendiger, 2025-03-17:
The data is very strange. There is always a different amount of data.
Can you make a recording of the data when the PC software is connected to the RS485B of the BMS? You can use another RS485 adapter to record the communication.

mmoe-milano, 2025-03-18:
Hello, I run 3xGobel GP-SR1-PC200 in a 'Daisy Chain' configuration without any problem. As @Fuendiger already said you have to start the Switch Setup on the front panel of each pack with address 2. Be aware that this Switch Setup must be binary coded! The RS485 communication cable between BSC and the first pack must be plugged into RS-485B of pack with address 2. From there you have to wire the daisy chain communication cables as usual to the other packs. 

[image]

shining-man, 2025-03-18:
@mmoe-milano The problem is that his version of the PC200 is not equipped with the DIP switches. This is where the problems come from.

martinarva, 2025-03-19:
> The data is very strange. There is always a different amount of data. Can you make a recording of the data when the PC software is connected to the RS485B of the BMS? You can use another RS485 adapter to record the communication.

You mean that i don't connect the Lilygo with BSC firmware but just some rs485 to USB and monitor the traffic on the RS-485B port of the first/master battery?

Fuendiger, 2025-03-19:
> You mean that i don't connect the Lilygo with BSC firmware but just some rs485 to USB and monitor the traffic on the RS-485B port of the first/master battery?

I mean that you are listening to the communication with the PC software and the BMS. I would like to see what the software sends to the BMS. 
Are the recordings between the PC software and the BMS?

I am surprised that only 3 BMS responded in the first recording. The requests to the BMS look like mine, with one small difference. I will make you a version to try something.

martinarva, 2025-03-21:
Would it be possible to connect battery to BSC via rs232? To my knowledge the protocol on the rs485b which is used in the daisy chain between batteries is not so standard and at least with my battery the issue is that master battery is already polling other batteries and this traffic will collide with BSC that is also polling all the batteries. As I cannot physically change dip switches so that all four batteries are slave, the issue will be there, unless we find out a way how to just monitor the serial for the batteries 2,3,4 and poll only battery 1.

Protocol on the RS232 port of the battery should be more universal and is used by many Chinese batteries like seplos, pace etc. There is a computer software BmsTools where you can get all four batteries data if they are daisy chained. I have a RS232 to USB dongle and i connect to adress 1 with BmsTools.

So my question -  wold there be a possibility to connect via RS232 port instead of the RS485B port, which is mainly for daisy chain?

martinarva, 2025-03-21:
Looks like there are two more polls from BMSTools to the rs232 serial:

```
9378 21/03/2025 10:30:09 IRP_MJ_WRITE UP STATUS_SUCCESS 7e 32 35 30 33 34 36 43 31 30 30 30 30 46 44 39 38 0d  ~250346C10000FD98. 18  COM4  
9386 21/03/2025 10:30:09 IRP_MJ_READ UP STATUS_SUCCESS 7e 32 35 30 31 34 36 30 30 36 30 32 38 35 30 33 31 33 36 35 33 33 32 33 30 33 30 34 31 32 44 33  ~25014600602850313653323030412D3 32  COM4  
9396 21/03/2025 10:30:09 IRP_MJ_READ UP STATUS_SUCCESS 33 33 31 33 39 33 30 33 39 32 44 33 31 32 45 33 30 33 31 32 30 46 35 37 46 0d  3313930392D312E303120F57F. 26  COM4  

9400 21/03/2025 10:30:10 IRP_MJ_WRITE UP STATUS_SUCCESS 7e 32 35 30 33 34 36 43 32 30 30 30 30 46 44 39 37 0d  ~250346C20000FD97. 18  COM4
9408 21/03/2025 10:30:10 IRP_MJ_READ UP STATUS_SUCCESS 7e 32 35 30 31 34 36 30 30 42 30 35 30 33 35 33 30 33 34 33 30 33 30 33 30 33 34 33 36 33 32 33  ~25014600B0503530343030303436323 32  COM4  
9418 21/03/2025 10:30:10 IRP_MJ_READ UP STATUS_SUCCESS 34 33 30 33 33 33 30 33 38 34 31 33 30 33 30 33 32 30 30 30 30 33 35 33 30 33 34 33 30 33 30 33  43033303841303032000035303430303 32  COM4  
9428 21/03/2025 10:30:10 IRP_MJ_READ UP STATUS_SUCCESS 30 33 34 33 36 33 32 33 34 33 30 33 33 33 30 33 38 34 31 33 30 33 30 33 32 30 30 30 30 45 44 44  03436323430333038413030320000EDD 32  COM4  
9438 21/03/2025 10:30:10 IRP_MJ_READ UP STATUS_SUCCESS 42 0d  B. 2  COM4  
```

Different CID2 - 43 31 and 43 32 instead of 34 34 and 34 32

Fuendiger, 2025-03-27:
This is not possible via RS232. The BSC wants to query each battery directly. This is intentional, as I do not want any collected data. The BSC is the “boss” and wants direct contact with every battery. This is only possible via RS485. As this is not possible in your case, you can either try to equip the DIP switches or connect each BMS individually to the BSC.

martinarva, 2025-03-27:
> This is not possible via RS232. The BSC wants to query each battery directly. This is intentional, as I do not want any collected data. The BSC is the “boss” and wants direct contact with every [...]

But you can query each battery direclty from RS232 when batteries are daisychained together. BSC can be the boss and query each battery like wheh it's connected via RS485B port. Here is example of RS232 queries: 

battery 1 - 7e 32 35 **30 31** 34 36 43 32 30 30 30 30 46 44 39 39 0d
battery 2 - 7e 32 35 **30 32** 34 36 43 31 30 30 30 30 46 44 39 39 0d
battery 3 - 7e 32 35 **30 33** 34 36 43 31 30 30 30 30 46 44 39 38 0d
battery 4 - 7e 32 35 **30 34** 34 36 43 31 30 30 30 30 46 44 39 37 0d

Fuendiger, 2025-03-27:
I don't think I can query each battery directly, because I only have a connection to the first pack via RS232. All further communication is then handled by the first pack with the others. Does the first pack actually give me the information from the other packs, or is it modified by the first pack? Other BMS manufacturers have also encountered this problem, with the master pack filtering out important data.

> battery 1 - 7e 32 35 30 31 34 36 43 32 30 30 30 30 46 44 39 39 0d
> battery 2 - 7e 32 35 30 32 34 36 43 31 30 30 30 30 46 44 39 39 0d
> battery 3 - 7e 32 35 30 33 34 36 43 31 30 30 30 30 46 44 39 38 0d
> battery 4 - 7e 32 35 30 34 34 36 43 31 30 30 30 30 46 44 39 37 0d

If these are the requests, how do they get to the respective battery? Why can't I do this directly via RS485? I suspect it's a collective query and not a direct 1:1 query. The whole thing is still a bit unclear to me, and there are many unanswered questions.
- What happens in the first battery when the request is made via RS232?
- How does it query the other batteries?
- What happens to the data in the master battery?
- Will I get the data back unchanged?

Such an implementation involves a lot of work, and everything needs to be clarified beforehand. So far, I don't see it as a good or clean solution, but I'm happy to be convinced otherwise.

martinarva, 2025-03-28:
I used PBMTools PC software that was connected to the main battery RS232 port. Then i used a software called Serial Port Monitor to monitor the serial traffic between BMSTools and master battery connected via RS232 port.

Now just  connected my PC to via USB to RS232 to the main battery RS232 port and used Serial Port Monitor software to send my own commands. There is no other traffic on the RS232, so only commands i send i get answers. 

- What happens in the first battery when the request is made via RS232?
  - there is sa response accoringly
- How does it query the other batteries?
  - See the sample below. To my understandind 99% same as over RS485B
- What happens to the data in the master battery?
  - I dont understand the question
- Will I get the data back unchanged?
  - yes

Battery1 request:
```
7E 32 35 30 31 34 36 34 32 45 30 30 32 30 31 46 44 33 30 0D
```

Battery 1 response
```
7E 32 35 30 31 34 36 30 30 31 30 30 30 31 31 30 30 43 46 36 30 43 46 35 30 43 46 34 30 44 30 35 30 44 30 33 30 43 46 35 30 43 46 34 30 43 46 46 30 44 30 32 30 43 46 33 30 43 46 33 30 44 30 35 30 44 30 33 30 43 46 35 30 43 46 36 30 43 46 35 30 36 30 42 45 33 30 42 45 30 30 42 45 31 30 42 44 37 30 42 45 30 30 42 45 31 31 34 45 35 43 46 39 46 31 35 36 32 30 33 34 45 45 31 30 31 31 36 35 31 34 30 31 42 35 31 34 30 35 31 34 30 35 31 34 30 36 33 43 46 37 42 30 30 30 30 44 43 43 35
```

Battery 2 request:
```
7E 32 35 30 32 34 36 34 32 45 30 30 32 30 32 46 44 32 45 0D
```

Battery 2 response
```
7E 32 35 30 31 34 36 30 30 31 30 30 30 31 31 30 30 43 46 36 30 43 46 35 30 43 46 34 30 44 30 35 30 44 30 33 30 43 46 35 30 43 46 34 30 43 46 46 30 44 30 32 30 43 46 33 30 43 46 33 30 44 30 35 30 44 30 33 30 43 46 35 30 43 46 36 30 43 46 35 30 36 30 42 45 33 30 42 45 30 30 42 45 31 30 42 44 37 30 42 45 30 30 42 45 31 31 34 45 35 43 46 39 46 31 35 36 32 30 33 34 45 45 31 30 31 31 36 35 31 34 30 31 42 35 31 34 30 35 31 34 30 35 31 34 30 36 33 43 46 37 42 30 30 30 30 44 43 43 35
```

Battery 3 request:
```
7E 32 35 30 33 34 36 34 32 45 30 30 32 30 33 46 44 32 43 0D
```

Battery 3 response
```
7E 32 35 30 31 34 36 30 30 31 30 30 30 33 31 30 30 43 46 34 30 43 46 33 30 43 46 33 30 44 30 35 30 44 30 38 30 43 46 33 30 43 46 33 30 44 30 35 30 44 30 33 30 43 46 33 30 43 46 33 30 44 30 31 30 44 30 35 30 43 46 34 30 43 46 33 30 43 46 33 30 36 30 42 45 33 30 42 45 35 30 42 44 45 30 42 44 32 30 42 44 43 30 42 45 35 31 32 39 42 43 46 39 42 31 35 33 42 30 33 34 46 45 34 30 30 38 45 35 31 34 30 31 42 35 31 34 30 35 31 34 30 35 31 34 30 36 34 43 46 43 45 30 30 30 30 44 43 39 39
```

Battery 4 request:
```
7E 32 35 30 34 34 36 34 32 45 30 30 32 30 34 46 44 32 41 0D
```

Battery 4 response
```
7E 32 35 30 31 34 36 30 30 31 30 30 30 34 31 30 30 43 46 34 30 43 46 35 30 43 46 35 30 44 30 38 30 44 30 31 30 43 46 33 30 43 46 34 30 44 30 34 30 44 30 31 30 43 46 34 30 43 46 33 30 44 30 31 30 44 30 34 30 43 46 34 30 43 46 35 30 43 46 34 30 36 30 42 44 42 30 42 45 31 30 42 44 43 30 42 43 41 30 42 44 39 30 42 44 45 31 32 33 44 43 46 39 43 31 35 37 37 30 33 34 46 46 37 30 30 38 45 35 31 34 30 31 42 35 31 34 30 35 31 34 30 35 31 34 30 36 34 43 46 37 42 30 30 30 30 44 43 39 31
```

All 4 request  commands use same logic. To access battery 2,3,4 ADR and INFO must be accordingly 32, 33, 34:

| Field    | Value | ASCII     | Hex (ASCII-encoded) | Description             |
|----------|-------|-----------|----------------------|-------------------------|
| SOI      | ~     | `~`       | `7E`                 | Start of Info           |
| VER      | 25    | `2`, `5`  | `32 35`              | Protocol version 2.5    |
| ADR      | 01    | `0`, `1`  | `30 31`              | Address = Battery 1     |
| CID1     | 46    | `4`, `6`  | `34 36`              | Iron Lithium battery    |
| CID2     | 42    | `4`, `2`  | `34 32`              | Get Analog Info         |
| LCHKSUM  | E0    | `E`, `0`  | `45 30`              | Fixed for LENID = 02    |
| LENID    | 02    | `0`, `2`  | `30 32`              | INFO is 1 byte          |
| INFO     | 01    | `0`, `1`  | `30 31`              | Target = Pack 1   **Must be same as ADR**      |
| CHKSUM   | FD30  | `F`, `D`, `3`, `0` | `46 44 33 30` | Full command checksum   |
| EOI      | \r    | —         | `0D`                 | End of Info             |

And this is the response logic should be same as already in BSC. The documentation of PACE (and to my understanding most of the china RS232 protocol: 

[72651=11625-PACE-RS232commuciation-protocal（PACE-RS232-TY16S）-20180705.pdf](https://github.com/user-attachments/files/19503600/72651.11625-PACE-RS232commuciation-protocal.PACE-RS232-TY16S.-20180705.pdf)

on top of the INFO request there are 3 more request types. here is the overview of all 4 types. I'm not sure if other info is needed for BSC, but maybe the balancing status is:
| # | Command (HEX)                                                                 | Function           | CID1 / CID2 | Description                                                                                                                                          |
|---|-------------------------------------------------------------------------------|--------------------|-------------|------------------------------------------------------------------------------------------------------------------------------------------------------|
[... cut ...]
| 3 | `7E 32 35 30 31 34 36 34 34 45 30 30 32 30 31 46 44 32 45 0D`                 | Get Warning Info   | 46H / 44H   | Requests warning and fault status: over/under voltage/current, temperature faults, MOS faults, protection states, and balance status.              |
| 4 | `7E 32 35 30 31 34 36 43 31 30 30 30 30 46 44 39 41 0D`                       | Get Software Version | 46H / C1H   | Requests the BMS firmware version string. Expected response: 20 ASCII characters, padded if needed.                                                  |

So to my understanding nothingh else that allowing RS232 coneection should be done to BSC as the protocol is already impemented.

martinarva, 2025-03-28:
So my plan for tomorrow is:
1. to  buy a MAX3232 Serial RS232 TTL converter, 
2. connect it to Lilygo TCAN485 UART- RX IO5 and  TX IO18 
3. Change  #define TCAN485_RS485_TX_PIN 22 to 5
4. change #define TCAN485_RS485_RX_PIN 21 to 18
5. then tweak GobelBms_PC200 a little bit to request the correct payload

i'll let you know how it goes.

martinarva, 2025-03-29:
So as expected, it worked!!

So I wired the MAX2232 to GPIO 5 (RX) and 18 (TX)

[image]

Only two changes

One in GobelBMS_PC200.cpp
  u8_lData[6] = 0xff -> u8_lData[6] = address;

and

 
BscSerial.cpp

```
  #ifdef LILYGO_TCAN485
  if(u8_devNr==2) // Hw Serial 0
  {
    Serial2.end();
    Serial2.begin(baudrate,SERIAL_8N1,TCAN485_RS485_RX_PIN,TCAN485_RS485_TX_PIN);
    serialDeviceData[u8_devNr].stream_mPort=&Serial2;
  }
```

to

```
  #ifdef LILYGO_TCAN485
  if(u8_devNr==2) // Hw Serial 0
  {
    Serial2.end();
    Serial2.begin(9600, SERIAL_8N1, 5, 18);  // RS232 on GPIO18/5
    BSC_LOGI(TAG, "RS232 active on GPIO18 (RX) and GPIO5 (TX)");
    serialDeviceData[u8_devNr].stream_mPort=&Serial2;
  }
```

So just pretty much hardcoded BSC to use the MAX2232 rs232 instead of built in RS484

Everything else worked out of the box and i got a result:

[image]

Fuendiger, 2025-03-29:
Then you probably didn't test it with the test version I made for you! In that one, I changed `u8_lData[6] = address;`.
It should have worked if you had tested it with an RS485 -> RS232 adapter.

martinarva, 2025-03-30:
> Then you probably didn't test it with the test version I made for you! In that one, I changed `u8_lData[6] = address;`.
> It should have worked if you had tested it with an RS485 -> RS232 adapter.

Yes, I tested the latest version from git with these two changes.

Fuendiger, 2025-03-30:
I have added a second entry to the device list. 
[image]

The difference in the RS232 version is only 'u8_lData[6] = address;'. In the RS485 version this must be 0xff. This should then also allow the data to be read via RS232. 
The change will be included in the next test version. 

Note: The change is initially only in the Insider version.

martinarva, 2025-03-31:
> I have added a second entry to the device list. [image]
>
> The difference in the RS232 version is only 'u8_lData[6] = address;'. In the RS485 version this must be 0xff. This should then also allow the [...]

I'm now planning to order the BSC official hardware and have a question. Is there a free GPIO-s to connect the RS232 adapter like MAX2232?

Fuendiger, 2025-03-31:
I recommend an RS485 to RS232 adapter. For interference immunity, I would use RS485 from the BSC to the BMS, and then connect it to RS232.

There are no free GPIOs. But you can, for example, remove U8 and close JP11 and JP12 (solder jumpers). Then you have RX/TX from serial 0 on X2.7/8.
The advantage is that the electrical isolation is maintained.

[image]  
[image]

tueftler0, 2025-03-31:
As i wrote in the chat: An RS232 connection is not a good choice for use in harsh environments.

You have connected it successful, so one short question about the wiring: 
The TX-RS232-Pin3 of the BMS is the Output, right? Sometimes the Chinese people have different descriptions than the rest of the world...

[image]

There is a single-device extension corrently in development. It's almost finished and worked flawlessly for eight beta testers. I could imagine adding the RS232 port there as well...

martinarva, 2025-04-02:
> As i wrote in the chat: An RS232 connection is not a good choice for use in harsh environments.
>
> You have connected it successful, so one short question about the wiring:
> The TX-RS232-Pin3 of the [...]

My batteries have a RJ11 port for RS-232. There are four wires:
1 - GND
2 - RX
3 - TX
4 - GND

But I have seen RJ-12 (6 wire) and even RJ-45 ports for RS-232 on different Chinese batteries.

tueftler0, 2025-04-02:
> RJ-11 four wires are the middle four wires of RJ-12, so it's the same.
Yes, thats the same mechanical part with different pinning.

> The TX-RS232-Pin3 of the BMS is the Output, right? Sometimes the Chinese people have different descriptions than the rest of the world...
Could you please send me an answer to that, please.

mmoe-milano, 2025-04-02:
I think we have to clarrify an additional software issue concerning RS232. PACE PC200 BMS exists in an older and newer hardware version. The older version is not be supported further on. The newer V2 version works with older firmware versions with a baudrate of 9600 whereas the last SW version switch has been switch to 115200 baudrate. https://docs.google.com/document/d/1LCbIgYqBTsav8lQxt6ITq0Jt3TlQ1zuLrp2W7DDivrQ/edit?tab=t.0

Fuendiger, 2025-04-03:
@martinarva  must have the 9600 baud version, otherwise it wouldn't have worked.
Is there any difference besides the baud rate?

mmoe-milano, 2025-04-03:
@Fuendiger I recognized that too, therefore I mentioned it. Maybe @martinarva can update its BMS firmware if possible. But all newer feature updates and fixes come up with 115200 baudrate. Another issue is that one must be aware that @martinarva tested it with a BMS tool, I guess this it the same as mine. The BMS tool iterates the pack address itself (see image)

[image]

martinarva, 2025-04-03:
> [@martinarva](https://github.com/martinarva) must have the 9600 baud version, otherwise it wouldn't have worked. Is there any difference besides the baud rate?

i can try the 115200 baudrate and let you know if it works.

otherwise it has been stable and working as it should for few days.

mmoe-milano, 2025-04-03:
Possible to update the firmware @martinarva of the BMS ?

martinarva, 2025-04-03:
> [@Fuendiger]Maybe [@martinarva](https://github.com/martinarva) can update its BMS firmware if possible. But all newer feature updates and fixes come up with 115200 baudrate. 

My battery manufacturer unfortunately went bankrupt, which means there’s no longer any support or firmware updates available. The BMS in my battery is manufactured by PACE, but they don’t offer direct support either, as their business model relies on battery manufacturers customizing and distributing the firmware themselves.

Although my battery uses the same PACE BMS hardware as GoberPower, there are key differences that make GoberPower’s firmware incompatible with my system. This was a major obstacle, and one of the primary reasons I started looking for alternatives—ultimately leading me to discover the BSC (Battery System Controller) project.

The most frustrating issue was related to forced charging behavior: when the battery’s SOC dropped below 15%, it would send a “Force Charge” command to the inverter, which then continued charging until the SOC reached 30% before the command was cleared. GoberPower faced the same problem but was able to release a firmware update to fix it. In my case, I’m stuck with the original firmware and had no way to address the issue—until now.

Thanks to BSC, I can finally bypass this limitation. I’m now able to discharge the battery all the way down to 0% SOC, and even at that point, the lowest cell voltage remains at 3.0V. That means I was previously leaving a significant portion of my usable capacity untapped. BSC has made a huge difference in maximizing the potential of my setup.

mmoe-milano, 2025-04-03:
@martinarva what I wanted to clarify - the latest firmware extension of @Fuendiger concerning alternative adaption 'PACE PC200 via RS232' works currently only with a baudrate of 9600 and cannot be used by others with a newer PACE PC200 BMS. For me this is not a problem cuz I wrote my own RS232 adaption which works in parallel to RS485. Of course I'm happy for you that you found your solution and @Fuendiger for doing it's gratefull job.

martinarva, 2025-04-04:
> [@martinarva](https://github.com/martinarva) what I wanted to clarify - the latest firmware extension of [@Fuendiger](https://github.com/Fuendiger) concerning alternative adaption 'PACE PC200 via [...]

Actually for me its not a problem as well, as i'm building my firmware myself. I also hardcoded the GPIO-s and could hardcode the baudrate as well.

However, there might me others wirth 9600 baudrate and maybe this could be a configurable in the UI? Also the RS232 module  GPIO configuration option from web UI would be nice. But a said, i can build my own firmware.

Fuendiger, 2025-04-04:
If the baud rate is the only difference, then I can simply provide another option in the BSC to select that. However, I haven't yet received confirmation that the baud rate is really the only change. I would need someone to test it.  

Configuring the GPIOs makes little sense in conjunction with the original BSC hardware. If someone doesn't want to use them, they have to change the code themselves. This is easily done.  
I've described above what you need to change on the original hardware to directly connect an RS232 transceiver (MAX232).  
https://github.com/shining-man/bsc_fw/issues/233#issuecomment-2767100306

asdt1803, 2025-04-25:
The RS232 protocol has not changed between hardware versions 1 and 2. I have 4 Gobel Packs in use. The first (master) was originally equipped with an hardware 1 BMS but I insisted to have that replaced so I can have the same firmware on all 4 packs. I read data from the master pack  for all 4 packs via RS232 with a python script to get the date into influxdb, visualize it in Grafana and also to integrate the data into HA. No changes necessary after replacing the BMS. The Baud rate changed in a later firmware revision. Both hardware revisions of the Gobel version of the Pace BMS initially used 9600 Baud for the RS232 communication. As I have the RS232 port in use, I would prefer to use the RS485C port at the master pack for the communication with BSC. This port is intended for the communication to a charger (like the CAN port) and has many different protocols to choose from. Maybe one of those protocols is already implemented in BSC or could be implemented?

shining-man, 2025-04-26:
It doesn't work on the RS485A port, regardless of the protocol used. The BMS uses an old and undocumented protocol on the RS485A port. I've done a lot of testing.

The RS232 port (currently at 9600 baud) and the RS485B port work.

closed 2025-04-26 by shining-man

martinarva, 2025-04-29:
> I recommend an RS485 to RS232 adapter. For interference immunity, I would use RS485 from the BSC to the BMS, and then connect it to RS232.
>
> There are no free GPIOs. But you can, for example, remove [...]

Hi. I just received the official BSC board and plan is to move to the official board but can't decide which way to go:
1. RS485 to RS232 adapater (https://www.aliexpress.com/item/1005005904514037.html). This still needs power which i would take from X4.5/7 (5v). So would it have a electrical isolation then?
2. remove U8 and close JP11 and JP12 and then use RX/TX from serial 0 on X2.7 for MAX3232. Again MAX3232 needs power so i would use the X4.5/7 (5v). <

First option will leave the BSC board intact but introduces a new probale point of failuse as it will be TTL - RS485 - RS232, second one breaks the U8, but then it will be TTL - RS232. 

@shining-man , what would be your recommendation?

tueftler0, 2025-04-29:
> 1. RS485 to RS232 adapater (https://www.aliexpress.com/item/1005005904514037.html). This still needs power which i would take from X4.5/7 (5v). So would it have a electrical isolation then?

Then, the galvanical isolation is shorted to the main power supply of the mainboard.
CAN isolation is still there.

> 3. remove U8 and close JP11 and JP12 and then use RX/TX from serial 0 on X2.7 for MAX3232. Again MAX3232 needs power so i would use the X4.5/7 (5v). <> 

You can perform this PCB changes as written.
U1_VCC and U1_GND are usable for powering then.

BUT
Exactly for these problems, we have develop a "Serial device extension" which you can find on bsc-shop.com .
These adaptor can supply with 8-70VDC galvanic isolated and have an additional RS232 communication port integrated.

martinarva, 2025-04-29:
> Then, the galvanical isolation is shorted to the main power supply of the mainboard. CAN isolation is still there.
>
>
> You can perform this PCB changes as written. U1_VCC and U1_GND are usable for [...]

You mean this one: https://bsc-shop.com/produkt/single-device-extension/

What will be the price for it and when will it be avaiable? Also can i power it from BSC board or  I need a separate power supply?

tueftler0, 2025-04-29:
Yes, exactly. This device.
The price depends on the selected configuration and is not fixed yet.
Ten beta testers have tested it - it is currently in production.
For a RS232 connection, you can power it with 3,3V-5V or the specified galvanically isolated internal 8–70 VDC power supply.

shining-man, 2025-04-30:
@martinarva 
The simplest and best way is the "serial device extension" mentioned by @tueftler0 .
I would convert from RS485 to RS232 just before the battery. RS485 has a higher immunity to common-mode noise.