shining-man/bsc_fw #234: Autobalancer - Triggerbefehl zum Balancen

erwinh-at, 2025-03-13:
Ich hätte gerne die Möglichkeit dem Autobalancer per Trigger (Digitaleingang) den Befehl zu geben einen Balancevorgang einzuleiten obwohl der Intervall noch nicht abgelaufen ist.
Nach erfolgreichem Balancing soll der Intervall wieder neu starten (Dauer wie eingestellt).

Mir persönlich geht es darum dass die Batterien per externem Befehl voll laden dürfen (z.B. vor einer Schlechtwetterphase), weil ich normalerweise die Ladung bei ca. 90% SoC beende (Zellenschonung).

RRonGit, 2025-03-15:
Das hätte ich auch gerne, vielleicht sogar über die API triggerbar

erwinh-at, 2025-03-29:
Außerdem könnte man den externen Befehl verwenden um bei Netzausfall die Lade-Limitierungen außer Kraft zu setzen, damit die PV auch ohne Überschusseinspeisung weiterhin produzieren kann bis der Akku endgültig voll ist.

shining-man, 2025-03-29:
> Außerdem könnte man den externen Befehl verwenden um bei Netzausfall die Lade-Limitierungen außer Kraft zu setzen, damit die PV auch ohne Überschusseinspeisung weiterhin produzieren kann bis der Akku endgültig voll ist.

Den Wunsch verstehe ich nich wirklich.
a) Was hat das mit dem Autobalaner zu tun?
b) Sehe ich als  extrem Gefährlich, wenn plötzlich alle Ladelimitierungen wegfallen. Dann wird geladen bis das BMS eingreift. Was soll der Sinn sein? Die Ladelimitierungen sollen doch den Ladeprozess steuern und die dürfen doch keinesfalls deaktivierbar sein.

erwinh-at, 2025-03-29:
Entschuldigung, habe ich vielleicht falsch rüber gebracht.
Ich meinte natürlich alle Begrenzungen die der Autobalancer sowieso deaktiviert, wie Ladestromreduzierung SoC, Ladespannung wird auf Balancespannung geändert usw.

So eine Funktion für den Fall Inselbetrieb wurde schon mal irgendwo gewünscht. Für das könnte man aus meiner Sicht den Autobalancer missbrauchen.

Fuendiger, 2025-04-02:
> Entschuldigung, habe ich vielleicht falsch rüber gebracht. Ich meinte natürlich alle Begrenzungen die der Autobalancer sowieso deaktiviert, wie Ladestromreduzierung SoC, Ladespannung wird auf [...]

So ganz habe ich noch nicht verstanden was der Wunsch ist. Kannst du dazu ein neues Issue aufmachen.
Mein Wunsch wäre: Ein Issue, ein Thema. Das macht es für mich übersichtlicher, da ich dann am Titel schon sehe um was es geht.

erwinh-at, 2025-04-02:
> So ganz habe ich noch nicht verstanden was der Wunsch ist. Kannst du dazu ein neues Issue aufmachen. Mein Wunsch wäre: Ein Issue, ein Thema. Das macht es für mich übersichtlicher, da ich dann am [...]

Dazu braucht es kein neues Issue, das war nur ein weiterer Gedanke zu der Funktion die ich mir gewünscht habe.
Also der Triggerbefehl zum Start des Autobalancer bringt für mich gleich diese Funktion mit.

Um den Hintergrund zu verstehen ein Beispiel:

Es ist ein sonniger Tag, der Akku ist bereits zu 90% voll und wird aufgrund der Einstellungen (z.B. SoC abhängige Ladestrombegrenzung) nicht mehr oder nur mehr sehr schwach geladen. Der PV Überschuss wird ins Netz eingespeist.
Jetzt kommt ein Netzausfall aus welchem Grund auch immer. Überschuss kann nicht mehr eingespeist werden, die Laderegler regeln ab bzw. (zumindest bei Victron) die AC-gekoppelten PV Inverter (im Inselnetz) werden per Frequenzanhebung herunter geregelt oder abgeschaltet.
In dem Moment des Netzausfalls bekomme ich von der Victron-Anlage oder einem Netzüberwachungsrelais einen Meldekontakt geschaltet, der an einem digitalen Eingang des BSC eingelesen wird. Dieser wiederum setzt mir einen Trigger mit dem dann der Autobalancer den Startbefehl bekommt. Somit wird die SoC abhängige Ladestrombegrenzung deaktiviert und der Akku weiter geladen. Die PV-Leistung wird nicht mehr abgeregelt.
Kommt 10 Minuten später das Netz wieder zurück, schaltet der Netzausfallkontakt wieder um. Das Balancing geht vermutlich weiter (muss ich erst testen ob für diese Funktion ein Impuls ausreicht) bis es abgeschlossen ist oder die PV Leistung nicht mehr ausreicht um weiter zu laden.

Danke für die Umsetzung, werde ich demnächst ausprobieren!

closed 2025-04-26 by shining-man